wills106/homeassistant-solax-modbus #238: Possible Missing Solis Data Value

J14111, 2022-12-12:
on the Solis official portal it shows you the power going to the battery in watts/kw but I cant find this parameter exposed, its possible its missing or they calculate the value from the volts/amps?

wills106, 2022-12-13:
You have "Battery Power" which is two way.

You also have "Battery Charge Today" & "Battery Discharge Today"

J14111, 2022-12-13:
Thanks I will look at that

J14111, 2022-12-14:
As its two way and it doesnt seem to be + or - do you know how I can differentiate as its messing my graphs up

wills106, 2022-12-14:
Does [2022.12.7b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.7b1) have Battery Input / Output Energy the correct way round?

J14111, 2022-12-14:
I will test thanks

J14111, 2022-12-14:
I am getting this 
[image]

I think it may be a shortcoming on the Solis as I have spoken to someone else, I think I will need to do some logic in my reporting to get the correct data unless you have a solution?

wills106, 2022-12-14:
add '_charge' to the end of it.

Do you see battery_input_power and battery_output_power though?

J14111, 2022-12-14:
Seems good so rat will have to check once the sun goes down :)

[image]

J14111, 2022-12-14:
It does not work correctly, Battry Input energy works the same as "Battery Power" used to, it shows both in and out. Battery Output Enegry shows nothing at all

wills106, 2022-12-14:
There is a register that tells you if the battery is charging or discharging. I guess you could try and build that into a template sensor so you know if "Battery Power" should be positive or negative.

wills106, 2022-12-14:
I don't know if it will work or not? But try [2022.12.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.7b2)

I have renamed battery_power_charge back to the original battery_power.
It this works I'll set "battery_power" & "battery_charge_direction" both to hidden / disabled by default.

J14111, 2022-12-14:
Thanks it shows as a 1% now and its discharging, is that what you expect?

[image]

wills106, 2022-12-14:
Well, it's not meant to be 1% (only 1). That's because I copied and pasted without checking 🤣
But other than that yes.

So "Battery Output Energy" should display a value and "Battery Input Energy" should be 0.
Is that correct?

wills106, 2022-12-14:
[2022.12.7b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.7b3) corrects the % and the incorrect 'DEVICE_CLASS_BATTERY'

J14111, 2022-12-14:
> Well, it's not meant to be 1% (only 1). That's because I copied and pasted without checking 🤣 But other than that yes.
> 
> So "Battery Output Energy" should display a value and "Battery Input Energy" should be 0. Is that correct?

Yes that appears to be working now but I wont know for sure until I see sun tomorrow now and the batteries charge, thanks so much for all your help. I will let you know how it goes.

[image]
[image]

wills106, 2022-12-14:
If it works fine I'll release it as a proper release and I'll also disable the view of "battery_power" & "battery_charge_direction" as they don't need to be visible.

J14111, 2022-12-15:
The sun has come up and it works like a charm thanks so much!

So Charge Direction 1 = battery discharging and 0 = battery charging?

[image]
[image]

wills106, 2022-12-15:
Yes that matches what I said here https://github.com/wills106/homeassistant-solax-modbus/issues/238#issuecomment-1351581196

closed 2022-12-15 by J14111