wills106/homeassistant-solax-modbus #239: Workaround for #48 and #226

mickemartinsson, 2022-12-12:
I have really problems with Idle-mode. We got a lot of snow and after discussions with the technical staff at my dealer, we discovered that lack of watts from the solar panels and a SOC below the minimum level, in my case 15% (for EPS backup), puts the inverter in Idle mode and a battery-wake-up do not make it Normal mode unless it is in Force-mode. They are in discussions with Solax about if the Idle-mode can be disabled (not only me here with problems of snow....).  I have found, hopefully, a workaround (until it is finally solved). 

I use Node-Red and do the following sequence:

*) Check if battery SOC is 16% or less -> switch from Self Mode to Force Time Use
*) Set Charge from Grid to Disable

Now it works like Self Mode but without the inverter going into Idle (Backup mode does not work)

When it is time to charge -> 

*) Set Fore TIme Use (just in case)
*) Set Charge from Grid to Enable

When it is time to not charge ->
*) Set Self Mode

So far, no Idle-mode, and my normal flows work. I normally have charging time from 00.00 -23.59 and just switch between Force and Self use.

richsimpson, 2022-12-13:
I do similar on the Gen 4 X1 Hybrid

"Self Use Mode" with the charge start/end both 2030hrs and Discharge 0000-2359hrs

When the SOC get to 13% my node looks for it then hitting 11% - so only effective when discharging. At that point it switches the inverter from "Self Use Mode" to "Manual Mode". In "Manual Mode" it is set to the sub setting of "Stop Charge & Discharge". This avoids the idle mode and sits nicely waiting for the next command which comes at 2030hrs and is to be in "Manual Mode" and "Force Charge" as I get to off peak tariff.

[image]

[image]

I also have odd thing where if the battery is full and when power from solar drops off, the battery doesn't always kick in so I have this running too which is basically turning a high power device on for a short time to 'jump start' the battery.

[image]

Has worked fine for a couple of weeks or so now since I ran in to the Idle issue. 

I also have an alert to flag to me if the inverter is in Idle mode and another if it goes into Locked mode for any reason (reboot etc).

mickemartinsson, 2022-12-13:
I have had a discussion with both Solax Service & Tech. Support Engineer and my supplier. I have told them I need to disable this Idle-mode totally. I hope they can provide some fixes. We do have snow until march....

They had an option that:
*) Set System off - for 1 minute
*) Set System on - then it will set the normal mode for a little period and then it will accept any changes

I don't think that's a good option so I will continue with mine until they can provide an option to disable Idle mode when batteries are present.

Here is my short Node Red - I do not have a manual mode on mine. 

[image]

richsimpson, 2022-12-13:
Yes I would stick to your logic and see if an update comes along to solve it. It is likely to knock something else out which is frustrating!

richsimpson, 2023-01-16:
I have just had a firmware update that seems to fix all of my issues. I haven't discovered the new bugs yet, there are bound to be some! 
I have posted the detail and version here if it is any help. Enabling the version sensors in Home Assistant will show what you currently have other than the battery items which I had to get from the solax cloud app.
https://github.com/wills106/homeassistant-solax-modbus/discussions/297
[https://github.com/wills106/homeassistant-solax-modbus/discussions/297](url)

wills106, 2023-01-16:
> battery items which I had to get from the solax cloud app

I did briefly have the BMS sensors in but ended up removing them as it broke the Integration for many people.