wills106/homeassistant-solax-modbus #240: Integration was working then stopped

willbiker, 2022-12-12:
Hi! I had wills modbus integration up and running pulling in registers and data nicely to my HA. Then it randomly stopped and I got an error saying inverter serial number can't be found. I updated HA and also updated the integration with no luck. I then deleted the integration and reinstalled with no luck. I fixed the ip of my inverter so it can't change and I can still ping it. Any ideas how I can get this working. Here is the log below.

Thanks in advance

Will
[image]

infradom, 2022-12-12:
One possibility: The inverter was probably already asleep when you reinstalled
Maybe it will work again when there is PV

Patrik1973, 2022-12-13:
> Hi! I had wills modbus integration up and running pulling in registers and data nicely to my HA. Then it randomly stopped and I got an error saying inverter serial number can't be found. I updated [...]

I don't think so, it isn't that inverter is sleeping...I have the same issue, the same errors and my inverter produce power.
In HA at the devices integration shows 4 entities only.

infradom, 2022-12-13:
@Patrik1973 Even though your problem looks similar, it is probably not the same. The @willbiker log shows that the system tried to awake the inverter "cannot wakeup inverter" so it was asleep. In the current version, we still have issues with waking up an inverter automatically, so that part does not work yet.

infradom, 2022-12-13:
@Patrik1973 In most cases, the system being unable to read the serial number is caused by bad wiring or bad ip addres/port configuration issues.
Were you able to communicate with the inverter in the past from within this integration?

infradom, 2022-12-13:
Correction @Patrik1973 : I see in other posts your system worked with previous versions. Sorry ...

Patrik1973, 2022-12-13:
@infradom Yes, on saturday the integration was stopped and shows 4 entities only, then I tried to reinstall it few times and situations is the same still.
At the attachment you can see many errors.

https://github.com/wills106/homeassistant-solax-modbus/issues/234#issuecomment-1345501318
[Modbus Error.txt](https://github.com/wills106/homeassistant-solax-modbus/files/10216526/Modbus.Error.txt)

infradom, 2022-12-13:
@Patrik1973 It is important that you install the integration when the inverter is awake
So you may need to remove / restart / reinstall at a proper time when the inverter is awake
Maybe you already tried that
In the mean time, I am examining your log ...

Patrik1973, 2022-12-13:
@infradom Yes I know it, I did it...ok I will try it again.
I have on HA last version installed and on integration SOLAX modbus I have 2022.12.4 version

willbiker, 2022-12-13:
Hi All, Thanks for the comments! Indeed the inverter status doesn't seem to make a difference. The IP is fixed and I can ping is so there should be no issues with connectivity as it worked  few weeks ago flawlessly. I am using the wifi dongle which the description said isn't supported however as as mentioned, it was working fine. Maybe I need to get an ethernet adapter instead

infradom, 2022-12-13:
Ok, I get it .. We really have problems with the wifi dongle. I personally prefer not to put any efforts in debugging systems that try to use modbus over pocketwifi (or pocketlan). It is simply unpredictable.

infradom, 2022-12-13:
@Patrik1973 Do you also connect through  a pocketwifi adapter? If so, sorry that explains a lot, but I cannot help you then ..
Have you submitted the details of your installation ? type of inverter, connection method, ... ? I did not find it in either thread

willbiker, 2022-12-13:
Sure understood. So an ethernet dongle may be the answer for me, worth a try.

I think the 'cannot wake' error may have been as a result of me pressing the 'unlock' button. I will invest in an ethernet dongle and will try again

Thanks

willbiker, 2022-12-13:
I assume an ethernet dongle is needed for ethernet connection for the x1 Hybrid inverters? There is no built in network port I assume? What method of network connection do you use?

infradom, 2022-12-13:
@willbiker  : Not an Ethernet (pocketlan) dongle, that wont work either but a modbus RS485 to Eth like https://www.waveshare.com/wiki/RS485_TO_ETH_(B)
Our wiki descirbes this

wills106, 2022-12-13:
Have a look in the Wiki for the list of supported RS485 - Ethernet adaptors.
Take note of the unworking / unreliable section below.

Patrik1973, 2022-12-13:
Yes, I also use the wifi pocket adapter, ok, if also LAN adapter is not supported made from Solax I will try to change a connection to other.

I'm sorry for my issues...las week I started PV and tried the integration and everything seems ok and suddenly everything failed.

My configuration is Solax X3 G4

infradom, 2022-12-13:
@Patrik1973 In the future I might develop a HA addon that converts serial RS485 modbus to TCP based modbus. That would remove the need for an external converter, a simple low cost USB-RS485 dongle would then be suffient. Unfortunately, I have a big backlog and need to do other things first. For my Gen4 X3: I use the Waveshare  RS485 to ETH (B) converter

Patrik1973, 2022-12-13:
@infradom Thanks a lot for your advice, the fastest for me if I order the Waveshare RS485 to LAN (B) converter.

wills106, 2022-12-13:
If your switch supports POE there is a POE version. Saves needing to supply power to it.

willbiker, 2022-12-15:
Hi Again.  I'm trying to learn all about integrating to my inverter using RS485 and converters. The Waveshare RS485 TO ETH (B) looks to be a reliable but what does the ethernet output connect to? I run HA on a RPi4, do I connect the eithernet directly to the RPI or to my router for a connection through the network?

My RPI can be moved next to my Inverter, would a RS485 TO USB, plugged directly into my RPI work also?

Thanks

Will

wills106, 2022-12-15:
Either a Waveshare RS485 TO ETH (B) plugged in to your router or RS485 to USB adaptor plugged directly to your RPI will work fine.
Just depends on which is easier to install.
RS485 to USB is the cheapest option, just take note of [No 2](https://github.com/wills106/homeassistant-solax-modbus#known-issues) you can filter those out if you configure your logger correct I believe.

wills106, 2022-12-17:
Closing down as Issue related to Pocket WiFi.

If you have further issues when you get an RS485 - Ethernet / USB adaptor then feel free to raise a new issue.

closed 2022-12-17 by wills106