wills106/homeassistant-solax-modbus #244: pv_power_total unit of measurement Sofar HYD 10ktl-3ph

juraurban, 2022-12-13:
PW Power total sensor has wrong decimal value.

[image]

Setting scale = 0.1 at this register, helps.

Thank you for fixing that in your code as well.

wills106, 2022-12-13:
Are you on 2022.12.x I thought I corrected that during 0.7.0bx before releasing the stable branch?

wills106, 2022-12-14:
With scale at 0.1 does it ever return 2 digits after the decimal ie 2.34kW or is it only ever 2.3kW?
I might need to drop the rounding to 1 as well if it's ever only 2.3

wills106, 2022-12-14:
PV Power Total is corrected in [2022.12.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.6)

wills106, 2022-12-15:
Have you managed to test this?
If so and it's correct I'll close the issue down.

darkrain-nl, 2022-12-17:
For my device the 0.01 and rouding 2 were working fine, now I get this:
[image]

wills106, 2022-12-17:
I have just checked and you have a PV only Inverter 'SS2E' don't you?

So it must have a slightly different rounding on PV Power Total

darkrain-nl, 2022-12-17:
> I have just checked and you have a PV only Inverter 'SS2E' don't you?
> 
> So it must have a slightly different rounding on PV Power Total

Yes that is correct, is there any smart way to fix that? 
I do have a workaround and that is to use a different sensor for this reading, I just use sensor.sofar_active_power_output_total which seems to be the same thing anyway in my specific case.

juraurban, 2022-12-17:
Sorry for late answer. Unfortunately I have had panels covered with layer of snow since last week therefore 0. Shows only 1 decimal
[image]

wills106, 2022-12-17:
> Yes that is correct, is there any smart way to fix that?

Yes I can have a PV only version and a Hybrid only version of the same sensor with the different scaling required.

juraurban, 2022-12-17:
[image]

juraurban, 2022-12-17:
[image]

juraurban, 2022-12-17:
Inverter shows 10W current production

wills106, 2022-12-17:
As PV Power Total is only designed from Sofar to be in 0.1 resolution it won't display anything less than 100w

wills106, 2022-12-17:
[2022.12.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.8) will correct "PV Power Total" scaling on KTLX-G3 Inverters

darkrain-nl, 2022-12-17:
> [2022.12.8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.8) will correct "PV Power Total" scaling on KTLX-G3 Inverters

Thanks, can confirm this works great!

wills106, 2022-12-17:
I'm going to close this down now as "PV Power Total" is now correct for the Hybrid and PV only Inverters.

closed 2022-12-17 by wills106