wills106/homeassistant-solax-modbus #245: Additional BMS data for Solax X1/X3 - G4

shaarkys, 2022-12-13:
Hi, 
would it be possible to include additional data from BMS ? It would be nice to see some details per installed  battery module, however not sure it's possible. Thank you, btw so far your integration works fine for X3-G4 (just reading values).

https://gbc-solino.cz/wp-content/uploads/2022/07/Hybrid-X1X3-G4-ModbusTCPRTU-V3.21-English_0622-public-version.pdf 

0x03:Read Holding Register（BMS Info） 
0x04:Read Input Register

wills106, 2022-12-14:
Which specific registers would you like?

I don't know if the holding registers will provide much useful Information?
I'm not sure if BMS_BatteryCapacity is the installed capacity (Like on Gen3) or the available energy.

shaarkys, 2022-12-14:
Thank you and apologies, I'm not ModBUS expert, I just though some of the useful data might maybe come from : 

 0x04:Read Input Register Parallel page 35 – 0x0204 - 0x02ED – slave1 data and so on, registers BatPower,BatVoltage, BatCurrent, BatFaultMessage a BatCapacity for each unit ?
Also maybe would be interesting to see BMS – BMS warning code page 30 and BMS_BatteryCapacity– 0x0026~0x0027  … but as you said, not sure it's installed capacity or available.

Happy to test it, if there is a way. I just realized we are pretty blind when it comes to batteries on Solax, because we just see some of the parameters and only those reported by BMS for all units. Eg. SOC means almost nothing - I guess it's average of available energy from each unit, while if they would be dis-balanced, you will almost not know...etc. 

I'm also discussing with another developer of another integration (not on HA), he is also not sure but if I can help anyhow with testing, happy to do so.

wills106, 2022-12-14:
Do you have more than one Inverter connected in parallel if you want to test the Registers on P35?

wills106, 2022-12-14:
> Also maybe would be interesting to see BMS – BMS warning code page 30

That isn't currently supported (decoding of individual bits) it's on the [Todo List V2](https://github.com/wills106/homeassistant-solax-modbus/discussions/200#discussioncomment-4242410) though.

shaarkys, 2022-12-14:
> more than one Inverter

Apologies, I have only one inverter, I though it's for slave battery modules... so you are saying, unsupported, right ;-(

wills106, 2022-12-14:
Added BMS Battery Capacity into 2022.12.6 like I say I don't know if it's installed capacity like on Gen3 or available capacity?

The P35 registers are for parallel connection of inverters.

With BMS info (holding registers) I'm only expecting it to return serial numbers to be honest? But I can have a look at them.

shaarkys, 2022-12-14:
> BMS Battery Capacity into 2022.12.6 like I say I don't know if it's installed capacity like on Gen3 or available capacity

Unfortunately reports 0Wh...

[image]

wills106, 2022-12-14:
There are a lot of entities disabled by default. They don't really provide anything useful day to day.

wills106, 2022-12-15:
Has "BMS Battery Capacity" changed at all during the day? Or still sat at 0?

wills106, 2022-12-15:
Just had another look at the docs, even though it doesn't say it's a unit32 register it kind of hint's it is one.
Could you see if [2022.12.8b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.8b1) returns something other than 0?

shaarkys, 2022-12-15:
I have installed 2022.12.8b1 but seems it's still 0W (also didn't update/change during night)

[image]

wills106, 2022-12-15:
Ok, was worth a try.

I am guessing it's one of the registers that never got implemented?

shaarkys, 2022-12-15:
Yep, maybe intentionally....seriously, I have 3 packs/modules, 17kWh and...I know nothing about their real capacity. I can maybe find a way to calculate it from 0-100%, but it will be anyway for whole pack itself.

Eg. just FYI, Deye, the report (at least on device itself) following details : 
[image]

wills106, 2022-12-15:
What BMS system is that? Is that on another Inverter?

shaarkys, 2022-12-15:
Yep, as I wrote, it's Deye inverter, it's for info only - such level of details should be provided by each inverter/BMS, but in case of Solax, we can be dreaming I guess. 

Thank you for your help anyway, looking forward for next development. :-)

closed 2022-12-15 by shaarkys

wills106, 2022-12-15:
I wasn't aware of a brand of Inverters called Deye.

Are those batteries wired in Parallel?
On the SolaX they are wired in series, so behave more like a single battery.

wills106, 2022-12-15:
Like I expected "0x03:Read Holding Register（BMS Info)"

Just returns the Serial Numbers of the BMS and Batteries. Which doesn't give much insight / practical use)

shaarkys, 2022-12-15:
> called Deye.

Deye manufactures Sunsynk, also makes Fusion, Inge and SolArk :-) 

Not sure whether it's parallel or series honestly, it's brother installation, so I didn't check in such detail....

> Just returns the Serial Numbers of the BMS and Batteries

Yes, those values can be found on the SolaxCloud as well - might be needed, but not so often. Eg. in my case, BMS reports different S/N then it's on the case and Solax support don't know why :-)

wills106, 2022-12-15:
The Firmware is hidden by default. But it doesn't quite tally up correctly.

Might not be decoding it right.