wills106/homeassistant-solax-modbus #246: Error on 2022.12.4 - KeyError: 'plugin'

t3lmoazevedo, 2022-12-14:
Updating from 0.6.4 to 2022.12.4 fails on the integration with:

Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 90, in async_setup_entry
    _LOGGER.debug(f"Ready to load plugin {config[CONF_PLUGIN]}")
KeyError: 'plugin'

[image]

infradom, 2022-12-14:
Furthermore, we did not receive contexxt information (Hardware platform, OS platform, HA version, inverter type, ...
So it is very difficult to provide support with so little information

t3lmoazevedo, 2022-12-14:
@infradom thanks for the tip.
Removing and reinstaling fixed the situation ;) 
Great support here!

closed 2022-12-14 by t3lmoazevedo