wills106/homeassistant-solax-modbus #247: [BUG].  Battery energy input and output don't match

markxroberts, 2022-12-14:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version: 2022.12.5
2. HA Version: 2022.12.5
3. Inverter Brand: Solax
4. Plugin Used: 
5. First 6 digits of Serial:  H4502D
6. Connection Method: Waveshare RS485 to ethernet

In my Gen 3 setup Battery input in a 24 hour period is always substantially more than battery output, rather than net zero.
I've only had the batteries for 3 months, but over that time, net input shows as 357kWh and output as 179kWh using SolaX Battery Input Energy Today and SolaX Battery Output Energy Today.  I have a 2 battery setup.

How can this be?

There are no relevant error logs.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.

wills106, 2022-12-14:
Just so you know your Inverter is a Gen4 and not a Gen3.

The Integration only returns what's stored in the Inverter so something must have skewed the data.
Were both batteries installed at the same time?
I believe some firmware versions had problems with batteries getting stuck at certain percentages. So perhaps the Inverter was miss calculating?

markxroberts, 2022-12-15:
Sorry, yes Gen4 inverter.   Thanks for speedy reply.  Both batteries installed simultaneously.  No issues with battery percentage getting stuck: I see a gradual increase in charge.  Unless you mean stuck when discharging?  Don't see that either.  Otherwise would perhaps expect that discharge greater than charge.  This is a persistent issue but of variable magnitude.  Averages at 2 fold difference, but sometimes more and sometimes less.  I'll try to contact supplier if not likely an issue with data extraction.

wills106, 2022-12-17:
Closing down as the Integration can only report what's stored in the Inverter.

closed 2022-12-17 by wills106