wills106/homeassistant-solax-modbus #250: solax_battery_awaken not working on SolaX G4 hybrid - using wrong register?

M0LTE, 2022-12-17:
In #226 I [learned](https://github.com/wills106/homeassistant-solax-modbus/issues/226#issuecomment-1349045643) from @infradom that the awake button is defined as:
send value 1 to write register with address 0x56 hex or address 86 decimal

For my Gen 4 SolaX hybrid inverter (with LFP batteries) this does nothing - the inverter does not change from Idle to Checking or anything else on pressing the button, nor does the new auto-wake functionality work - any settings changes revert in the app shortly after being set.

I have queried the command with SolaX and received the following two slightly contradictory replies:

> 0x0056 is for X3-G2, and I checked your device is a G4 version, this address is not valid for G4 devices. If you want to wake up a G4 machine please try power control, and set a charging/discharging target to make the battery work.

> The 0x0056  register is for Lead-acid battery. If the inverter go to idle, you can use remote power control to wake it.

The second reply contained two screenshots:

https://i.imgur.com/pU2uQm6.png

https://i.imgur.com/2SbUE9Z.png

Hopefully the information above is helpful in getting solax_battery_awaken and auto-wake running.
Thanks as always!

wills106, 2022-12-17:
We are aware of Modbus power control registers (0x7e) they are used with the write_registers function which isn't complete to write to multiple registers are once.

At the moment write_registers can only write to a single register.
But we need to write to 5 registers at once.

wills106, 2022-12-17:
I'm closing this as it's the same / related to #226 so we can keep the conversation in the same issue and not spread over multiple.

closed 2022-12-17 by wills106

M0LTE, 2022-12-17:
Apologies - I felt it was distinct enough to warrant a separate issue - it's not specifically auto-wake, just wake.

wills106, 2022-12-17:
It's all kind of linked together.

Waking the Inverter up / waking the Inverter up so it can charge / discharge etc.