wills106/homeassistant-solax-modbus #263: [BUG]

klio-klio, 2022-12-18:
Hello, after the update do not get all data. Before everything worked wonderfully

**Mandatory details**
1. Integration Version: last
2. HA Version: Home **Assistant 2022.12.7**
3. Inverter Brand: Solax H3-Hybrid-8.0-D-E
4. Plugin Used:
5. First 6 digits of Serial:H3DE08F7......
6. Connection Method: Network Wired

Something went wrong reading from modbus
22:13:35 – (FEHLER) SolaX Inverter Modbus (benutzerdefinierte Integration) - Die Nachricht ist zum ersten Mal um 22:01:20 aufgetreten und erscheint 6 mal
SolaX error reading holding registers at device 1 position 0x206
22:01:20 – (FEHLER) SolaX Inverter Modbus (benutzerdefinierte Integration)
 
Vollständige Protokolle anzeigen
2022-12-18 21:49:45.602 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration solaxcloud which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-12-18 21:49:45.604 WARNING (SyncWorker_4) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-12-18 22:01:20.396 ERROR (MainThread) [custom_components.solax_modbus] SolaX error reading holding registers at device 1 position 0x206
2022-12-18 22:01:20.396 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 320, in read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 397, in read_modbus_registers_all
File "/config/custom_components/solax_modbus/const.py", line 182, in value_function_grid_import
2022-12-18 22:03:35.394 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 320, in read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 397, in read_modbus_registers_all
File "/config/custom_components/solax_modbus/const.py", line 182, in value_function_grid_import
2022-12-18 22:06:05.409 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 320, in read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 397, in read_modbus_registers_all
[image]

File "/config/custom_components/solax_modbus/const.py", line 182, in value_function_grid_import
2022-12-18 22:08:35.412 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 320, in read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 397, in read_modbus_registers_all
File "/config/custom_components/solax_modbus/const.py", line 182, in value_function_grid_import
2022-12-18 22:11:05.428 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 320, in read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 397, in read_modbus_registers_all
File "/config/custom_components/solax_modbus/const.py", line 182, in value_function_grid_import
2022-12-18 22:13:35.435 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
File "/config/custom_components/solax_modbus/__init__.py", line 320, in read_modbus_data
File "/config/custom_components/solax_modbus/__init__.py", line 397, in read_modbus_registers_all

wills106, 2022-12-18:
**Can you please go back to 2022.12.6**

This is the same as https://community.home-assistant.io/t/universal-solar-inverter-over-modbus-rs485-tcp-custom-component-formally-solax-no-pocket-wifi/140143/933?u=wills106

Also #262 

These BMS Info registers were introduced as Investigation for #245

I'm going to remove them as it appears not all Inverters like / have these registers available and they don't provide anything useful other than the serial numbers for your batteries.

klio-klio, 2022-12-18:
I'm back on Home Assistant 2022.12.6
![Screenshot 2022-12-18 235702](https://user-images.githubusercontent.com/76972213/208324020-ec82424f
[image]
-69a4-4d69-aba8-b88cbd94a636.jpg)
[image]

![Screenshot 2022-12-18 235722](https://user-images.githubusercontent.com/76972213/208324026-3e82abbb-bc0
![Screenshot 2022-12-18 235746](https://user-images.githubusercontent.com/76972213/208324055-d0cd66e1-2e69-402d-81a2-79a920962af
[image]
f.jpg)
5-4561-a
[image]
9c3-df0ff573c0b7.jpg)

wills106, 2022-12-18:
Did you remove the Integration from the integration page before downgrading and restarting HA and then adding it back again on the Integration page?

klio-klio, 2022-12-19:
No, now I have tried several variants. always restarted after removing integration.  Which variant should actually work? always the same result. I don't see through it anymore. Thank you for your efforts.

wills106, 2022-12-19:
2022.12.6 should work fine. Do you get errors in your log with 2022.12.6? Are you using the built in ethernet and not trying to connect to the Pocket Wifi?

klio-klio, 2022-12-19:
Internal network only

klio-klio, 2022-12-19:
2022.12.6 HA logs

lowprize, 2022-12-19:
> Did you remove the Integration from the integration page before downgrading and restarting HA and then adding it back again on the Integration page?
I do.

Version 12.07 works fine for X3-Hybrid-10.0-E

wills106, 2022-12-19:
@lowprize Does your Inverter serial start different to H3DE or H3UE?

lowprize, 2022-12-19:
@wills106 H3UE10....

lowprize, 2022-12-19:
@wills106 works with version 12.7
Version 12.8 does not.

wills106, 2022-12-19:
Ok, I thought you meant the latest 2022.12.8 was working.

lowprize, 2022-12-19:
Right, sry for misunderstood, Im bfu

klio-klio, 2022-12-19:
Serial:H3DE08F7...... Not working yet?

wills106, 2022-12-20:
I have removed these BMS Registers in [2022.12.9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.9)

klio-klio, 2022-12-20:
Hey Super!!!! functioned. also works with Home Assistant 2022.12.7 ? Can I support the project somewhere? sorry for my English. I have to translate everything.

lowprize, 2022-12-21:
Confirmed, version 12.9 works with H2UE..

wills106, 2022-12-21:
> Can I support the project somewhere?

There is a Ko-fi button in the readme, or you can help translate the config_flow if your language hasn't already been translated.

I'm going to close this Issue down and the BMS registers causing the problem have been removed.

closed 2022-12-21 by wills106