wills106/homeassistant-solax-modbus #264: [Serial Number cannot be read]

alienatedsec, 2022-12-19:
**Describe the bug**
Cannot make it working for some reason.

**Mandatory details**
1. Integration Version: 2022.12.08
2. HA Version: 2022.12.4
3. Inverter Brand: RHI-6K-48ES-5G Single
4. Plugin Used: What is that?
5. First 6 digits of Serial: 603105
6. Connection Method: Wi-Fi

**Detailed Error Log**
```2022-12-19 14:24:27.150 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2022-12-19 14:24:27.151 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2022-12-19 14:24:27.202 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2022 12
2022-12-19 14:24:27.203 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2022-12-19 14:24:27.231 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Solaxv01: trying to determine inverter type
2022-12-19 14:24:32.245 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] Solaxv01: reading serial number from address 0x80ec failed; other address may succeed
2022-12-19 14:24:32.249 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read Solaxv01 0x80ec serial number: None, swapped: False
2022-12-19 14:24:32.254 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] Solaxv01: cannot find serial number, even not for other Inverter
2022-12-19 14:24:32.254 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized Solaxv01 inverter type - serial number : unknown
2022-12-19 14:24:32.277 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2022-12-19 14:24:32.305 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}
2022-12-19 14:24:33.262 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2022-12-19 14:24:33.270 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2022-12-19 14:24:33.274 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2022-12-19 14:24:33.286 INFO (MainThread) [custom_components.solax_modbus.button] hub.awake button: None
2022-12-19 14:24:55.417 INFO (SyncWorker_7) [homeassistant.loader] Loaded solax from homeassistant.components.solax
```

**Additional context**

I am wondering how to fix this. It works with other integrations e.g., https://github.com/StephanJoubert/home_assistant_solarman but has many issues with data, so I am looking for alternatives.

Any advice?

wills106, 2022-12-19:
I would look / ask in here #188 
But it looks like people are using port 8899 and not the default 502 that is shown in the configuration option.

alienatedsec, 2022-12-19:
I think the port number was the first thing I checked.

Solarman requires providing the stick S/N. Mine is 10 digits starting with `4093xxxxxx`. Its firmware version is `MW3_15_0501_1.18`

I will also check the plugin

wills106, 2022-12-19:
You are probably best checking on that post how others are connecting. I thought the WiFi stick didn't provide access. But they are using something other than a RS485 adaptor by the looks.

alienatedsec, 2022-12-19:
@wills106 My WiFi stick works with Solismon and Solismod, which is based on Modbus communication. Any ideas on how we could have this supported?

alienatedsec, 2022-12-20:
Hi @wills106 [this wrapper is based on pysolarmanv5](https://github.com/jmccrohan/ha_pyscript_pysolarmanv5) and it introduces reading/writing to Modbus registers of the Solarman-based WiFi sticks.

[image]

I also wrote a script to read and write registers - [it's based on examples from Solismon3](https://github.com/slayer1551/solismon-3/blob/master/examples/client_example.py)

[image]

wills106, 2022-12-20:
I'm not sure why your Inverter isn't connecting if other people are using the WiFi stick. Unless there are different version of the stick?

The serial 6031 has been tested by other users.

alienatedsec, 2022-12-20:
@wills106 I got it... the WiFi stick I have needs the Modbus RTU (pysolarmanv5) to communicate as it doesn't support [native Modbus TCP](https://github.com/jmccrohan/pysolarmanv5/issues/5). 

The main question is if there is an easy way to integrate [pysolarmanv5](https://pysolarmanv5.readthedocs.io/en/stable/) into your integration.

wills106, 2023-01-05:
I have had a quick look at the protocol and I'm not sure if I have time at the moment to support it.

I don't think the current multi register read code would work as it looks like they build multiple reads differently and I am not sure if two register read is automatically turned into a 32bit register. So we would have to be careful not to have the auto builder read a block of two registers.

We would also have to modify the number, selects & sensor files etc to use the alternative method and either rework a plugin file just for solarmanv5 or translate it with perhaps a checkbox on initial setup to tell the Integration to use an alternative method to Modbus.

alienatedsec, 2023-01-05:
Thanks @wills106 I thought it would be the case. For now, the DLS-L should be enough for me now, but expanding the support to DLS-W will make this integration available to many others.

closed 2024-01-23 by alienatedsec