wills106/homeassistant-solax-modbus #269: Missing radiobuttons on configuration

gelse, 2022-12-21:
**Describe the bug**
When creating a new config (after removing previous installation / upgrade), the radio buttons for the inverter type are missing, thus leading to invalid user input:
`User input malformed: value must be one of [] for dictionary value @ data['plugin']`

**Mandatory details**
1. Integration Version: 2022.12.12
2. HA Version: 2022.12.7

**Detailed Error Log**
there is no error log, it is an UI error.

[image]

gelse, 2022-12-21:
as a fallback it would be nice to have a how-to somewhere how i can configure the integration via YAML.

wills106, 2022-12-21:
Have you upgraded from 0.6.4 or earlier without removing the Integration from the Integration page, restarting HA and adding it back to the Integration page?

gelse, 2022-12-21:
no, i followed the instructions by the book.  
i removed the integration from HA
i removed the integration from HACS
i restarted HA
i added the integration in HACS
i restarted HA again
i tried to add he integration in HA

gelse, 2022-12-21:
and also, as i said in the description above, there are NO errors of solax in the logs of HA.

wills106, 2022-12-21:
You are somehow missing the whole section from the config flow where you select which plugin you want.

On the Integration page have you tried forcing a refresh?
Ctrl + F5
Then try adding it again?

gelse, 2022-12-21:
yes, that was the first thing i tried (i am a developer myself ;-) ) - i just re-tried it, same behaviour.

wills106, 2022-12-21:
I have just done a fresh HACS download on a separate machine and I can't replicate this.

Has HACS downloaded the plugin_sofar plugin_solax files etc?

gelse, 2022-12-21:
after removal via HACS:
```
root@omv:~/links/homeassistant/config/custom_components# ls
alexa_media  edgeos  elasticsearch  hacs  monitor_docker  myjdownloader  openmediavault  spotcast  xiaomi_cloud_map_extractor  xiaomi_miot
```
then adding it via HACS:
```
root@omv:~/links/homeassistant/config/custom_components# ls
alexa_media  edgeos  elasticsearch  hacs  monitor_docker  myjdownloader  openmediavault  solax_modbus  spotcast  xiaomi_cloud_map_extractor  xiaomi_miot
```
and the content of solax_modbus seems fine to me.

wills106, 2022-12-21:
I'm on HA 2022.12.5 I'll try 2022.12.7 later today in case they made a change.

gelse, 2022-12-21:
aaaand another information:
i just tried to configure it with edge browser (normally: brave), which i NEVER used before to access home assistant at all, and i had the exact same behaviour.
as a last resort i try to configure it via Android in a minute.

but i am a bit ... nervous, i just want to get it working again now. i need the statistics.

wills106, 2022-12-21:
I'll upgrade to HA 2022.12.7 in a bit to see if I can replicate the issue.

gelse, 2022-12-21:
regarding logs - i set solax_modbus to debug in the config.

all i get in the logs are the following 3 lines:
```
2022-12-21 17:41:34.733 DEBUG (MainThread) [custom_components.solax_modbus] using pymodbus library 2.x
2022-12-21 17:41:34.749 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2022 12
2022-12-21 17:41:34.749 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
```

gelse, 2022-12-21:
My guess from reading the code:
somehow the dynamic loading of the plugins introduced in 469fb8082bf0344ae5aa110c00c056b1e09a1717 does not work.
perhaps it actually HAS something to do with the fact that i am running it via docker ? some weird permissions maybe? i try to tinker around a bit with that ...

wills106, 2022-12-21:
My main system is Core in Docker on Unraid.

The other system I tried was HAOS on PI4

gelse, 2022-12-21:
i added a few lines of debug messages, and indeed - it does not load any plugins:
```
2022-12-21 20:02:02.471 DEBUG (MainThread) [custom_components.solax_modbus.config_flow] Plugin path: custom_components/solax_modbus/plugin_*.py
2022-12-21 20:02:02.471 DEBUG (MainThread) [custom_components.solax_modbus.config_flow] Loaded plugins: []
```

the lines i added are in `config_flow.py`:
L59 (directly before the `PLUGINS=` ): `_LOGGER.debug(f"Plugin path: {PLUGIN_PATH}")`
L63 (directly after the `PLUGINS=` ): `_LOGGER.debug(f"Loaded plugins: {PLUGINS}")`

wills106, 2022-12-21:
When you do `ls -l` do you have read only for the different groups other than owner? Or has HACS removed read only some how?

[image]

I'm just going to upgrade HA to 2022.12.7

gelse, 2022-12-21:
it looks exactly that way, all of them are 644

gelse, 2022-12-21:
i just realized - PLUGIN_PATH is a relative path, i guess glob tries to find it from the CWD.
what if the CWD is not what we expect ?

gelse, 2022-12-21:
i guess that's it!
i added the following:
```
_LOGGER.debug(f"Current directory: {os.getcwd()}")
```
and got the following in the logs:
```
Current directory: /run/s6-rc:s6-rc-init:hjCanB/servicedirs/svc-homeassistant
```
i then attached to the container and checked what the directory is:
```
root@acbf3661c008:/run/s6-rc:s6-rc-init:hjCanB/servicedirs/svc-homeassistant# ls
down  event  notification-fd  run  supervise
```

which is NOT the root of the config.
my guess: there is some way to get the actual absolute path of the home-assistant configuration directory path - and that is missing here.

i guess the reason is because i am using the linuxserver.io docker image

wills106, 2022-12-21:
Is your config folder mapped to a drive on your NAS or however you are running this? Or is it all stored in your docker image?

gelse, 2022-12-21:
and so it is:

[image]

what i changed - and i admit it is a dirty hack that only works temporarily, but i think it should point you in the right direction:

in solax_modbus/const.py i changed the following lines:
```
DEFAULT_PLUGIN   = "/config/custom_components/solax_modbus/plugin_solax.py"                                                                                                                                                                                           PLUGIN_PATH      = "/config/custom_components/solax_modbus/plugin_*.py"     
```

which in MY case is the correct location.

i did open heart surgery here, so this bug is NOT closed, but i think it should be straightforward to implement now.

wills106, 2022-12-21:
This is how my setup is mapped
/config <-> /mnt/user/appdata/Home-Assistant-Core

gelse, 2022-12-21:
this has nothing to do with mapping of volumes of the docker container, as HA is running inside and it does not know of anything of the outside world.

this has to do with the special way linuxserver.io "simplifies" their images, as they always somehow change the running software for easier volume mappings.

the only way you can solve that (and i learned that the hard way in my job) - never rely on the working directory. NEVER.

infradom, 2022-12-21:
I think the absolute path might also work on most installations /config/custom...

infradom, 2022-12-21:
Just tested, and it seems to work with the absolute path on my HAOS
I had to reinstall the integration however ...

wills106, 2022-12-21:
If people don't remove the Integration prior to updating they are greeted with:

```
2022-12-21 20:43:25.462 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 372, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 95, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'custom_components.solax_modbus.plugin_'
```

If you try to reconfigure the Integration you loose the plugins.

Deleting and adding the Integration back again doesn't work.

It needs a full removal of the Integration from the Integration page and restarting HA and adding it back in again. For the path changes to take effect.

We'll have a look into a solution that doesn't cause all of the other users pain.

gelse, 2022-12-22:
i did not have to reinstall when i changed the path, altho i HAD to restart HA.
if i read the logfiles correctly, the integration, and thus the __init__.py, is loaded either on startup when it is configured, or during the first attempt to add a integration configuration.

which actually i just proved, because i made a few more changes and now it will work for every installation:
in const.py:
add this as first line:
```python
import pathlib
```

change this
```python
DEFAULT_PLUGIN   = "custom_components/solax_modbus/plugin_solax.py"
PLUGIN_PATH      = "custom_components/solax_modbus/plugin_*.py"
```
to this
```python
DEFAULT_PLUGIN   = f"{pathlib.Path(__file__).parent.absolute()}/plugin_solax.py"
PLUGIN_PATH      = f"{pathlib.Path(__file__).parent.absolute()}/plugin_*.py"
```

whoever wants it can take credit. ;-)

infradom, 2022-12-22:
Thanks, but I am still not sur this survives an upgrade, as the result path is probably stored persistently in the config.
I will examine in more detail to see if we need to adapt the string read from config also at startup

infradom, 2022-12-22:
And why not use absolute paths immediately? The problem of incompatibility lies in the config file entry that needs to be adapted ... I will check the config entry first so see which kind of path it contains in old and new

infradom, 2022-12-22:
checkedd: The config entry plugin contains a long path in the old version, probably relative,  in the new version: absolute
`'plugin': '/config/custom_components/solax_modbus/plugin_solax.py'`
So when starting up init.py reads this path and determines the name withj const.py getPluginName's function, which probable strips the name in a wrong way...

So we need carefull testing, and I am not sure adding this pathlib library solves the problem (as we can work as well with absolute paths). But there are more small changes needed is what I believe. I will make a proposal ..

gelse, 2022-12-22:
> Thanks, but I am still not sure this survives an upgrade, as the result path is probably stored persistently in the config.

thats a good concern, but my patch would at least fix the problem for people who have a custom installation directory and/or a different CWD. It of course helps nothing in terms of when you change the file location of the plugins in an upgrade.

> why not use absolute paths immediately? 

well my patch does that, but it calculates the absolute path depending on the location of the current file, which should be correct, wherever the HA config says that the configuration files reside. Unless, of course, in some future version of the integration, you change the location of the plugin files. Then you would have to change that code as well, but that's just obvious.

> as we can work as well with absolute paths

no we can't because there is no guarantee that the configuration directory is /config/

infradom, 2022-12-22:
OK thanks, all I wanted to say was that the pathlib patch is probably not sufficient if we want our users to be able to upgrade seamlessly

gelse, 2022-12-22:
yes, i agree with that.

if you want i can take a look into it - calculating the directory on startup and using that throughout the integration should be sufficient, but i would have to read into the details to find every place where the stored path is used.

infradom, 2022-12-22:
I am working on a solution... the const.py getPluginName() function is probably not needed anymore after another change last weeks

gelse, 2022-12-22:
actually i think it's the opposite.
if i read that code in __init__.py correctly, it maps any path from the config to a relative filename via getPluginName.

```python
    # ================== dynamically load desired plugin
    _LOGGER.debug(f"Ready to load plugin {config[CONF_PLUGIN]}")
    plugin_path = config[CONF_PLUGIN]
    if not plugin_path: _LOGGER.error(f"plugin path invalid, using default {DEFAULT_PLUGIN}; config dict: {config}")
    plugin_name = getPluginName(plugin_path)
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus') 
    if not plugin: _LOGGER.error(f"could not import plugin {plugin_name}")
    setPlugin(name, plugin)
    # ====================== end of dynamic load
```

i just wonder, why the plugin path is stored in the config in the first place, instead of just the plugin name.

gelse, 2022-12-22:
> ModuleNotFoundError: No module named 'custom_components.solax_modbus.plugin_'

oh my ...
that means, that getPluginName returns empty. would be interesting, what plugin_path is in line 94

infradom, 2022-12-22:
The modifications I am currently making make sure that the config entry for plugin just constains 'solax' or 'solis', not a path anymore. I do want to keep supporting the old path based entry for a while to avoid yet another breaking update.

infradom, 2022-12-22:
I used your contribution in a new PR, but I also made the plugin-config independent of paths, and added code to support old and new config entries.
Tested in an upgrade context and tested in a clean new install context
Further tests are welcome (see my PR)

wills106, 2022-12-22:
The PR closed this down automatically.

I'm just going to test the code change.

wills106, 2022-12-22:
@gelse does [2022.12.14b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.14b1) Work fine for you before I release the stable version?

I have tested it on my Docker install and it works smoothly as a code update.

I have also tested it on HAOS as much as I can without connecting to an actual Inverter.

gelse, 2022-12-23:
i was able to upgrade even without removing the old configuration, integration still works and the GUI also still works.

gelse, 2022-12-23:
btw you have a typo in the markdown of the index page:
> homsassistant-solax-modbus

wills106, 2022-12-23:
> i was able to upgrade (to 2022.12.14b1) even without removing the old configuration, integration still works and the GUI also still works.

Thanks, I'll close the issue down then.

closed 2022-12-23 by wills106

gelse, 2022-12-23:
but ... i just saw the release-notes, and i did not find the error message about the old style plugin name yet... i keep checking my logs

wills106, 2022-12-23:
It will be changed in the stable release.

gelse, 2022-12-23:
great, thanks a lot!

but i am feeling tempted to make a full code review like i would at work.... 
... must ... resist ... the ... urge ... ;-)

commit d2c0c9d2 (2022-12-22): make plugin method more path-independent

Thanks to @gelse
Also made it non-breaking change (config_flow)
fixes #269 and replaces PR #271

Tested: simple upgrade
Tested: remove, restart and reinstall