wills106/homeassistant-solax-modbus #277: Waveshare settings

willbiker, 2022-12-24:
Hi All!

I had this integration working through my wifi stick until recently when it stopped so I decided to switch to RS485.

I have a Solax X1 inverter and I have just setup a Waveshare RS485 to Ethernet B converter. I have installed Vircom and the device has been found. I have given the device an IP (which I can successfully ping) and have changed the Transfer protocol to Modbus_TCP which has changed the port to 502. 

Are there any other settings I need to configure? I am getting an "cannot find serial number" error.

Thanks in advance

[image]

willbiker, 2022-12-24:
Just to add to this, my device doesnt appear to be visible on the main Vircom landing screen...should it be?
[image]

wills106, 2022-12-24:
I am guessing you have a Gen4 Inverter? As the Gen3 has built in ethernet.
If so the Baud rate is set wrong, it should be 19200.
Also you have the IP Address and Gateway on two different Networks.

wills106, 2022-12-31:
@willbiker 
Did you get your Inverter connected ok?
Just wondering if I can close this issue down or not?

wills106, 2023-01-05:
Closing issue dow as no further response.
If you are still having issues feel free to reopen the issue.

closed 2023-01-05 by wills106