wills106/homeassistant-solax-modbus #278: [BUG]

littlecake, 2022-12-24:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:2022.12.13
2. HA Version:2022.12.8
3. Inverter Brand:SOLAX
4. Plugin Used:
5. First 6 digits of Serial:H34T10
6. Connection Method:Modbus

Logger: custom_components.solax_modbus
Source: /config/custom_components/solax_modbus/__init__.py:412 
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues)) 
First occurred: 22. prosince 2022 14:17:42 (486 occurrences) 
Last logged: 09:33:02

solax error reading input registers at device 1 position 0x0
solax error reading input registers at device 1 position 0x100
solax error reading holding registers at device 1 position 0x7d
solax error reading holding registers at device 1 position 0x0
solax error reading input registers at device 1 position 0x66
NoneType: None

Hello , plase its possible if some communication problem appears to hold the old values ?
If the problem appears values about PV jumps to 0.
I seen this problem probably after upgrade from 0.6.4

wills106, 2022-12-24:
How exactly are you connected to the Inverter?
RS485 - Ethernet Adaptor?
RS485 - USB Adaptor?

littlecake, 2022-12-24:
I am connected over ethernet

S pozdravem
Michal Koláček

wills106, 2022-12-24:
Ok I thought you might have been trying to connect with a Pocket WiFi.

When you were getting these errors was there enough Solar PV and Battery for the Inverter to be awake or was it in sleep (idle) mode?

littlecake, 2022-12-26:
It was when the sun was shining and the power plant was generating. It is possible that errors also occur due to hardware (but I can't detect it), I just thought that when a reading error occurs, it might be better not to write 0 values but to ignore them and wait for the next successful reading

infradom, 2022-12-26:
@wills106 : maybe we should remove the sleepmode=SLEEPMODE_ZERO attribute for the hybrids and keep it only for the MIC style devices. Currently the MIC devices dont even have this attribute for PV_ entities
It seems to be hard to keep everyone happy with these sleep states...

wills106, 2022-12-31:
Yes I agree to remove them. It seems only the X3 MIC needs these.

littlecake, 2023-01-04:
Thank you, the removing lines with sleepmode=SLEEPMODE_ZERO was really helpfull

wills106, 2023-01-04:
I'll push a proper update out shortly with some other corrections and improvements.

wills106, 2023-01-04:
This should be Implemented in [2023.01.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.1) is everything is fine are you able to close the Issue down.

closed 2023-01-04 by littlecake