wills106/homeassistant-solax-modbus #286: [BUG] HA unresponsive and slow when its polling

blackwhitebear8, 2023-01-03:
**Describe the bug**
SInce the last update our HA instance is slow unresponsive and breaks remot eaccess via nabu casa when its polling data from the inverter. When its dark outside so the inverter is off the HA instance becomes responsive again.

**Mandatory details**
Integration Version: 0latest
HA Version: Home Assistant latest
Inverter Brand: solis
Plugin Used: solis
First 6 digits of Serial: 110CA2
Connection Method: Serial (DSD TECH SH-U10 rs485 )

**Detailed Error Log**
there are no errors

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.

wills106, 2023-01-04:
When you say you are on the latest version of the Integration is that the stable release or the beta release?
What version were you on before when you didn't have the issue?

blackwhitebear8, 2023-01-04:
i am on stable asfar i know for the plugin
and the issue started after the plugin got updates for other inverters basicly so the last change after my last issue worked great

wills106, 2023-01-04:
If you look in manifest.py it will tell you which version you have installed.
Code got introduced to deal with sleeping Inverters partway through the development leading up to 2022.12.1 release.

blackwhitebear8, 2023-01-06:
HA version: Home Assistant 2022.12.8
Frontend-versie: 20221213.1 - latest

intergration version: 2022.12.13 but i see 2023.01.01 is out i will be trying that one now

wills106, 2023-01-06:
If you connect to your HA instance locally do you get these slow downs? Or is it only when you connect via Nabu Casa?

Are you still getting the Blocking Calls or did you amend your logging like I suggested [here](https://github.com/wills106/homeassistant-solax-modbus/issues/214#issuecomment-1328011729)?

wills106, 2023-01-06:
If you remove the RS485-USB dongle / the Integration does it go back to normal response?

blackwhitebear8, 2023-01-06:
Yes

blackwhitebear8, 2023-01-06:
and if the inverter is sleeping it is responsive aswell again

infradom, 2023-01-06:
What is your polling rate? Lower than 5 seconds may cause such problems

blackwhitebear8, 2023-01-06:
2 seconds but we never had problerms before for months it was after an update almost an month ago

infradom, 2023-01-06:
We did generalize the polling method and cannot guarantee that the system behaves exactly as before when it comes to timings. 
2 seconds may work, but I would not bet on it ... 
When the inverter is asleep, we slow down the polling by a factor 10. This explains the different behaviour

infradom, 2023-01-06:
Furthermore, each version supports more registers, so this also increases the load on the system.

wills106, 2023-01-06:
Have you tried going back from 2022.12.8 to see when the problem first started (Previous releases).
Polling at 2s on a 9600 baud is probably pushing the limits of the Inverter. These Solis don't seem the best of communications as it is and they don't follow Modbus protocols. We had to reduce the multi block reads down to 48 registers as they don't comply with the Modbus limit which is over double that.

Also when changing versions have you tried to remove the Integration from the Integration page, restarting HA and adding the Integration after upgrading. To make sure there aren't any bit's of code stuck in HA>

blackwhitebear8, 2023-01-06:
correction* i see that it was set to 1 second i put it at 2 seconds interval and things look already better.

blackwhitebear8, 2023-01-06:
it prop wasnt able to get everything within the 1 second becouse everything is normal again now

wills106, 2023-01-06:
If you get any further slowdowns / read errors I would try changing the poll to 5s before trying anything else.

blackwhitebear8, 2023-01-06:
well that kinda defeats the purpouse of readig it out via HA we wanted live xd the solis app was limited to 300seconds

infradom, 2023-01-06:
5 seconds is already a lot better than 300 ..
Dont push us too far ;-)

blackwhitebear8, 2023-01-06:
there was some update a month ago for wich i had to readd the inverter thats prop when i typed 1 sec instead of 2 :idk: well thanks for the help tho haha

wills106, 2023-01-06:
I'll close this issue down then if you are back to running as normal.

closed 2023-01-06 by wills106