wills106/homeassistant-solax-modbus #288: Integration is using deprecated `DEVICE_CLASS_*` constants

frenck, 2023-01-03:
**Describe the bug**

This custom integration uses deprecated `DEVICE_CLASS_*` constants in its codebase.

The `DEVICE_CLASS_*` constants have been deprecated and replaced in Home Assistant Core 2022.12 (over a year ago). I would highly suggest updating/migrating this integration to the new enums.

For example, for the device classes supported by the `sensor` platform, there is now a `SensorDeviceClass` enum. So if a sensor previously used the `DEVICE_CLASS_ENERGY` constant, it should now use `SensorDeviceClass.ENERGY`. Other platforms (like `binary_sensor`, and `number`) provide similar enumerations for their supported device classes.

The migration thus only consists of replacing constants with an enumeration member and is, therefore, very low impact and should be fairly straightforward.

If I can help resolve any questions regarding this change or migration, feel free to ask or respond to this issue. I'm happy to help!

Kindest regards,

../Frenck

**Mandatory details**
1. Integration Version: All
2. HA Version: since 2021.12 and newer
3. Inverter Brand: n/a
4. Plugin Used: n/a
5. First 6 digits of Serial: n/a
6. Connection Method: n/a

**Detailed Error Log**

n/a

**Additional context**

n/a

wills106, 2023-01-03:
Thanks, we have started to migrate over. For example `plugin_sofar.py` is using the new method:

```
    SofarModbusSensorEntityDescription(
        name = "Voltage L1",
        key = "voltage_l1",
        native_unit_of_measurement = ELECTRIC_POTENTIAL_VOLT,
        device_class = SensorDeviceClass.VOLTAGE,
        register = 0x48D,
        scale = 0.1,
        rounding = 1,
        allowedtypes = HYBRID | PV,
    ),
```

But the other Inverter brands / sub versions still need doing.

wills106, 2023-01-05:
At the moment for NumberEntities we don't declare any device classes, but we are using native_unit_of_measurement:

```
    SolaxModbusNumberEntityDescription( name = "Export Control User Limit",
        key = "export_control_user_limit", 
        register = 0x42,
        fmt = "i",
        native_min_value = 0,
        native_max_value = 2500,
        scale = 1,
        native_step = 100,
        native_unit_of_measurement = POWER_WATT,
        read_scale_exceptions=EXPORT_LIMIT_SCALE_EXCEPTIONS,
        allowedtypes = HYBRID | AC | GEN2 | GEN3 | GEN4,
        max_exceptions = MAX_EXPORT,
        icon="mdi:home-export-outline",
    ),
```

Would you recommend we also include the corresponding 'NumberDeviceClass.x' for each entity where we declare the 'native_unit_of_measurement' ?

frenck, 2023-01-05:
If the entity support setting a matching device class, yes, that would be best to use.

wills106, 2023-01-05:
Thanks for the feedback.

frenck, 2023-01-05:
No problem, any time 👍 

../Frenck

wills106, 2023-01-18:
I have migrated over to:

- NumberDeviceClass
- SensorDeviceClass
- SensorStateClass

Just need to work on converting POWER_WATT to UnitOfPower.WATT etc

wills106, 2023-01-21:
Closing down as I believe [2023.01.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.4) removes the last part of ```DEVICE_CLASS_*``` and I have migrated over to ```native_unit_of_measurement = UnitOf123.XYZ,``` for example ```POWER_WATT``` = ```UnitOfPower.WATT``` where possible.

closed 2023-01-21 by wills106

frenck, 2023-01-22:
Thanks for addressing this @wills106 👍 

../Frenck