wills106/homeassistant-solax-modbus #292: Sofar error reading holding registers at device

darkrain-nl, 2023-01-08:
I see this as well, but I don't think it is on every read cycle.
Would it work to have the integration set to like 30s for polling? Should that help to prevent it?

I see different positions it fails on, but that might be because my inverter is pure PV...
[image]

_Originally posted by @darkrain-nl in https://github.com/wills106/homeassistant-solax-modbus/issues/228#issuecomment-1344515467_

This seems to be gone away by changing polling to every 5 seconds, only the last poll of the day still gives the error, but that is because at that point the inverter is sleeping...
Just thought to share this.

wills106, 2023-01-11:
> This seems to be gone away by changing polling to every 5 seconds

What poll rate were you set to initially?

The default baud for Sofar and Solis Inverter's is quite low at 9600 and I think they struggle with faster poll rates.
The SolaX X3 MIC is the same 9600 baud and struggles at less that 5s poll rate.

In the Beta 2023.01.2b2 I changed the logging level to not report minor miss reads.
I'll be releasing the stable 2023.01.2 soonish anyway.

darkrain-nl, 2023-01-11:
> 

It was initially left default, so 15s, but 5s runs way better at least for me.
With 15s it happened multiple times a day, now only once a day at the end of the day.

I am curious how other systems behave...

darkrain-nl, 2023-01-13:
With the release no longer treating this as error I shall now close the issue.

Thanks again for the work on this integration.

closed 2023-01-13 by darkrain-nl

chfink0, 2023-01-19:
Hello there, I updated yesterday to the most recent 2023.01.3 in order to be able to change "Charge User Mode" but with no success.

When I set the Charger Use Mode to Timing Mode, it reverts back to the former Use Mode Self Use after a while. The log shows:
[image]

Any help is appreciated!

wills106, 2023-01-19:
I am guessing you are using a USB-RS485 adaptor? Do you also get this when reading sensors?

The blocking call's isn't an issue in itself (reading), it's listed as a [known issue](https://github.com/wills106/homeassistant-solax-modbus#known-issues) You can filter it out by setting your logger to ignore them.

At the moment writing to Sofar Inverters isn't working #228 I'm finding it a bit difficult to trouble shoot when I don't own a Sofar Inverter.

chfink0, 2023-01-19:
Yes, absolutely right, sorry, I actually read about that known issue when installing, but happened to forget in the meantime as it was working flawlessly. 

Ok than, understood, if I could be of assistance and you would like to pursue the writing issue, I'd be glad to help.

wills106, 2023-01-19:
Are you using plugin_sofar or plugin_sofar_old ?

chfink0, 2023-01-19:
I am using the plugin_sofar with a HYD 15KTL-3ph

wills106, 2023-01-19:
Thanks, just wanted to check which Inverter you were running. It will probably be best to follow along on that linked Issue I posted earlier.