wills106/homeassistant-solax-modbus #293: You cannot set "Export limit" more than 2500w

otava5, 2023-01-09:
Does anyone have an idea why I can't set more than 2500W in HA Export, while I can on the inverter, in the Solax app as well. thank you for the advice
[image]
[image]

wills106, 2023-01-11:
As a safety feature if the exact model isn't known it defaults to 2500w.

What Model and Gen Inverter do you have? I'll also need your first 4 or 6 digits of the serial to configure the Inverter.
You can see how it works by [looking here](https://github.com/wills106/homeassistant-solax-modbus/blob/755bad6b7bb17d8a66eb182e22fcc7f6c72a52bb/custom_components/solax_modbus/plugin_solax.py#L257)

Then I can update the Integration.

otava5, 2023-01-11:
It is a Solax X3 G4 10KW Inverter (Solax X3-Hybrid-10.0-M) with serial number H34C10. Thank you

wills106, 2023-01-11:
Ok, H34C is a new set of Serial Numbers, I have come across H34A, H34B & H34T so far.
What's the max charge / discharge Amp of the battery on this version? 25A or 30A? (Check on the Inverter / Cloud, not this Integration)

I'll update the Integration shortly.

otava5, 2023-01-11:
30A

wills106, 2023-01-11:
Just released [2023.01.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.2)

It's contains the MAX_CURRENTS & MAX_EXPORT for your Inverter series now and a few other changes.
If everything works ok, are you ok closing the issue down?

otava5, 2023-01-11:
Everything is fine. Thank you

closed 2023-01-11 by otava5