wills106/homeassistant-solax-modbus #295: Solax Timed Forced Charge Nort Working

J14111, 2023-01-11:
[image]

With these settings and setting the mode switch to Timed Charge/Discharge as you can see i nthe image it does not force charge, in face it seems to want to discharge at max all the time then in "Timed Charge/Discharge" unless Timed Discharge Current is set to 0 is this a bug or am I doing something wrong?

wills106, 2023-01-12:
I don't know how much testing has been done on the Solis work modes.
I don't own a Solis Inverter, so I can only go of the docs.

You might be best to start up a new discussion or use this one #188 ?

J14111, 2023-01-12:
Thanks I will start a new discussion, I am happy to do some testing for you if you have time.

Do you have a link to the Solis modbus information? I could not find any when I looked.

wills106, 2023-01-12:
I would expect there to be a newer document by now as that's quite old. https://community.home-assistant.io/t/solis-inverter-modbus-integration/292553/54 It's over 4 years old now.

J14111, 2023-01-12:
Thanks I will have a look

wills106, 2023-01-12:
There are some 2021 documents on the Internet. But they are for the older Inverter or at least not for Inverters in plugin_solis.py

The 2021 document refers to plugin_solis_old.py

i-jk, 2023-01-22:
Hi @J14111, when I first set up this module I tried using the Energy Storage Control setting but it also didn't work on my on my Solis RHI 5K 48ES 5G.

I think the issue is trying to use the slider controls for Timed Charge Start/End hour/minute controls in Home Assistant... These perhaps should not be used because [these time values need to be written together as a group](https://community.home-assistant.io/t/solis-inverter-modbus-integration/292553/5), so altering one value at a time from HA confuses the inverter which I think leads to the situation we found.

I have it working and here's how:

1. Go to the inverter's physical controls (you only need do this once), to `Advanced Settings` then `Storage Mode Select` and then turn it on.
2. In the next screen, go to `Self use` -> `Time of use` to get the screen of charge/discharge windows.
3. Ensure the `Optimal Income` (odd name but maybe its different on other firmware versions) is set "Stop" -- NOTE: this is equivalent to "Auto" in the Home Assistant Energy Storage Control select, which sets modbus register 43110 to value 33.
4. On the first charge block, set the start to 00:00 and the end to 23:59. This means the inverter will _allow_ charging in this period, but only when the `Optimal Income` is "Run".
5. Make sure you don't have any overlapping blocks of time in the other areas or it won't save, then escape and save and send.

Now in HA you should be able to set Energy Storage Control to 'Timed charge/discharge' (AKA register 43110 to value 35) and the inverter will begin charging at the current value set on its `Time of use` page, or whatever the BMS allows if lower. _(I haven't tested altering the Timed charge/discharge current values in HA to see if they will work properly -- i'll report back as soon as that is tested.)_

To recap: the schedules on the inverter simply _allow_ the charging/discharging in the windows defined, but only when `Optimal Income` is "Run" -- which is the same setting as Energy Storage Control in HA set to "Timed charge/discharge". So IMHO it's easier to set the inverter to allow charging any time (or any time you want to permit it), then control the actual charging yourself via HA UI or automations or whatever.

Hope this helps, and thank you so much @wills106 for all your hard work on this integration :)

wills106, 2023-01-23:
I can adapt the write_registers for register 43143 - 43150

Can you change "Timed Charge Current" & "Timed Discharge Current" or do they revert back as well?

Really wish when they write these documents they were clearer on which functions to use.

J14111, 2023-01-23:
@i-jk thanks Jim yes that makes perfect sense! as you change one slider it seems to throw the others out that would explain it thanks!

wills106, 2023-01-23:
Just released [2023.01.5b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b1)

Moved the following to write_method = WRITE_DATA_LOCAL,

Timed Charge Start
Timed Charge End
Timed Discharge Start
Timed Discharge End
Changing the times by them selves don't write to the Inverter, once you have finished setting all times, you press the Button "Update Charge/Discharge Times" The values will then be written to the Inverter in a write_registers block of multiple registers being written at once.

J14111, 2023-01-23:
Thanks so much @wills106 I will give it a try.

J14111, 2023-01-23:
I get this error when I try to move one of the time silders:
[image]

wills106, 2023-01-23:
Can you try removing the Integration from the Integration page, restarting HA and adding it back in.
If you name the Integration the same it wont affect your Dashboard.

I split the number and sensor entities in the background. So that might be causing bit of a problem in the background?

wills106, 2023-01-23:
Ok, I'll try something and push another update out.

wills106, 2023-01-23:
Could you please try [2023.01.5b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b2) they should all default to zero now until you change them to the time your want.

J14111, 2023-01-23:
Looks better and no errors, seems to work I set it to force charge and it worked, I will do some more testing but it seems you nailed it again, many thanks.

[image]

wills106, 2023-01-23:
Just been thinking about the way I have decoupled the number entities from the sensor entities.
You won't be able to see changes made from either the Inverter LCD or the Cloud app.

So in [2023.01.5b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b3) I have tried a different method.
Your numbers should now match what's stored in the Inverter.

Would you be able to perform a couple of tests for me?

Do they load up with what's stored in the Inverter (Assuming they are not all set as zero in the Inverter)
Could you change one of the number entities for a time slot. But don't press the Button "Update Charge/Discharge Times"
Does that number entity revert back to the stored value after a short time? (This will normally happen in relation to your poll interval)

If the number doesn't revert back, what happens if you change the value on the Inverter.
Does the Inverter update the number entity with a new value?
If it doesn't does restarting HA cause the number to then display the new stored value in the Inverter?

J14111, 2023-01-23:
will give it a go now

wills106, 2023-01-23:
Use [2023.01.5b4])https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b4) instead.

J14111, 2023-01-23:
it seems to load with what was set previously and if I change it without hitting update it then jumps back at the next poll

wills106, 2023-01-23:
I thought it would do that. I'll have another think about how to do an initial check of the sensor, without the sensor then interfering with the number. Before you get chance to press the button.

J14111, 2023-01-23:
yeah that could be an issue with low polling times, still at least its working better than before, thanks

J14111, 2023-01-23:
would sending/updating all fields when one is changed work/be possible?

wills106, 2023-01-23:
In [2023.01.5b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b2) you could just change one of the time slots, press the button and it would all get written to the Inverter. (Assuming the rest are set in HA how you want them)

HA would know what it's changed the times to. I would expect them to survive a restart of HA.
HA / The integration wouldn't know if the Inverter has been changed through either the LCD or Cloud app though. I don't know how important that is?

A function could be built on top of [2023.01.5b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b4) to only update the time sensors if the value has actually changed and not just because it's been repolled.

I would have to think how I could achieve that.

J14111, 2023-01-23:
There is something janky going on with these sliders, no matter what you set them to they go to 7A unless you set them to 0 then they jump to 100A 

[image]

[image]

wills106, 2023-01-23:
Those with other registers around them might need writing in a single block as well?
Shame there isn't a newer document about. This one is nearly 5 years old now. It mentions write_register and write_registers but doesn't give any indication when to use one or the other.

wills106, 2023-01-23:
In [2023.01.5b6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b6) I have reverted back to 2023.01.5b2 so can you program the times without them reverting back before you can press the button.

i-jk, 2023-01-23:
These sliders perhaps need a scaling factor fix? Current for these 3 appears to be scaled by 10x, so 1A = 10 when written to a register - for example 43118 for discharge current.

(FYI I'm basing this on working examples by various people, e.g. https://youtu.be/kgU-JFLQIFo)

wills106, 2023-01-23:
Ahh looks like I have the sensor as 0.1 resolution but the write is 1.

I'll correct it.

wills106, 2023-01-23:
Hopefully I have it the correct way round in [2023.01.5b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b7)

i-jk, 2023-01-23:
Thanks! 

Ok so I tried b7... Good news is the timed charge and discharge sliders seem to work great. I confirmed looking at the inverter's screen and they're perfect.

Bad news is the values for the timed windows have all gone to 0 (hours and minutes). However in the inverter my settings are still here. I left the integration page and went back and they are still at 0. I never pressed the update button, so this is a UI/local state issue.

wills106, 2023-01-23:
I have moved the timed values back to [2023.01.5b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b2) behaviour.

> In [2023.01.5b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b2) you could just change one of the time slots, press the button and it would all get written to the [...]

The times don't get the information from the Inverter. But you can manually change them and program the Inverter.
The problem with [2023.01.5b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b4) is that the values would go back to the Inverter values on the next poll cycle. So chances are you wouldn't be able to set all 8 times and hit the "Update Charge/Discharge Times" button before they reverted back.

wills106, 2023-01-23:
At the moment you will need to set all 8 timers how you want. Press the "Update Charge/Discharge Times" button.

Then if you wanted to amend you would change them and press the button again.
You would have to see if the current times in the Integration survive a restart or not.

If you change the times on the Inver / Cloud etc the Integration won't update the values displayed in the Integration though.

Ideal the read only values ie "RO Timed Charge Start Hours" will in the future be able to update "Timed Charge Start Hours".
But only if the value has actually changed and not just because it's been repolled again.

wills106, 2023-01-29:
@J14111 @i-jk Is this now working how it should be?

J14111, 2023-01-29:
Yes seems to be working fine for me, now I just need to get on an economy 7 tariff, thanks so much :)

wills106, 2023-05-06:
I have reopened this issue to try and couple the Times and the RO Times back together, so you have feedback on what the times are set to now before changing them.

wills106, 2023-05-06:
@J14111 

I have just released [2023.05.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b2)
I have replaced all of the RO Times with just Times and implemented "prevent_update = True," to all Times.

Can you let me know if you can now change the various times and they stick giving you enough time to press one of the "Update Charge/Discharge Times" buttons. As previously they would keep reverting back to the stored time on each poll cycle.

wills106, 2023-05-12:
@J14111 @i-jk 
Just wondering if either of you have tried the newer Beta with the Times coupled back up?

wawyed, 2023-05-16:
Hi, I was able to update the discharge times with Beta 3.  Although I can still see it resetting back to the previous value if I'm not fast enough to press Update. My polling is set to 10seconds

wills106, 2023-05-22:
@J14111 @i-jk @wawyed 
Just released [2023.05.2b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b5)
Could someone test out the Charge/Discharge programming. You should now have 120s to press the "Update Charge/Discharge Times" button without the values revert back on each polling cycle.

J14111, 2023-05-22:
Just tested and it doesnt revert for me, just stays at what I set even if I dont press the update button

J14111, 2023-05-22:
Nope on latest beta, it changes back in a few seconds for me, (I have my polling set to 5 sec)

wawyed, 2023-05-23:
Same here it reset after 10 seconds ( my polling time)

wills106, 2023-05-24:
Are you able to test [2023.05.2b6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b6)
We are trying to understand how 'entity_registry_enabled_default' affects the prevent update code introduce to prevent the numbers from altering back on each polling cycle.

wills106, 2023-05-27:
New release [2023.05.2b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2b7) moves the  'prevent_update = True,' to number entities allowing the sensors to be disabled by default. So you don't need to remove the Integration or enable the disabled sensors.

@J14111 @i-jk @wawyed Could you let me know if it works as intended?

wills106, 2023-05-29:
Closing down as [2023.05.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.2) addresses this issue.

closed 2023-05-29 by wills106