wills106/homeassistant-solax-modbus #307: Suggestions for improvements on Solis inverters

i-jk, 2023-01-25:
Hello, I'm using `plugin_solis` and have noticed a minor bug, plus have some suggestions to make...

**Bug:** The Inverter Storage Control sensor doesn't show the name of the mode name, just the number. Perhaps this is due to a type mismatch (should be `fmt = "i"`)? Note this same issue also affects the mode select widget, which is blank (or rather shows the placeholder text).

**Suggestion: fix charge and discharge current limitation of 70A**
The `native_max_value` of several number sliders for the charging and discharging current are limited to 70A. This is obviously dependent on the specification of the inverter, e.g. my 5kW 48V hybrid should be 100A. I wonder if it's possible to either:
1. establish the inverter specs from the serial and set these accordingly
2. use the appropriate `BMS` value for charge max and discharge max, when available, or fallback to safe 70A.

**Suggestion: number fields instead of sliders?**
The sliders are rather fiddly to use tbh... A simple number field might be better, is this possible?

If you agree with the above, perhaps we can give the hour/minute timed charge and discharge values the same treatment? Can we group the hour and minute fields together to display like a proper time? This is a nice to have, obviously :slightly_smiling_face:  

Thanks again for such an awesome integration! :raised_hands:

wills106, 2023-01-26:
> Inverter Storage Control sensor doesn't show the name of the mode name, just the number.

Looks like there is a potential mismatch between the rear sensor and the select entitie.

> charging and discharging current are limited to 70A

This was only set to 70A as that's what's in the Modbus docs.
If we get enough data I can use an approach like used on the SolaX Inverters:

https://github.com/wills106/homeassistant-solax-modbus/blob/4264bc6e0c84e96b9232c1611f0dcf7e0d064aeb/custom_components/solax_modbus/plugin_solax.py#L242-L244

> TBH the sliders are rather fiddly to use on larger current numbers like 100A

HA itself automatically configures sliders or number boxes on it's own.
I think there is an option to override this automatic allocation, I can have a look into it.

> Can we group the hour and minute fields together to display like a proper time?

There isn't actually support for the equivalent to an input.date_time for Integrations.
You should be able to create a time helper and have the Integration use that I believe. But I haven't found an example on how to do this yet.

i-jk, 2023-01-26:
>> Inverter Storage Control sensor doesn't show the name of the mode name, just the number.

> Looks like there is a potential mismatch between the rear sensor and the select entitie.

These work though... when I set the value and the name appears correctly until the next update from the inverter, at which point the name is replaced with the number. It's a display issue only, for both the select and the sensor that point at the same underlying value.

wills106, 2023-01-26:
The select for "Energy Storage Control Switch" has the number - name conversion.

https://github.com/wills106/homeassistant-solax-modbus/blob/4264bc6e0c84e96b9232c1611f0dcf7e0d064aeb/custom_components/solax_modbus/plugin_solis.py#L323-L338

But I forgot to include them on the Sensor

https://github.com/wills106/homeassistant-solax-modbus/blob/4264bc6e0c84e96b9232c1611f0dcf7e0d064aeb/custom_components/solax_modbus/plugin_solis.py#L737-L743

The sensor should look a bit like this SolaX one, but with Solis info

https://github.com/wills106/homeassistant-solax-modbus/blob/4264bc6e0c84e96b9232c1611f0dcf7e0d064aeb/custom_components/solax_modbus/plugin_solax.py#L3573-L3586

i-jk, 2023-01-26:
FYI the spec sheet for the RHI 48V hybrids is here: https://www.ginlong.com/Uploads/file/Solis_datasheet_RHI-(3-6)K-48ES-5G_AUS.pdf, and the variants look like this:

* Max charge/discharge for 3K, 3.6K models = 62.5 A
* Max for 4.6K, 5K and 6K models = 100 A

wills106, 2023-01-27:
[2023.01.5b11](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b11)

Includes the fix for "Energy Storage Control Switch" and the introduction of Charge / Discharge Values based on Inverter Model.

i-jk, 2023-01-30:
I tried beta 17... after a restart, the sliders are now limited to the value I had (25A at the time) and the widget HTML confirms this limit. But I see in your commit you set the `native_max_value` to 20? 60 would be a better safe minimum... 

either way the `max_exceptions = MAX_CURRENTS` bit doesn't seem to work presently.

wills106, 2023-01-30:
Your Inverter should only default to 20A if it isn't declared here:

https://github.com/wills106/homeassistant-solax-modbus/blob/0b3a6d8461b53cceadca05b1279556c59df463c1/custom_components/solax_modbus/plugin_solis.py#L138-L144

I have set the default to be lower than the known lowest Amps that an Inverter supports.
That way if an unknown Inverter connects they can't accidently set their Inverter higher than Intended, without properly adding in support for that model.

Is your Inverter stuck at 20A?

wills106, 2023-01-30:
Doesn't matter, I see the typo now...

wills106, 2023-01-30:
Should be corrected in [2023.01.5b19](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b19) so your value is 100A

i-jk, 2023-01-30:
Another (major) issue with b11+: the storage controls do not work in automations any more since the fix for the name/number conversion...

[image]

Whereas there were three named values before, there is just 'available' and 'unknown', and the value 33 was there when I loaded. This was all working with b7.

wills106, 2023-01-30:
Will it accept:
Auto Mode

Also does 2023.01.5b19 fix the battery charge limits?

i-jk, 2023-01-30:
Where has the drop down with all the available values gone? None of my automatons that depends on this cake with any more... I'll rolling back to b7

i-jk, 2023-01-30:
And yes, b19 fixed the current limits, thank you 🙂

wills106, 2023-01-30:
Does it help if you change the following:

https://github.com/wills106/homeassistant-solax-modbus/blob/6138e69c7c7b94c30608ac3999946ac8794ca114/custom_components/solax_modbus/plugin_solis.py#L807-L820

to:
```
    SolisModbusSensorEntityDescription(
        name = "Energy Storage Control Switch",
        key = "energy_storage_control_switch",
        register = 43110,
        #register_type = REG_INPUT,
        scale = {  
                33: "Auto Mode", 
                35: "Timed Charge/Discharge", 
                37: "Off-Grid Mode", 
                41: "Battery Awaken", 
                43: "Battery Awaken + Timed Charge/Discharge", 
                },
        allowedtypes = HYBRID,
    ),
```

wills106, 2023-01-30:
Also when you went from b7 to b11+ did you remove the Integration, restart HA and add it back in again?
When there are major changes and you just upgrade, HA doesn't cleanly make all the changes. You need to remove the Integration, restart and add it back in again. That way it loads all the code fresh.

i-jk, 2023-01-30:
I'm going to try to avoid removing the integration as a lot of statistics, automations and dashboards depend on the entities it provides. A restart works fine for all other HACS integrations I've used, including this one.

I'll test the above later and let you know.

i-jk, 2023-01-30:
Ok so it looks like I'll need to replace/update every mention of that entity in scripts and automations... the "state" condition doesn't work any more, but the "device" version appears to.

[image]

I'll continue testing and let you know.

i-jk, 2023-01-30:
Ok so, apart from the state condition no longer working, everything seems ok now. It had been working at beta 7 but not since the testing tonight.

As far as I can see there are two "Energy Storage Control Switch" entities, one read-only sensor, and the select. The select always works and it's available values appear in the drop down, but the sensor now [only shows the numeric value of the register as screenshot above](https://github.com/wills106/homeassistant-solax-modbus/issues/307#issuecomment-1409295288) .

When displayed in an entity card, the sensor shows the correct text.

i-jk, 2023-01-30:
(btw the above is with all the change you suggested in https://github.com/wills106/homeassistant-solax-modbus/issues/307#issuecomment-1409321051)

wills106, 2023-01-31:
The read only sensor is normally disabled by default. It's just used as feedback for the select.

> I'm going to try to avoid removing the integration as a lot of statistics, automations and dashboards depend on the entities it provides.

Unfortunately sometimes there is a big enough change that results in you having to remove the Integration, restarting HA and adding it back in such as going from 0.6.4 - [2022.12.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2022.12.1)
As long as you keep the Integration name the same Dashboards will pick up again, as long as there isn't any name changes of sensors.

wawyed, 2023-02-05:
Happy to create a different thread but has anyone notince that the house load seems to round to 0 when the load is around 100W? Is this an issue with the logging stick/inverter or some rounding on the tool itself?

wawyed, 2023-02-05:
Also the grid_import_today seems to instead of starting at 0kwh starts at 4. something kwh everyday?

i-jk, 2023-02-19:
@wawyed I'm pretty sure your issues are due to old/or of the box firmware. I have them too but I have yet to contact support to schedule an update - especially as it would need my to revert back to the solarman-based

wawyed, 2023-03-19:
I don't know if anyone mentioned but `BMS Battery Current` has the wrong scale. It should be `0.1` not `0.01`. It should be the same as `BMS Battery Charge Limit`

wills106, 2023-03-21:
Just checked a newer version of the Modbus doc and it is indeed 0.1 and not 0.01 as per the older version of the Modbus doc.

wawyed, 2023-04-05:
This might be a weird one but I'm trying to calculate Energy Consumption and Energy Generated Consumed via entity templates by combining values as follows:

```
      - name: "Consumption today"
        device_class: "energy"
        unit_of_measurement: "kWh"
        icon: mdi:home
        state_class: total_increasing
        state: >
          {% set generationToday = states('sensor.solis_inverter_power_generation_today') | float(0) %}
          {% set importToday = states('sensor.solis_inverter_grid_import_today') | float(0) %}
          {% set exportToday = states('sensor.solis_inverter_grid_export_today') | float(0) %}

          {{ generationToday + importToday - exportToday }}
      - name: "Energy generated consumed today"
        device_class: "energy"
        unit_of_measurement: "kWh"
        icon: mdi:home
        state_class: total_increasing
        state: >
          {% set generationToday = states('sensor.solis_inverter_power_generation_today') | float(0) %}
          {% set exportToday = states('sensor.solis_inverter_grid_export_today') | float(0) %}

          {{  (generationToday - exportToday) | round(2) }}
      - name: "Energy cost saved today"
        unit_of_measurement: "GBP"
        device_class: "monetary"
        icon: "mdi:cash"
        state_class: total_increasing
        state: >
          {% set generationConsumedToday = states('sensor.energy_generated_consumed_today') | float(0) %}
          {% set importPrice = states('sensor.smart_meter_electricity_import_unit_rate') | float(0) %}

          {{  (generationConsumedToday * importPrice) | round(2) }}
```

Unfortunately I've had an instance where the reset of the daily values have deferred which led me to incorrect calculations. 

solis_inverter_power_generation_today reset to 0 at 23:59:39

where as both:

solis_inverter_grid_import_today
solis_inverter_grid_export_today

reset at 23:59:29

I'm not sure how this can be solved from my side so I was wondering if you had any ideas on how to tackle this. 

Thank you very much.