wills106/homeassistant-solax-modbus #309: [BUG] Solax Gen2 doesn't work anymore since 2023.01.X updates

blb4github, 2023-01-29:
**Describe the bug**
In december the integration worked still for me but with one of the updates the connection to the inverter didn't work anymore. I tried all versions currently available via HACS but non of them is working.

**Mandatory details**
1. Integration Version: 2023.01.1 t/m 2023.0104
2. HA Version: 2022.12.8 and 2023.1.7
3. Inverter Brand: Solax
4. Plugin Used: Solax Interver Modbus
5. First 6 digits of Serial: 43F947
6. Connection Method: ethernet

**Detailed Error Log**
2023-01-29 00:49:45.029 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 00:50:32.248 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 00:50:32.286 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 59, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
  File "/config/custom_components/solax_modbus/__init__.py", line 317, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(172.31.0.21:502)]
2023-01-29 00:50:32.300 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-01-29 00:50:37.310 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 00:50:37.311 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 59, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
  File "/config/custom_components/solax_modbus/__init__.py", line 317, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(172.31.0.21:502)]
2023-01-29 00:50:37.312 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-01-29 00:50:37.313 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-01-29 00:50:37.317 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2023-01-29 00:51:12.622 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 00:51:32.657 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 00:53:49.848 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 00:56:19.877 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 00:58:50.062 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 01:01:20.109 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 01:03:07.490 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 01:03:07.492 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x0
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 59, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
  File "/config/custom_components/solax_modbus/__init__.py", line 317, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(172.31.0.21:502)]
2023-01-29 01:03:07.494 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-01-29 01:03:12.501 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 01:03:12.507 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 59, in _read_serialnr
    inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
  File "/config/custom_components/solax_modbus/__init__.py", line 317, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
[... 13 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/core.py", line 1792, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 958, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 116, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 79, in async_press
    self._hub.write_register(unit=self._modbus_addr, address=self._register, payload=self._command)
  File "/config/custom_components/solax_modbus/__init__.py", line 342, in write_register
    res = self._lowlevel_write_register(unit, address, payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 333, in _lowlevel_write_register
    return self._client.write_register(address, payload[0], **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 92, in write_register
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(172.31.0.21:502)]
2023-01-29 01:03:37.826 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out
2023-01-29 01:05:52.838 ERROR (MainThread) [pymodbus.client.sync] Connection to (172.31.0.21, 502) failed: timed out

wills106, 2023-01-29:
As it's not able to read your Serial number it appears there are no communications with your Inverter.
Has the ip address changed? Has the Inverter gone into a fault mode? Might be worth power cycling your Inverter if neither of them.

blb4github, 2023-01-29:
Thanks, at last I did get it working! IP address was the same but I guess powering down the inverter completely did do the trick.

closed 2023-01-29 by blb4github