wills106/homeassistant-solax-modbus #310: [Feature Request] Sync inverter time - Solis

J14111, 2023-01-29:
Hi,

Would it be possible to have an option in the plugin to automatically sync the inverter time daily or weekly with the homeassistant clock at say 2am? this would resolve daylight savings issues, it looks like its possible to set the time via registers on the Solis:

[image]

wills106, 2023-01-29:
It won't be automatic.
But I can get you a sync button like we have on SolaX.
Then run an automation against it every so often.

wills106, 2023-01-29:
Just released [2023.01.5b14](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b14) it should give you the functionality you want.

J14111, 2023-01-29:
Thanks will give it a go, does it check the time and only sync if needed, just so I know and can avoid excessive register writes?

wills106, 2023-01-29:
It's purely manual.
Just a sensor you can enable if you wish to see the RTC and a button to Sync it to HA.

So you would need an automation to press the button one a week or once per month etc

wills106, 2023-01-29:
Updating the RTC should work. But the sensor for RTC won't as reading it wrong.
I'll do an update.

wills106, 2023-01-29:
Corrected in [2023.01.5b15](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b15)

J14111, 2023-01-29:
Thats brillaint thanks!

wills106, 2023-01-29:
Once you have tested it, are you ok closing the Issue down?

J14111, 2023-01-29:
Ok will do, I have tried it and no error but I will have to go in the loft to check on the screen, I know I was about 5 mins fast before, I will close now and reopen if needed, thanks again.

closed 2023-01-29 by J14111

J14111, 2023-01-29:
It works perfectly, I set the time wrong on the inveter then synced and it updated fine, I have set an automation to run it every sunday at 2:02am so that should cover keeping it in sync and DST changes but I guess I will get duplicate wonky data when the clocks go back but thats not much of an issue. Thanks so much!

wills106, 2023-01-29:
Glad it's helped.