wills106/homeassistant-solax-modbus #311: [BUG] Suddenly cant read data from Solax

zejdajan, 2023-01-29:
Suddenly cant ready any values from Solax inverter, everything worked more than month nicely.
I didnt change the version or settings.
Tried also rollback to older plugin version.
Inverted is working, solax cloud is receiving date, also I cant read data over LAN with CURL.
Maybe Solax pushed some firmware update that broke it?
When re-adding integration I am not getting any errors.

**Mandatory details**
1. Integration Version:2023.01.4
2. HA Version:2023.1.7
3. Inverter Brand:Solax X3 gen4
4. Plugin Used:Solax Interver Modbus
5. First 6 digits of Serial:SVA7TY
6. Connection Method: TCP/IP with wifi dongle

Output of CURL realtime data: 
{"sn":"SVA7TXXXX","ver":"3.003.02","type":14,"Data":[2375,2380,2376,11,10,10,75,71,56,202,2558,2368,3,2,84,54,4998,4999,4998,2,0,0,0,0,0,0,0,0,0,0,0,0,0,1,61906,65535,0,0,0,23400,0,0,2329,65534,65490,1,31,3832,256,1576,7433,5889,100,0,15,0,0,0,0,0,0,0,0,0,0,0,0,0,3237,0,2,152,0,0,1005,0,1214,0,0,0,3400,0,0,1,0,0,130,0,31640,3,0,0,680,0,0,0,0,0,0,0,0,0,1,32,1,17,37,256,2628,1800,80,350,108,81,32,32,10,1115,256,9252,9252,0,0,0,0,3240,3229,43182,3,21302,14389,18757,13111,16688,12611,13106,21302,14389,18757,13111,16688,12611,13106,21302,14389,18758,12855,16696,12867,14642,0,0,0,0,0,0,0,0,0,0,0,0,0,0,515,3330,516,1025,0,23400,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"Information":[8.000,14,"H34A08I4214149",8,1.27,0.00,1.26,1.09,0.00,1]}

Assistant log: 

2023-01-29 09:12:36.673 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration nibe which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 09:12:36.674 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 09:12:36.675 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration xiaomi_cloud_map_extractor which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 09:12:36.676 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration cz_energy_spot_prices which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 09:12:36.676 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 09:12:36.677 WARNING (SyncWorker_1) [homeassistant.loader] We found a custom integration tesla_custom which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-29 09:12:43.706 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-01-29 09:12:43.808 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-01-29 09:12:43.809 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-01-29 09:12:43.809 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2023-01-29 09:12:46.912 WARNING (MainThread) [homeassistant.config_entries] Config entry 'HP Color LaserJet MFP M277dw (A3F6C6)' for ipp integration not ready yet: Invalid response from API: Error occurred while communicating with IPP server.; Retrying in background
2023-01-29 09:12:48.236 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for chilli-box @ 10.10.30.151: Error connecting to ('10.10.30.151', 6053): [Errno 113] Connect call failed ('10.10.30.151', 6053)
2023-01-29 09:12:54.258 WARNING (MainThread) [homeassistant.helpers.entity] Update of camera.xiaomi_cloud_map_extractor is taking over 10 seconds
2023-01-29 09:13:09.264 WARNING (MainThread) [homeassistant.components.switch] Setup of switch platform nibe is taking over 10 seconds.
2023-01-29 09:13:09.266 WARNING (MainThread) [homeassistant.components.sensor] Setup of sensor platform nibe is taking over 10 seconds.
2023-01-29 09:13:09.284 WARNING (MainThread) [homeassistant.components.water_heater] Setup of water_heater platform nibe is taking over 10 seconds.
2023-01-29 09:13:09.284 WARNING (MainThread) [homeassistant.components.climate] Setup of climate platform nibe is taking over 10 seconds.
2023-01-29 09:13:32.189 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if (t>0) %} {{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}")
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 423, in async_render
render_result = _render_with_context(self.template, compiled, **kwargs)
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1950, in _render_with_context
return template.render(**kwargs)
File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 1301, in render
self.environment.handle_exception()
File "/usr/local/lib/python3.10/site-packages/jinja2/environment.py", line 936, in handle_exception
raise rewrite_traceback_stack(source=source)
File "<template>", line 1, in top-level template code
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1729, in forgiving_int_filter
raise_no_default("int", value)
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1411, in raise_no_default
raise ValueError(
ValueError: Template error: int got invalid input 'unknown' when rendering template '{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if (t>0) %} {{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}' but no default was specified
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 540, in async_render_to_info
render_info._result = self.async_render(variables, strict=strict, **kwargs)
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 425, in async_render
raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: int got invalid input 'unknown' when rendering template '{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if (t>0) %} {{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}' but no default was specified
2023-01-29 09:13:32.193 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unknown' when rendering template '{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if (t>0) %} {{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}' but no default was specified') while processing template 'Template("{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if (t>0) %} {{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}")' for attribute '_attr_native_value' in entity 'sensor.battery_in_2'
2023-01-29 09:13:32.196 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if t<=0 %}{{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}")
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 423, in async_render
[... cut ...]
2023-01-29 09:13:32.198 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('ValueError: Template error: int got invalid input 'unknown' when rendering template '{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if t<=0 %}{{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}' but no default was specified') while processing template 'Template("{% set t = states('sensor.solax_battery_power_charge').split()[0] | replace(",", "") | int %} {% if t<=0 %}{{states('sensor.solax_battery_power_charge')}} {% else %} 0 {% endif %}")' for attribute '_attr_native_value' in entity 'sensor.battery_out'

wills106, 2023-01-29:
You are getting an error in your log. When sharing your log it's easier to type SolaX into the search box to filter anything not related to SolaX out.

> 2023-01-29 09:12:43.706 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-01-29 09:12:43.808 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-01-29 09:12:43.809 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-01-29 09:12:43.809 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

You don't have any communications with the Inverter.
I am guessing that Serial you have provided is your Pocket WiFi and not the Inverter.

I [don't officially](https://github.com/wills106/homeassistant-solax-modbus#support-modbus-over-rs485--tcp) support the Pocket WiFI as the old ones fall under three groups:
- Work fine
- Intermittently work
- Don't work at all

According to SolaX the only Pocket WiFi that's advertised as working with Modbus is the Pocket WiFi 3.0 with Firmware V3.004.03, but I don't have a means to validate this as I don't own a Pocket WiFi

closed 2023-01-29 by wills106

zejdajan, 2023-01-29:
Hi, thanks for reply. 
Serial of inverter is: H34A0XXXX.
It was working very good on a first attemp since december. 
It looks like it connects to the inverted but there is a data reading error, however the data are coming-tested with CURL with IP of inverted.

wills106, 2023-01-29:
Unfortunately the older Pocket WiFi seem to be unreliable at working.

msgre, 2023-04-13:
Hi guys,
just FYI: I have Pocket Wifi 2.0 with SN starting with "SXE.." letters. I am able to communicate with it through REST API (`curl -d "optType=ReadRealTimeData&pwd=<PWD>" -X POST http://<IP>`) and also `homeassistant-solax-modbus` plugin. My Pocket Wifi is identified as device with hostname `espressif` by router.

Anyway, yesterday I found that friend of mine have same Pocket Wifi 2.0 device connected to Solax inverter with SN starting with "SWR.." letters, identified as `MXCHIP`. We need to send slightly different curl to get data (`curl -H "'X-Forwarded-For':'5.8.8.8'" -s -w '\n' -X POST "http://<IP>" --data 'optType=ReadRealTimeData&pwd=admin'`), but we have no luck with `homeassistant-solax-modbus` plugin. There was bunch of error logs about unsucesfull connections to dongle, reading S/N and several others.