wills106/homeassistant-solax-modbus #313: [Feature Request]

J14111, 2023-01-29:
Thanks for all the amazing work, I have another request when you have the time, would it be possible to have this as a slider:

[image]

I have had a look but I cant figure out which register it is, it would help my stop my hot water diverter kicking in when I dont want it to as I can disable any export.

Thanks.

wills106, 2023-01-29:
What's the Min - Max it will allow you to set it to?
I guess it could be 43051 or 43081?

J14111, 2023-01-29:
Min is defo 0w and I think max is 10000w on my inverter but I can double check the max tomorrrow if needed

wills106, 2023-01-29:
Looks like in the docs they can be + / - so might not be them?
I can add the sensors in to see if they alter when you change them on the LCD / Cloud portal?

J14111, 2023-01-29:
ok sounds like a good plan

wills106, 2023-01-29:
If you want to experiment with just the sensor, you could easily add the following into plugin_solis
As long is it's within the 'SolisModbusSensorEntityDescription' section it doesn't matter on the order, but I prefer to keep them in register order. As I find it easier to keep track of them.

To try a different register just change the register number.

```
    SolisModbusSensorEntityDescription(
        name = "Backflow Power",
        key = "backflow_power",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        register = 43051,
        allowedtypes = HYBRID,
    ),
```

J14111, 2023-01-29:
ok will have a look thanks

wills106, 2023-01-29:
The register you are after should be in section 5.4 as they are the read / write ones.
Registers in section 5.2 are read only and need the addition of ```register_type = REG_INPUT,```

Once you look at the code in any plugin_x it becomes quite easy to workout what's going on.
Most if it is just defining what a register is and how it needs to be displayed (Icon, unit type etc)

wills106, 2023-01-29:
If you want to add in an aditional sensor using a different register make sure the following are unique to other sensors.
```
name = "Backflow Power",
key = "backflow_power",
```
I always try to keep the name and key the same, bar case and replacing spaces with _

So:
```
name = "New Sensor",
key = "new_sensor",
```

J14111, 2023-01-29:
I got lucky its the second one I tried!, I think its 

SolisModbusSensorEntityDescription(
        name = "Backflow Power1",
        key = "backflow_power1",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        register = 43052,
        allowedtypes = HYBRID,

[image]

J14111, 2023-01-29:
Its the only thing I think that is set to 10000w on my inveter but I can double check by changing it tomorrow so we know 100%

wills106, 2023-01-29:
Yea changing it on the Inverter will let you know for sure.
As the docs indicate that register should be a %
But docs are not always accurate!

J14111, 2023-01-29:
Yeah they were translated from Chinese I believe so a mistake could have been made, I will check it tomorrow morning, thanks!

J14111, 2023-01-30:
thats not the right setting, it has a different value, the ranges on my inverter are 0-9900

J14111, 2023-01-30:
I have tried quite a few and cant find it, I think its a newish feature in more recent firmware, mine also has the option for on/off that is not in that screenshot from the manual, I might try speaking to Solis support to see if they will tell me.

wills106, 2023-01-30:
Would be nice to have a document that isn't 5 years old!
They have most on their site bar the one we need. Last time I looked there was just a dead link.

J14111, 2023-01-30:
Yeah, I have logged a ticket I will also ask them if they have a document.

J14111, 2023-01-30:
If they wont provide the info is there an easy way to get a dump of all the registers, I should be able to spot it fairly easily?

J14111, 2023-01-30:
I managed to figure it out buy dimping the degisters, changing it and then dumping and comparing, it seems to be 43073, however it doesnt store the plain value:

0w is stored as 0
100w is stored as 256
9900w is stored as 25344

so it seems each w is multiplied by 2.56 then stored? that may make more sense to you?

But im 99% sure this is the correct register.

wills106, 2023-01-30:
How are you extracting these registers?
Are you using a tool like Gmodbus?

J14111, 2023-01-30:
yeah, a note for the slider it only lets you change it in 100w increments

wills106, 2023-01-30:
I think the endian is swapped. Is there an option in Gmodbus to swap the Endian?
Not used it in a while. Like if it's reading Little endian can you select Big.

J14111, 2023-01-30:
Im getting really confussed now, 3 different programs, 3 different readings lol

[image]

HA:

[image]

wills106, 2023-01-30:
So is Backflow Power 1 the on off switch? 16 = on and 0 = off? Or could it be something else?
Backflow Power 2 matches your LCD but only representing first two digits. So 1 = 100w, 5 = 500w, 50 = 5000w

J14111, 2023-01-30:
Ah yes I think you are right, I set it to off and 100w on the LCD and I got:

[image]

J14111, 2023-01-30:
Ok sounds good

wills106, 2023-01-30:
Out of curiosity do you have a second or more blocks where you get set Charge / Discharge Periods on your Inverter in addition to the one in the Integration?

- Timed Charge Start
- Timed Charge End
- Timed Discharge Start
- Timed Discharge End

J14111, 2023-01-30:
Yes I think there are 3 or possibly 4 blocks, this was included in recentish firmware I beleive from what I have read.

wills106, 2023-01-30:
Are you on that firmware or not?

J14111, 2023-01-30:
yeah I have the 3 or 4 blocks I believe I am on the latest

wills106, 2023-01-30:
Can you try [2023.01.5b17](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b17)

I'll disable sensor "Backflow Power" in the future as it isn't needed going forward. But for now can you see if it's scaled correct.
If it's 9900 on the Inverter, does "Backflow Power" read 9900?

If that's correct you can then try the Number associated to it.
You also have the select switch for it.

i-jk, 2023-01-30:
Vaguely offtopic but... @wills106 have you seen the modbus doc from the Solis site? https://ginlongsolis.freshdesk.com/support/solutions/articles/36000340158-modbus-communication-for-solis-inverters

Not sure if it's newer than what you presently are using as reference?

wills106, 2023-01-30:
Unfortunately, despite being a newer doc. That is actually for plugin_solis_old

J14111, 2023-01-30:
Yes it read 9900 in HA but I cant change it, I get this error if I move the slider:

[image]

wills106, 2023-01-30:
I let the copied code as a float by mistake instead of an int [2023.01.5b18](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.01.5b18)

J14111, 2023-01-31:
Thanks mine is the 3.6 model I moved to b19 but still seeing:

[image]

wills106, 2023-01-31:
If you change:
https://github.com/wills106/homeassistant-solax-modbus/blob/6138e69c7c7b94c30608ac3999946ac8794ca114/custom_components/solax_modbus/plugin_solis.py#L276-L289
to:
```
    SolisModbusNumberEntityDescription(
        name = "Backflow Power",
        key = "backflow_power",
        register = 43074,
        fmt = "i",
        native_min_value = 0,
        native_max_value = 9900,
        native_step = 100,
        scale = 100,
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        allowedtypes = HYBRID,
        entity_category = EntityCategory.CONFIG,
    ),
```
does it work?

J14111, 2023-01-31:
No error now with that change, I will have to confirm it changes the setting on the inverter a bit later

wills106, 2023-01-31:
Could you also confirm some sensors for me.
"Battery SOC" = The live battery SOC, it's not you minimum SOC you want it to discharge to.

On your Inverter do you have two more Battery SOC you can set.

Solis Backup Mode SOC & Solis Overdischarge SOC or something like that?

J14111, 2023-01-31:
Yes battery SOC is live and correct

There are two settings one is the SOC it stops using the battery at and the second is the force charge setting where the inverter will force a charge if it hits that level.

I will check the others but I think this is still current:

[image]

i-jk, 2023-01-31:
Thanks for sharing @J14111. Be aware that timed charge seems to be it's own mode even if it's in the self-use menu on the inverter.

I'd say that the 'Energy Storage Control' in HA is good but not quite a match.

* `Auto` in HA is "self use mode" and works fine
* `Timed charge/discharge` is correct and works fine
* `Off-grid mode` is also correct
* `battery awaken` and `battery awaken + Timed charge/discharge` I'm not sure about... I've tried them but can't really say what happened tbh

Which leaves "feed in priority" and "EPS mode unaccounted for... This could be the two "battery awaken" starts above?

i-jk, 2023-01-31:
@wills106 there are several battery charge values in the system - some from BMS (which limits over current), some for the depth of charge for the batteries overall, some for limiting charge in certain modes. Tbh it'll take a while to discover what does what.

I'm not using those since I now use automations to control battery charge and discharge current via Home Assistant via basic charge and discharge current. I base this on state of charge and temperature to protect the batteries success they're in a cold garage.

I also automate the switch from auto (self use) to off-grid when SOC < 20% -- and back, depending on time of day and solar energy input. Finally I use the octopus integration to force charge overnight on the cheapest electricity and am about to include Solcast data to attempt to charge only what is needed based on the likely solar harvest next day.

This integration currently allows all this cool stuff to work seamlessly so, again, thank you 🍻

J14111, 2023-01-31:
> I'd say that the 'Energy Storage Control' in HA is good but not quite a match.
> 
>     * `Auto` in HA is "self use mode" and works fine
> 
>     * `Timed charge/discharge` is correct and works fine
> 
>     * `Off-grid mode` is also correct
> 
>     * `battery awaken` and `battery awaken + Timed charge/discharge` I'm not sure about... I've tried them but can't really say what happened tbh
> 
> 
> Which leaves "feed in priority" and "EPS mode unaccounted for... This could be the two "battery awaken" modes above?

Exactly the same for me

J14111, 2023-01-31:
@wills106 someone on facebook pointed me to this, they are regsiter addresses extracted from the android solis app, may be of some use to you? 
[assets.zip](https://github.com/wills106/homeassistant-solax-modbus/files/10545769/assets.zip)

wills106, 2023-01-31:
I can take a look. But there is a possibility they don't match up to the Modbus registers.
The SolaX ones don't, some are close.

wills106, 2023-01-31:
Just had a look at they appear to be actual registers.
I'll try an take a proper look later today.

wills106, 2023-02-01:
> No error now with that change, I will have to confirm it changes the setting on the inverter a bit later

Did you manage to confirm 'Backflow' is working correctly and writing to the Inverter as intended?

J14111, 2023-02-01:
Just checked both backflow settings work perfectly thanks

closed 2023-02-01 by J14111

J14111, 2023-02-02:
@wills106 I just signed up to Octopus and used your referal code to say thanks again!

wills106, 2023-02-02:
Thanks, just seen it.
I think they are a very good company and I like their innovative tariffs they offer.

Out of curiosity were you able to fully complete the signup online or do you still have to phone up to move over?

J14111, 2023-02-02:
I was getting ready to phone them but its all online again now which is great, I need to get a smart meter even though I dont want one so I can get paid for export, I think im going to be generating a lot more than I can store/use in the summer so makes sense

J14111, 2023-02-02:
Yeah I will probably go on one of their smart tariffs over the winter to fill the batteries up, not much point now

Baker0052, 2023-04-07:
On my Solis-S5-EH1P46K-L when i set Backflow Power to 300W it sets it to 3.000.000W

Baker0052, 2023-04-07:
Ok changing the "Scale =0.01" to "Scale = 100" seems to work.
Had to restart HA.