wills106/homeassistant-solax-modbus #318: [BUG] Scale Today's Solar Energy wrong; must be 0.1 instead of 0.001

blb4github, 2023-02-01:
**Describe the bug**
Since today I have PV panels connected to my SolaX SK-SU3000E so I'm looking into the related details in the integration. 
It looks like the scale for Today's Solar Energy is wrong; it stayed the whole (grey) day at 0,00 kWh where Today's Yield did go up to 0,6 kWh. I did change the scale in plugin_solax.py from 0.001 to 0.1 and now I do get 0.4 which is equal to what I see in the Solax app when I connect local to the inverter.

**Mandatory details**
1. Integration Version:
2. HA Version:
3. Inverter Brand: Solax
4. Plugin Used:
5. First 6 digits of Serial:
6. Connection Method:

**Additional context**
plugin_solax.py:
    SolaXModbusSensorEntityDescription(
        name = "Today's Solar Energy",
        key = "today_yield",
        native_unit_of_measurement = UnitOfEnergy.KILO_WATT_HOUR,
        icon = "mdi:solar-power",
        device_class = SensorDeviceClass.ENERGY,
        state_class = SensorStateClass.TOTAL_INCREASING,
        register = 0x50,
        register_type = REG_INPUT,
        unit = REGISTER_S32,
        scale = 0.1,      # BLB 20230201 changed from 0.001
        rounding = 2,
        allowedtypes = GEN2,

wills106, 2023-02-08:
Just released [2023.02.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.02.6) should fix the rounding error.

closed 2023-02-08 by wills106

blb4github, 2023-02-09:
It's working for me as well, thanks!