wills106/homeassistant-solax-modbus #320: [BUG]

sagpsch, 2023-02-03:
**Describe the bug**
A clear and concise description of what the bug is.
Hello, just started with HA on a Pi4 and tried to get data from my Q.HOME⁺ ESS HYB-G3 (inverter, Matebox incl. Smartmeter, 2 batteries) into HA. Already bought Waveshare RS485 TO Ethernet Converter (B) and connected it via RS485 with inverter and LAN (same  IP-range as Inverter and Pi4). Integration shows 1 device with 250 entitites. I retrieve some data from inverter (e.g. Serial number, Backup Discharge Min SOC, Discharger Start Time 1) which cleary shows setup data from my inverter. But I do not receive any live data from inverter for PV1, PV2, Grid usage etc. but can see data in Webinterface https://www.portal-q-cells.us/

**Mandatory details**
1. Integration Version: 2023.02.4
2. HA Version: Core 2023.2.1 / Operating System 9.5
3. Inverter Brand: Q-cells Q.HOME⁺ ESS HYB-G3
4. Plugin Used:
5. First 6 digits of Serial: H34B10
6. Connection Method: ModBus TCP using WaveShare RS485 TO ETH (B)

**Detailed Error Log**
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/const.py:213
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 15:09:48 (25 occurrences)
Last logged: 16:09:38

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 410, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 490, in read_modbus_registers_all
    self.data[descr.key] = descr.value_function(0, descr, self.data )
  File "/config/custom_components/solax_modbus/const.py", line 213, in value_function_grid_export
    val = datadict["measured_power"]
KeyError: 'measured_power'
**Additional context**

I tried to follow strictly the installation instruction on Wiki for Waveshare and installed integration when I was sure that inverter is online and active.

wills106, 2023-03-30:
Did you manage to resolve your issue?

wills106, 2023-04-05:
Closing issue down as no response.

If you have further issues please raise a new issue or reopen this issue.

closed 2023-04-05 by wills106