wills106/homeassistant-solax-modbus #321: [BUG] Upgrade to 2023.02.4 breaks installation on X3-Hybrid-G1/G2

mickemartinsson, 2023-02-04:
**Describe the bug**
Updating from > 2023.02.01 breaks installation and Solax custom integration does not start

**Mandatory details**
1. Integration Version: > 2023.02.01
2. HA Version:  Home Assistant 2022.12.8 Frontend-version: 20221213.1 - latest 
3. Inverter Brand: Solax
4. Plugin Used: 
5. First 6 digits of Serial: H3UE10H
6. Connection Method: Cable

**Detailed Error Log**
Logger: homeassistant.setup
Source: setup.py:210
First occurred: 08:11:12 (1 occurrences)
Last logged: 08:11:12

Setup failed for custom integration solax_modbus: Unable to import component: cannot import name 'UnitOfApparentPower' from 'homeassistant.const' (/usr/src/homeassistant/homeassistant/const.py)

wills106, 2023-02-04:
I think 'UnitOfApparentPower' is missing from HA 2022.12.8 https://github.com/home-assistant/core/pull/83451
When I was working on https://github.com/wills106/homeassistant-solax-modbus/issues/288 I would have been on HA 2023.01.x
I will probably need to set this Integration to use a minimum HA version of 2023.01.1

mickemartinsson, 2023-02-04:
Yes, I can confirm that first update to HA >2023.01.X and then update to integration > 2023.02.xx works fine. 
So, yes, no update without a minimum HA version of 2023.01.1

wills106, 2023-02-04:
I'll update the readme.

wills106, 2023-02-05:
Updated readme.

closed 2023-02-05 by wills106