wills106/homeassistant-solax-modbus #322: [BUG] sofarsolar: cannot find serial number, even not for other Inverter

shairportMulti, 2023-02-04:
Error on start up

1. Integration Version: 2023.02.4
2. HA Version: 2023.2.1
3. Inverter Brand: SofarSolar HYD 15KTL-3PH
4. Plugin Used: 
5. First 6 digits of Serial: SP1ES1
6. Connection Method: serial

2023-02-04 15:24:06.943 WARNING (MainThread) [custom_components.solax_modbus.plugin_sofar] sofarsolar: reading serial number from address 0x445 failed; other address may succeed
2023-02-04 15:24:06.983 ERROR (MainThread) [custom_components.solax_modbus.plugin_sofar] sofarsolar: cannot find serial number, even not for other Inverter
2023-02-04 15:24:06.988 ERROR (MainThread) [custom_components.solax_modbus.plugin_sofar] unrecognized sofarsolar inverter type - serial number : unknown

wills106, 2023-02-04:
Have you made sure you are not connected to the USB - RS485 device while still being connected via your software "solaranzeige.de" or with the windows software solarmonitor you mention?

Inverters starting with SP1ES1 are known to work.

shairportMulti, 2023-02-04:
a second master (esp) is connected the inverter, but the esp and solaranzeige.de and the windows software are working in parallel

shairportMulti, 2023-02-05:
Everything is working fine as soon as your integration is connected as only one.

closed 2023-02-05 by shairportMulti