wills106/homeassistant-solax-modbus #323: Battery power total rounding to 2 decimal points

juraurban, 2023-02-04:
Hi, 
Sofar HYD 10 KTL 
can you please fix for me rounding of Battery power total to 2 decimal points ... to the same as Sofar Battery Power 1
[image]

wills106, 2023-02-04:
Register 667 is only rounded to 0.1 according to the documentation:

https://github.com/wills106/homeassistant-solax-modbus/blob/f4254203f16b1f78d5a9f13210e7c937feb178dd/custom_components/solax_modbus/plugin_sofar.py#L1837-L1848

[image]

juraurban, 2023-02-04:
Thanks for comment, I am closing this one.

closed 2023-02-04 by juraurban