wills106/homeassistant-solax-modbus #324: solax-modbus not running on homeassistant 2023.2

redzoro01, 2023-02-04:
I upgraded my installtion to homeassistant 2023.2.1
Unfortunately solax-modbus installation can´t be loaded anymore.
Error message is:
2023-02-03 14:06:33.317 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solax_modbus: No module named 'pymodbus.client.sync'

I am back to HA 2023.1.7 and all is working fine again.

Maybe this gives a hint:
[HA core is getting pymodbus 3.1.1]

https://github.com/WillCodeForCats/solaredge-modbus-multi/pull/238

Would be great, if this integration would work with new homeassistant versions.

Many thanks for your great work

wills106, 2023-02-04:
I can't replicate this issue.

I'm on HA 2023.2.1, what version of the integration are you on?
0.5.x or 0.6.x by any chance?

wills106, 2023-02-04:
Same issue? https://github.com/wills106/homeassistant-solax-modbus/discussions/319

redzoro01, 2023-02-04:
Yes seems to be the same issue

<html><body>

domain | "solax_modbus"
-- | --
name | "SolaX Inverter Modbus"
version | "0.5.6"
documentation | "https://github.com/wills106/homsassistant-solax-modbus"
issue_tracker | "https://github.com/wills106/homsassistant-solax-modbus/issues"
requirements |  
0 | "pymodbus>=2.5.3"
dependencies | []
codeowners |  
0 | "@wills106"
config_flow | true
iot_class | "local_polling"

</body>
</html>

redzoro01, 2023-02-04:
I will install newer version tomorrow and do the update to HA 2023.2 again.

wills106, 2023-02-04:
As you are on quite an old version of the Integration, remove the Integration from the Integration page.
Update the Integration, restart HA.
Then reconfigure the Integration on the Integration page.

If you name the Integration the same, you dashboards should be the same.
There was a major config change between 0.6.x and 2022.12.x

redzoro01, 2023-02-05:
After upgrading the integration and upgrade to HA 2023.2 all works fine.

Many thanks

closed 2023-02-05 by redzoro01