wills106/homeassistant-solax-modbus #325: Feature request

blackwhitebear8, 2023-02-05:
Can we get an error sensor that if the in my case solis inverter gets into an error state we can read that out via home assistant.

wills106, 2023-02-06:
It's already on the Todo List V2 https://github.com/wills106/homeassistant-solax-modbus/discussions/200#discussioncomment-4242410

wills106, 2023-04-26:
I still have a bug in the current beta for Solis without a battery / not enough battery to last the night.
I have a workaround to push out as a Beta. If that works I might push out a proper release and look at fixing it properly later.

wills106, 2023-05-01:
I'm closing this issue as [2023.05.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.1) is released.
Which includes what you requested.

closed 2023-05-01 by wills106