wills106/homeassistant-solax-modbus #326: [BUG] 'Please do not use unit=, convert to slave=' when using the sync RTC button

darkrain-nl, 2023-02-06:
**Describe the bug**
When I press the button to sync RTC the following error shows in the log
```
Please do not use unit=, convert to slave=.
```

**Mandatory details**
1. Integration Version: 2023.02.1
2. HA Version: 2023.2.2
3. Inverter Brand: Sofar
4. Plugin Used: plugin_sofar
5. First 6 digits of Serial: SS2ES1
6. Connection Method: RS485 ethernet 

**Detailed Error Log**
```
2023-02-06 12:00:00.005 INFO (MainThread) [homeassistant.components.automation.sync_rtx_for_sofar] Sync RTX for sofar: Running automation actions
2023-02-06 12:00:00.005 INFO (MainThread) [homeassistant.components.automation.sync_rtx_for_sofar] Sync RTX for sofar: Executing step device automation
2023-02-06 12:00:00.005 ERROR (MainThread) [pymodbus.client.mixin] Please do not use unit=, convert to slave=.
```
**Additional context**
Seems the error might pymodbus related.

[image]

wills106, 2023-02-06:
I am guessing a change was introduced with HA 2023.02.x but I can't actually find the change just yet.
I'll try an update later today.

darkrain-nl, 2023-02-06:
Yes there have been two dependency updates, first to pymodbus 4.1.0 and then to 4.1.1 The last PR was this one https://github.com/home-assistant/core/pull/86513 

I have searched in the actual pymodbus code, but with my skills that didn't lead to the solution...

wills106, 2023-02-07:
Just released [2023.02.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.02.5) should be fixed now.

darkrain-nl, 2023-02-07:
Yes, it is fixed, pressed the button twice and no error anymore.

closed 2023-02-07 by darkrain-nl