wills106/homeassistant-solax-modbus #329: [BUG] Missing history data from time of day  - get stuck, every 24h in the same time, need restart

filipi992062, 2023-02-12:
Describe the bug
   At a certain time, the integration gets stuck and visually stops recording the values that are plotted in the energy graph, after restarting the entire device, everything will be added at the end of the hour as the energy panel plots data in the graphs, no data will be lost. I run the system on a Raspberry pi4, 8GB, I use an RS485 to USB converter for RS485 data.
   This happens with iron regularity every approx. 24 hours, always between 10-11 am. It doesn't matter if I restart in the morning or in the afternoon or in the evening, the time when these errors appear is constant

**Mandatory details**
1. Integration Version: 2023.02.7b1
2. HA Version: 2023.02.3 /Operating syste 9.5
3. Inverter Brand: SOLAX X3 G4 10kW
4. Plugin Used: SolaX Inverter Modbus
5. First 6 digits of Serial:  H34B10
6. Connection Method: RS485 TO USB

**Detailed Error Log**

Logger: homeassistant.util.async_
Source: util/async_.py:182
First occurred: 10:49:47 (89483 occurrences)
Last logged: 11:27:12

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 318: return self._client.read_holding_registers(address, count, **kwargs)
Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 324: return self._client.read_input_registers(address, count, **kwargs)

[image]

[image]

[image]

Goblaski, 2023-02-14:
I assume you use the RS485 USB module. I see a lot of errors with the blocking calls. Might flood the log.

What you could try is to add the following lines to your configuration file:

```
logger:
  default: warning
  logs:
    homeassistant.util.async_: error
```

This suppresses the warnings that comes from the blocking call due to the RS485 USB module (as well as other integrations that might give this warning). Due note that this is a potential solution (but not a full fix). The original author recommends RS485 to Ethernet modules for this (what I am also going for in the future as a fix).

wills106, 2023-02-14:
The blocking calls aren't actually the issue.
The logger can be setup to ignore them like @Goblaski mentions. Although ideally that needs addressing properly, although it's difficult when I don't connect over RS485 at all.

> What is the polling frequency you have? Most inverters don't like it if you go below 5 seconds.

That is good advice! @filipi992062 you are not trying to run with a poll rate of 1 second are you?

If it's the same time each day do you have any cron jobs running at that time?

wills106, 2023-03-31:
@filipi992062 are you still having issues? Were you trying to poll too fast and since reduced the frequency?

wills106, 2023-04-05:
Closing issue down as no response.

If you have further issues please raise a new issue or reopen this issue.

closed 2023-04-05 by wills106

filipi992062, 2023-04-16:
I reduced the frequency, once 15sec, problem solved