wills106/homeassistant-solax-modbus #334: Two solax inverter integration

klio-klio, 2023-02-19:
Hello, I tried to integrate my second inverter today. Had to find out that the display is wrong. The Entetis were integrated from old into new. Battery is displayed in new even though none is connected. PV power is the same. Do I have to pay attention to anything? The old inverter is the X3 8000 hybrid old generation still with network interface. The new one is the X3 Hybrid 15.0 D Modbus connection via COM connection.
[image]

wills106, 2023-02-19:
Have you given them both unique names? Ie SolaX West, SolaX SouthEast etc

klio-klio, 2023-02-19:
Of course... name. Thank you. Mea culpa!!!

wills106, 2023-02-20:
Is everything running ok now? If so could you close the issue down?

closed 2023-02-21 by klio-klio