wills106/homeassistant-solax-modbus #336: [BUG]

CristiXP, 2023-02-20:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:2023.02.6
2. HA Version: 2023.2.5
3. Inverter Brand:X3-Hybrid-G4-15.0
4. Plugin Used:Solax Inverter Modbus
5. First 6 digits of Serial:H34A15
6. Connection Method: modbus

Solax house entity sensor.solax_house_load not sure it is displaying the correct value. Can you please check? it seems that the value is increasing once the PV power increases also. Is the version being determined from PV power - injection power or is something read directly from the inverter? 

This is how mine looks daily when production and read from modbus

[image]

Versus the one I read from Solax API

[image]

wills106, 2023-03-26:
I have just released [2023.03.2b9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b9) which reintroduces an alternative "House Load Alt" this is using the old calculation.

Let me know if it seems better and doesn't mirror the PC Curve as much.

CristiXP, 2023-03-26:
Updated, still no change in values.
Solax House is value from API, Solax House Load from moddbus.

[image]

CristiXP, 2023-03-26:
Seems to be a value next to the correct one. I will keep you posted.[image]