wills106/homeassistant-solax-modbus #340: [BUG] Sensors producing garbage - integration has to be restarted

alienatedsec, 2023-02-24:
**Describe the bug**
The integration seems to crash occasionally and produces some garbage for the sensors until reloaded.

**Mandatory details**
1. Integration Version: 
`2023.02.7b3` and it happened with earlier versions too
2. HA Version: 
`2023.2.5`
3. Inverter Brand:
[RHI-6K-48ES-5G](https://www.ginlong.com/rhi_inverter1/1952.html)
4. Plugin Used:
`solis`
5. First 6 digits of Serial:
`603105`
6. Connection Method:
`Waveshare RS485 to ETH (B) PoE` version. I described it [here](https://github.com/fboundy/ha_solis_modbus/issues/4#issuecomment-1436737744) in more detail

[image]

**Detailed Error Log**
From around 08:35 this morning until I restarted the integration at around 09:10, the sensor reported incredible battery usage (around 500kWh) and it affects the energy dashboard. It also reports 0 for other sensors - seems like the integration is not reading data from the gateway.
[image]

[image]

That wasn't the first time, and it happens occasionally with no real trigger. Firstly, I thought it was the problem with my unusual setup or even the gateway crashing and producing those results. However, noticed that when the integration is restarted, it gets back to normal. Also, my datalogger relies on the gateway, and it produced correct figures for other integration and the cloud between 08:35 - 09:10. It is worth noting that if the integration is to poll data `less` often (every 15 instead of 5 seconds) it crashes `more` often. It hasn't crashed on me for the last 4 days since I changed it to poll data every 5 seconds.

[image]

I attach the log files for the period it occurred.
[Problem.txt](https://github.com/wills106/homeassistant-solax-modbus/files/10823087/Problem.txt)

alienatedsec, 2023-02-26:
Another issue from this morning is the same scenario. I got it back to normal after reloading the integration.

Logs attached
[Problem day 2.txt](https://github.com/wills106/homeassistant-solax-modbus/files/10833448/Problem.day.2.txt)

wills106, 2023-03-05:
If you reduce block_size to say 40, does that improve the situation?
https://github.com/wills106/homeassistant-solax-modbus/blob/a61f25e52f3768d50087ff86c18b2c78fe1a7ae3/custom_components/solax_modbus/plugin_solis.py#L1670

alienatedsec, 2023-03-05:
> If you reduce block_size to say 40, does that improve the situation?

I made a change, so it will take some time to confirm its working correctly - I also attached the logs in relation to the following warning towards the end of the log file

```
2023-03-05 12:48:33.508 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2023-03-05 12:48:33.512 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2023-03-05 12:48:33.515 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2023-03-05 12:48:33.532 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None
2023-03-05 12:48:45.805 WARNING (MainThread) [homeassistant.components.button] Setup of button platform solax_modbus is taking over 10 seconds.
2023-03-05 12:48:45.806 WARNING (MainThread) [homeassistant.components.select] Setup of select platform solax_modbus is taking over 10 seconds.
2023-03-05 12:48:45.809 WARNING (MainThread) [homeassistant.components.number] Setup of number platform solax_modbus is taking over 10 seconds.
```
[block_size40 logs.txt](https://github.com/wills106/homeassistant-solax-modbus/files/10891869/block_size40.logs.txt)

alienatedsec, 2023-03-06:
@wills106 is there any limitation in reducing the block size to 40? I cross-checked with other plugins and Solax has 100. Also, I looked into the documentation and the max recommended seems to be the same (100 bytes).

[image]

https://www.photovoltaikforum.com/core/attachment/202315-modbus-solis-rhi-5k-pdf/

So far, limiting to 40 is working ok with no issues, but I will need another week to feel more comfortable with it.

wills106, 2023-03-10:
> pyModbus `3.1.1` unable to handle blocks bigger than 40 bytes

It won't be that pyModbus is unable to handle blocks larger than 40 registers (80 bytes)
SolaX and Sofar return 100 registers and can return more if needed.

It will be either the Inverter is struggling to provide so many registers or on occasion is returning garbage and pyModbus doesn't know what to do with it / #346 is another way to deal with it if 40 blocks doesn't help.

> Will you update the block_size to 40 in the next release?

I can do that yes.

alienatedsec, 2023-03-10:
@wills106 Thanks for the clarity - I appreciate your time on this project, especially when you have no access to equipment.
Once you release a change, you can close this issue. It works fine with 40 registers (80 Bytes).

J14111, 2023-03-14:
Yes:

[image]

alienatedsec, 2023-03-16:
@wills106 I just had the below this morning

Unable to download a full log at the moment

‘’’
This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:270 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 09:09:33 (85764 occurrences) 
Last logged: 12:25:33

SolisLAN: read failed at 0x81ff: meter_apparent_power_total
SolisLAN: read failed at 0x8201: meter_power_factor
SolisLAN: read failed at 0x8202: meter_grid_frequency
SolisLAN: read failed at 0x8203: meter_grid_import_total
SolisLAN: read failed at 0x8205: meter_grid_export_total
‘’’

alienatedsec, 2023-03-16:
This time it produced some weird things - 8% battery. On previous occasions, it was 0%.

[image]

Still, Producing garbage for other sensors. 

[image]

alienatedsec, 2023-03-16:
So far it worked best with `2023.02.7b3` and the `block_size=40` - I am going to revert to that version after downloading the logs.

alienatedsec, 2023-03-16:
@wills106 I am currently on `2023.03.2b3` with the `block_size=40` - I can see the `block_size=40` would be the only change for the `2023.03.2b4`

alienatedsec, 2023-03-16:
> Your new timed registers? Which you added here [alienatedsec/homeassistant-solax-modbus@main...solis_timeslots](https://github.com/alienatedsec/homeassistant-solax-modbus/compare/main...solis_timeslots)

yes

wills106, 2023-03-16:
I'll work on adding them in and making sure ```ignore_readerror = True,``` works for this additional sensor block.

wills106, 2023-03-16:
I have just released [2023.03.2b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b5) with the new time blocks and the missing ```ignore_readerror = True,```

alienatedsec, 2023-03-24:
@wills106 I am currently on [2023.03.2b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b7) since its was released. I can see no 'garbage', but the following in the logs. Those are since I restarted yesterday.

```
2023-03-23 18:40:21.086 INFO (SyncWorker_1) [homeassistant.loader] Loaded solax from homeassistant.components.solax
2023-03-23 20:36:04.071 INFO (SyncWorker_0) [homeassistant.loader] Loaded solax_modbus from custom_components.solax_modbus
2023-03-23 20:36:04.071 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-03-24 00:25:48.739 INFO (MainThread) [custom_components.solax_modbus] read_error : SolisLAN cannot read input registers at device 1 position 0x81e3
```

Even the `read_error`, it works well and doesn't show any issues, but wondering if that is related to anything in the integration.

alienatedsec, 2023-03-30:
Seems to be stable since [2023.03.2b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b7)

Thanks for now. I hope you can progress with those two discussions at some point and I can contribute to #365 if needed:
- https://github.com/wills106/homeassistant-solax-modbus/discussions/188#discussioncomment-5416341
- https://github.com/wills106/homeassistant-solax-modbus/discussions/365#discussion-5013711

alienatedsec, 2023-03-30:
@wills106 I can have a go

alienatedsec, 2023-03-31:
@wills106 I hope that works for you
https://github.com/alienatedsec/homeassistant-solax-modbus/wiki/Installation-Notes#solis-inverters

wills106, 2023-03-31:
That seems good.
Is there a way to do a PR and push it back to me?

alienatedsec, 2023-03-31:
Unfortunately not - edit your page and paste the below

```
## SolaX Inverters
**Only the SolaX Gen2 & SolaX Gen3 Hybrids have built in Ethernet.**
- **(The Qcells Q.VOLT HYB-G3-3P is a SolaX Gen4 despite the G3 naming)**

All other inverters do not have a fixed Ethernet or WiFi interface. 
A pocketLAN or pocketWiFi adapter can be used to connect the inverter to the cloud.
The inverter however does not provide the Modbus protocol on its pocket LAN or WiFi interface. 
Modbus communication must be done through the serial RS485 COM port.

This port is marked COM on the inverter, some user manuals by mistake claim that this is an Ethernet port.
DO NOT CONNECT THIS COM PORT TO AN ETHERNET SWITCH - your Ethernet switchport will die if you do !!!
The RS485 interface uses only the pin4 (blue wire) and pin5 (white-blue). Polarity matters, so make sure pin4 (blue wire in typical cables) is connected to the A lead of your adapter and pin5 (white-blue wire) to the B lead of your adapter.
RS485 cables can be fairly long, so 25 meter or more is possible. You can use the blue pair of a standard cat5 or cat6 cable.

To connect the inverter to your HA, you will need either:
- A RS485 to Ethernet adapter (cost 25 €) - This is the preferred way
- A RS485 to USB adapter (cost 2 €) - This is a simple way, but HA may report thousands of warnings "blocking call inside the event loop"

Make sure you verify your inverter's baud rate with the [Wiki Page](https://github.com/wills106/homeassistant-solax-modbus/wiki/Serial-Baud-Rates) and make sure it matches the baud rate of the Ethernet to RS485 adapter
You can confirm the baud rate on the LCD screen on the front of your Inverter, within the Menu's.

The Modbus address be default is 1. Some people have reported that their Modbus address was 4.

In case you can read parameters, but cannot modify them, your inverter is probably still locked. See main README file for more information on how to unlock.

Attention: This work is based on the common protocol documentation for the X1 and X3 device. The documentation is not always clear which parameters do not apply to the X1 inverter, so some of the entities that appear may not have a meaning on X1. 

We have tested with following RS485 adapter devices:
## Option 1: Waveshare RS485 to ETH (B): (DIN rail mounted model)
> ![image](https://user-images.githubusercontent.com/11804014/160612700-35e494ee-174b-4137-83eb-7b26f9e67296.png)
> > > > ![image](https://user-images.githubusercontent.com/11804014/162572603-5e991bef-bd6d-490c-bcfa-21f1fa7464cb.png)

  - connects through Ethernet to your home network router/switch
  - use the "vircom" tool (Windows) to set the IP address matching your lan (or to let the device use DHCP, it does not use DHCP by default)
  - configure it to use "Modbus TCP to RTU", make sure the work mode is "TCP Server" and the server listens to port 502. Make sure the baudrate matches the baudrate of your inverter (see:  settings > advanced settings > modbus > baud rate)
![image](https://user-images.githubusercontent.com/11804014/162570111-adb222a5-0b44-4eac-bcab-d3a83375af34.png)

  - In "More Advanced Aettings...",
    - **SolaX** set Modbus Gateway Type to "Simple Modbus to TCP/IP mode"
    - **Sofar Solar** set Modbus Gateway Type to "Auto query storage type"

![image](https://user-images.githubusercontent.com/11804014/161312831-8f571b57-84d1-45e4-80b4-08a440c6ab66.png)
  - all other parameters can be left default
 - you can verify the config by opening the webpage of your adapter: http://<waveshare_ip_addr>. It should show something like:
![image](https://user-images.githubusercontent.com/11804014/162569619-0ca4dde8-503f-4966-901d-1b8a6800fb7a.png)
 - In the unlikely case you have multiple inverters connected to one RS485 adapter, you may need to modify the multi-host flag (untested). The system will increase the gap between messages, which will slow down the communication throughput. Make sure your polling interval is long enough (the act led should not light up all the time).

  - if all is well, the PWR LED should be red, the LINK LED should initially be green and become permanently blue as soon as HA has made the connection. The ACT LED should be off most of the time but should flash green (send) and blue (receive) briefly every 15 seconds (the default polling interval). From the waveshare documentation:
> 1) If the Link light is not green, the network cable is not connected properly, please check the
network cable. 
> 2) If the Link light is not blue (only consider the TCP working mode), the host computer
software does not have a serial port The server establishes a connection, please consider
whether the IP address is configured in the same network segment. 
> 3) If the Active light is green, it means that the network port is sending data, but there is no
serial device returning data. Please check whether the baud rate is configured properly and
whether the RS485 positive and negative are connected reversely
  - this device seems to have a low power consumption (does not become warm; I did not measure the power)

When configuring the SolaX Modbus integration, specify the network IP address of the Waveshare device, not the address of the SolaX.
Do not tick the Serial RS485 flag; the serial port is ignored in this setup.

### Multiple inverters connected to one Waveshare adapter
[... cut ...]
- A combined method of Solis Cloud and Home Assistant using [any Datalogger](https://github.com/alienatedsec/solis-ha-modbus-cloud#supported-solis-dataloggers) as well as wiring, and how to connect and configure Waveshare devices to work with this integration are described in detail at https://github.com/alienatedsec/solis-ha-modbus-cloud

**(*) Disconnecting Solis Cloud is to prevent Solis engineers and your installer to support and monitor your PV installation.**

The comparison of functions and dependencies:
![](https://github.com/alienatedsec/solis-ha-modbus-cloud/blob/master/images/datalogger-table.png)
```

wills106, 2023-03-31:
Just updated thanks. Seems odd how you can clone a Wiki but not push back.

alienatedsec, 2023-03-31:
You can clone, but not `fork` a wiki page; hence, why there is no `push`.

alienatedsec, 2023-04-02:
@wills106 Unfortunately, this issue came back with the latest `2023.03.2b16` - which started after the midnight
[SolaXLogs.zip](https://github.com/wills106/homeassistant-solax-modbus/files/11131309/SolaXLogs.zip)

[image]

[image]

wills106, 2023-04-02:
I don't think b16 would have reintroduced this error. I think the register is unreliable.

I have released [2023.03.2b17](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b17) I have added ```sleepmode = SLEEPMODE_LASTAWAKE,``` to "Battery SOC" like I have done on some of the "Total" and "Today" sensors.

wills106, 2023-04-05:
Is your "Battery SOC" behaving now?

alienatedsec, 2023-04-05:
@wills106 it looks normal since the b17 release, but the SOC is not my concern, but all other sensors producing extreme numbers and messing with the graphs at the same time until the integration is restarted. I only inserted SOC to present timelines and when it started. 

If this will happen again, we could reopen the issue with further information.

alienatedsec, 2023-04-05:
It didn't take long before the next problem occurred. This time, I had an automation to restart SolaX integration automatically when the battery SOC is at 8% or 0% - in theory, that should never happen. See the logs, but those are not showing anything useful to me on this occasion.

[SolaXError_b18.zip](https://github.com/wills106/homeassistant-solax-modbus/files/11161486/SolaXError_b18.zip)

[image]

So when the battery SOC drops to this weird (garbage) value, it also gives other (garbage) values, and I would not point at the unreliable register, which I believe is the same issue since my very first post in this thread.

[image]

[image]

[image]

That gives some false figures used for the energy dashboard - look at the slot 15:00 - 16:00 when the issue happened.

[image]

If you look at the previous logs I uploaded in the past, there is more indication of which sensors produce garbage until the integration is restarted.

alienatedsec, 2023-04-05:
@wills106 previous examples below

```
2023-02-26 07:35:10.497 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_power_generation_this_month, value dropped from 320.0 to 2023-02-26T07:31:20.320450+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.497 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_power_generation_last_month, value dropped from 269.0 to 2023-02-26T07:31:20.320642+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.497 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_power_generation_yesterday, value dropped from 10.4 to 2023-02-26T07:31:20.320914+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.498 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_power_generation_this_year, value dropped from 589.0 to 2023-02-26T07:31:20.321087+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.498 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_power_generation_last_year, value dropped from 1021.0 to 2023-02-26T07:31:20.321254+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.498 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_battery_charge_today, value dropped from 3.5 to 2023-02-26T07:31:20.328894+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.499 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.solislan_battery_discharge_today from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 500.2 (500.3) with last_updated set to 2023-02-26T07:31:45.327452+00:00. Please report it to the custom integration author.
2023-02-26 07:35:10.500 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_grid_import_today, value dropped from 10.0 to 2023-02-26T07:31:20.329730+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.500 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.solislan_house_load_total from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 15630352.6 (15630352.7) with last_updated set to 2023-02-26T07:31:40.328177+00:00. Please report it to the custom integration author.
2023-02-26 07:35:10.500 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:31:25.361911+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.500 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:31:35.330527+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.500 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:31:50.369277+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:32:05.327419+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:32:15.331270+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:32:30.338283+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:32:40.351080+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:33:00.345133+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:33:15.351697+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:33:35.368193+00:00, triggered by state with last_updated set to 0.0
2023-02-26 07:35:10.501 INFO (Recorder) [homeassistant.components.sensor.recorder] Detected new cycle for sensor.solislan_house_load_today, value dropped from 6553.5 to 2023-02-26T07:33:50.353994+00:00, triggered by state with last_updated set to 0.0
[... cut ...]
2023-02-26 07:35:15.326 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0x80ec: serialnumber
2023-02-26 07:35:15.328 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0x80fe: rtc
2023-02-26 07:35:15.331 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0x8105: power_generation_total
2023-02-26 07:35:15.332 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0x8107: power_generation_this_month
```

alienatedsec, 2023-04-17:
It has been several days since it occurred last - I guess, I will close this issue and it will happen again :) Some latest logs below

```
2023-04-16 02:13:58.515 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.solislan_meter_reactive_power (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.586 seconds. Please report it to the custom integration author.
2023-04-17 10:30:40.958 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for number.solislan_timed_discharge_start_hours_slot_2 (<class 'custom_components.solax_modbus.number.SolaXModbusNumber'>) took 0.900 seconds. Please report it to the custom integration author.
```

alienatedsec, 2023-05-20:
I have now two waveshares at different baud rates

TCP Server at 19200 and the inverter the same
TCP Client at 9600 to match dataloggers defaults

Works fine (so far). The cloud is reporting as normal and the integration is polling data. The inverter is not supporting anything higher than 38400.

_Originally posted by @alienatedsec in https://github.com/wills106/homeassistant-solax-modbus/issues/401#issuecomment-1555884056_

wills106, 2023-05-20:
Another option could be to adapt Infradom's https://github.com/infradom/ha-addon-modbusspy instead of having so many Waveshares.
At the moment ModbusSpy passively listens via a USB-RS485 adaptor the Modbus traffic and relays it as Modbus over TCP server so another device can connect and read data as if it was directly connected. (It can only read and not write to the main datasource)
You could look into adding a sencond USB Interface in adition to the Modbus over TCP connection. Then any devices that want to read data can, withough interfering with the main connection.

I don't have anything to attach a Waveshare to so I can't test to see if having multiple connected in your setup are causing any issues or not. It's also potentially a lot cheaper to have a couple of USB-RS485 adaptors connected running ModbusSpy compared to the multiple Waveshare setup.

alienatedsec, 2023-05-20:
@wills106 this sounds great and thanks for some alternatives; however, there are several obstacles I could think about:
- at the moment, my equipment is separated - HA in the office running on my QNAP - unlikely to connect any USBs there and the inverter is far away in the garage :)
- reading only is an option, but it limits many abilities - e.g., remote support (firmware, settings), remote control via Solis Cloud (who would need this if they have HA but still), maybe more etc.
- Waveshares are already doing all they need to do - queries storage and queuing - yes, it is more expensive, but everything comes with the price

I still think that those false readings are due to Waveshare losing the connection with the inverter. Two ways to address it:

1. improve the communication link between Waveshare and the inverter - Increase baud rate, trying some alternative settings as per the above

2. the integration to handle unexpected readings as false readings and reject those (not sure it is possible). An example: the battery SoC dropping from 95% to 0% is unexpected and should be discarded. The PV total power dropping from 4000W to 0W is likely to be expected and could be instant.

I am not a programmer and many things are still on my ToDo list :), but having some exposure in the industry and ChatGPT, gives an idea of how to handle such reading issues. The below was created based on my assumptions and some can be amended accordingly. I could have attempted to include it in the code, but I have no time to digest all things that I am missing at the moment (knowledge). Even as an example, it refers to many things that would not be relevant (setting the Modbus TCP IP) or even incorrect (32000 is not a holding register), but I am sure it could give a clue about how to handle unexpected readings.

There is another problem which could prevent the below effort from being successful. There is no easy way from HA to register that unexpected reading unless it is verified as unexpected - again something that I would need to expand my knowledge on this.

```
from pymodbus.client.sync import ModbusTcpClient as ModbusClient
from pymodbus.exceptions import ConnectionException
import time
import logging

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.INFO)

MODBUS_IP = 'your_modbus_device_ip'
REGISTER_ADDRESS = 32000

def read_battery_status(client):
    # assuming the battery status is a single register
    result = client.read_holding_registers(REGISTER_ADDRESS, 1, unit=1)

    # checking for connection issues
    if not result.isError():
        return result.registers[0]
    else:
        log.error(f"Error reading battery status: {result}")
        return None

def main():
    client = ModbusClient(MODBUS_IP)
    prev_status = None  # setting initial status to None
    while True:
        try:
            client.connect()

            current_status = read_battery_status(client)
            if current_status is not None:
                if current_status == 0 and (prev_status is None or prev_status > 5):
                    log.error("Unexpected battery status reading, resetting...")
                    # here you can put the code that restarts your home assistant integration
                    client.close()
                    time.sleep(10)  # assuming a 10 second delay for the reset
                    client = ModbusClient(MODBUS_IP)
                else:
                    prev_status = current_status
            client.close()
            time.sleep(1)  # delay before the next reading

        except ConnectionException:
            log.error("Error connecting to Modbus device, retrying...")
            client.close()
            time.sleep(10)  # delay before retrying

if __name__ == "__main__":
    main()
```

alienatedsec, 2023-06-05:
I had another problem overnight.

[image]

The pattern is that it drops the battery to `0` and I am using the below automation to reload the `SolaX` integration when this happens - I have been using it since shortly after I raised this issue the first time - that is to avoid long-term data corruption. As mentioned earlier, the integration has to be restarted for everything to go back to normal. My automation also sends a Telegram message, so I am aware of the problem at the time.

```
alias: Reload SolisLAN
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.solislan_battery_soc
    to: "0"
condition: []
action:
  - service: telegram_bot.send_message
    data:
      title: SOLISLAN CRASHED
      message: Problem with SolisLAN - Restarting Service
  - service: homeassistant.reload_config_entry
    data: {}
    target:
      device_id: 159ba7c9a31.....................
mode: single
```

While I agree with @wills106 that Solis is not the most reliable regarding comms on RS485, I think we should mitigate it somehow with the integration. 

**Now, I see that the integration is registering data despite it having reading errors**. @infradom do you have any ideas on how to prevent the integration from registering any data when read errors occur? If I don't reload the integration it keeps registering data for sensors despite not getting it from the inverter, and the below errors are generated. In my earlier posts and logs, you can find some randomly generated data for many sensors if the integration is not reloaded.

```
2023-06-04 04:25:36.432 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for number.solislan_battery_charge_current (<class 'custom_components.solax_modbus.number.SolaXModbusNumber'>) took 0.792 seconds. Please report it to the custom integration author.
2023-06-04 09:23:36.418 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.solislan_meter_grid_import_total (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.723 seconds. Please report it to the custom integration author.
2023-06-04 15:42:29.111 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for number.solislan_timed_charge_start_hours_slot_3 (<class 'custom_components.solax_modbus.number.SolaXModbusNumber'>) took 0.895 seconds. Please report it to the custom integration author.
2023-06-04 22:41:04.711 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.solislan_grid_export_today (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.940 seconds. Please report it to the custom integration author.
2023-06-04 23:40:15.270 INFO (MainThread) [custom_components.solax_modbus] read_error : SolisLAN cannot read holding registers at device 1 position 0xa7ff
2023-06-04 23:52:22.566 WARNING (MainThread) [homeassistant.helpers.entity] Updating state for sensor.solislan_house_load_yesterday (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.864 seconds. Please report it to the custom integration author.
2023-06-05 00:00:10.747 WARNING (Recorder) [homeassistant.components.sensor.recorder] Entity sensor.solislan_power_generation_yesterday from integration solax_modbus has state class total_increasing, but its state is not strictly increasing. Triggered by state 40.7 (42.8) with last_updated set to 2023-06-04T22:59:51.857109+00:00. Please report it to the custom integration author.
2023-06-05 01:02:18.067 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa86e: battery_discharge_current
2023-06-05 01:02:18.068 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa885: timed_charge_current
2023-06-05 01:02:18.069 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa886: timed_discharge_current
2023-06-05 01:02:18.070 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa887: timed_charge_start_h
2023-06-05 01:02:18.070 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa888: timed_charge_start_m
2023-06-05 01:02:18.071 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa889: timed_charge_end_h
2023-06-05 01:02:18.072 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa88a: timed_charge_end_m
2023-06-05 01:02:18.072 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa88b: timed_discharge_start_h
2023-06-05 01:02:18.073 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa88c: timed_discharge_start_m
2023-06-05 01:02:18.074 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa88d: timed_discharge_end_h
2023-06-05 01:02:18.075 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa88e: timed_discharge_end_m
2023-06-05 01:02:18.077 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa891: timed_charge_start_h_2
2023-06-05 01:02:18.078 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa892: timed_charge_start_m_2
2023-06-05 01:02:18.079 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa893: timed_charge_end_h_2
2023-06-05 01:02:18.081 WARNING (MainThread) [custom_components.solax_modbus] SolisLAN: read failed at 0xa894: timed_charge_end_m_2
[... cut ...]
2023-06-05 01:02:19.262 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa842 backflow_power to block with start 0xa841
2023-06-05 01:02:19.263 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa86c
2023-06-05 01:02:19.263 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86c battery_chargedischarge_current to block with start 0xa86c
.........................
```

Everything is back to normal until the next issue occurs.

I can do testing with some experimental code if that is required.

alienatedsec, 2023-06-12:
It turns out the problem can be resolved another way. Thanks for the support @wills106 @infradom 
Resolution with the Waveshare firmware update described here https://github.com/alienatedsec/solis-ha-modbus-cloud/discussions/17 and settings amendments as per the official guideline https://github.com/alienatedsec/solis-ha-modbus-cloud#configuration

closed 2023-06-12 by alienatedsec

bilalhp, 2024-11-16:
@alienatedsec I still see spikes time to time although it is a lot less with the new firmware and config, but it happens and ruins my statistics. Another difference is that the integration does not get stuck anymore and corrects the data at a new reading. However, the spikes are there forever in the history view.

Do you not encounter any issues like this anymore?

alienatedsec, 2024-11-16:
No more spikes for me.

@bilalhp did you also change the inverter's baud rate?

I have changed it from 9600 to 38400. You have to make changes on both, the inverter and `TCP Server`. Leave `TCP Client` at 9600 as this has to match your datalogger.

I am not sure how often you pull the data but the above will help with any potential bottlenecks on your system.

bilalhp, 2024-11-16:
I have not changed the baud rate. I will try that.

I am trying the understand the root cause.

- Is the data reported by the inverter wrong? Then, there is nothing to do with it.
- Is the data getting corrupted on RS485 level? Then, I can try better cabling.
- Is it because the poll rate is too frequent? Causing some kind of collusion. It shouldn't, because the Modbus server supposed to handle it.

And how increasing the baud rate helps exactly? If it is a corruption of data, it is more likely with an increased baud rate.

bilalhp, 2024-11-16:
@alienatedsec and my polling frequency is 15 seconds.

alienatedsec, 2024-11-16:
@bilalhp I pull every 5 seconds. I have found it working better this way. Honestly, I don't remember pulling with any longer intervals than every 5 seconds.

bilalhp, 2024-11-16:
@alienatedsec apparently, my Solis inverter doesn't allow to change the baud rate. I cannot find it in the settings or manual.

bilalhp, 2024-11-25:
@alienatedsec reducing interval to 5 made it worse. I get spikes multiple times a day.

Now I set the interval to 20 secs.

I cannot change the baud rate on this Solis inverter. I think it is not possible.

I need to understand the root cause of this spikes. Is it a conflict happening on the bus? Waveshare master device supposed to be able to prevent this conflict, and it should in fact know when the data on the bus is corrupted. So, how come it reports garbage data?

Can the situation be improved by better cabling?

Those are the questions I have right now.

BTW, I switched back to "Auto-query storage type" and increased the bus conflict detection interval.