wills106/homeassistant-solax-modbus #341: [BUG] Plugin stops updateing once a day

J14111, 2023-03-03:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version: 2023.02.6
2. HA Version: 2023.3.1
3. Inverter Brand: Solis
4. Plugin Used: Solis
5. First 6 digits of Serial: Dont have to hand
6. Connection Method: TCP

I have noticed the last 2 or 3 days the plugin seems to stop updating every day sometime between 3-6am it only seems to effect this plugin and a restart of HA resolves it, is anyone else seeing this?

I will check the logs when it happens tomorrow and update

wills106, 2023-03-05:
This is potentially the same issue as #340

J14111, 2023-03-05:
Thanks i will have a look, it stopped about 10am yesterday. I think it started happening after a recent HA update

alienatedsec, 2023-03-06:
@J14111 you are using TCP, but how? is that through a data logger? If so, which one do you have?

J14111, 2023-03-06:
Im using the USR-W610 RS485/TCP adapter

alienatedsec, 2023-03-06:
Despite using Waveshare, I found that the 'Instruction Time Out' had to be extended to around 5088ms (moves every 32)
[image]

It will be equivalent to your 'polling time out' settings
[image]

J14111, 2023-03-06:
Thanks I will give it a go, it has worked fine for the past few months without issue though.

I have checked mine and I acutally have that turned off??

[image]

J14111, 2023-03-06:
Interestingly turning that on and setting it to 5088 just crashed my HA instance, just restarting now.

wills106, 2023-03-06:
@J14111 I would try this for now https://github.com/wills106/homeassistant-solax-modbus/issues/340#issuecomment-1455065380

J14111, 2023-03-06:
> @J14111 I would try this for now [#340 (comment)](https://github.com/wills106/homeassistant-solax-modbus/issues/340#issuecomment-1455065380)

Thanks I will test that now.

My HA is really not happy with that modbus polling setting turned on above for some reason.

J14111, 2023-03-07:
@wills106 Didnt fix it crashed again today around 7am 
[image]

J14111, 2023-03-09:
Still happening, I have set an automation to restart HA when it stops.

There is a HA update out today so will see if that fixes

wills106, 2023-03-09:
The following PR #346 may help?
Not had time to implement it yet though.

It might be that the Inverter is sending bad data every so often and the Integration gets stuck.

J14111, 2023-03-09:
yeab but odd it only started happening this week

wills106, 2023-03-09:
Were you on a really old HA installation before?

Just wondering if you were on an old pyModbus v2.5.3
HA went to pyModbus v3.1.1 for a bit and is now on v3.1.3

I don't know if pyModbus version would alter how it handles bad data?

J14111, 2023-03-09:
No I usually update within a day or so of new versions so dobut I was on an old version before, I will keep an eye on it for a day or so and report back

alienatedsec, 2023-03-10:
did you apply a lower block size? @J14111

J14111, 2023-03-10:
> did you apply a lower block size? @J14111

Yeah it didnt make any difference

alienatedsec, 2023-03-10:
@J14111 What did you do to make it stable then? was it PR #346 ?

J14111, 2023-03-10:
I have no idea, my logic was a HA update started it so an HA update would eventually fix and it seems to have done

wills106, 2023-03-10:
pyModbus was updated in HA 2023.03.x and I matched the version in this Integration to it.
So maybe the change from pyModbus 3.1.1 to 3.1.3 solved the issue?

That PR doesn't actually do anything yet. I need to implement it in the plugin_xyz

J14111, 2023-03-11:
Appears this is still an issue, it just stopped updating, i restarted ha manually about an hour ago which makes it even odder as it doesnt seem like its a buffer or something filling up, anything else i can try?

wills106, 2023-03-11:
Just released [2023.03.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b2) implementing #346 
I have left the blocksize at 48 for now.
If you are still getting errors can you manually decrease the blocksize to 40 and let me know the new block configuration.
It will look something like:

> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning holding block: 0xa841 0xa86f [43073, 43074, 43116, 43117, 43118]
> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning holding block: 0xa885 0xa88f [43141, 43142, 43143, 43144, 43145, 43146, 43147, 43149, 43150]
> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning input block: 0x80ec 0x811d [33004, 33022, 33029, 33031, 33033, 33035, 33036, 33037, 33039, 33049, 33050, 33051, 33052]
> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning input block: 0x811d 0x8147 [33053, 33054, 33055, 33056, 33057, 33073, 33076, 33079, 33081, 33083, 33093, 33094]
> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning input block: 0x8166 0x8197 [33126, 33128, 33129, 33130, 33132, 33133, 33134, 33135, 33139, 33140, 33141, 33142, 33143, 33144, 33147, 33148, 33149, 33161, 33163, 33164, 33165, 33167, 33168, 33169, 33171, 33172, 33173]
> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning input block: 0x8197 0x819d [33175, 33176, 33177, 33179, 33180]
> 2023-02-26 10:13:14.932 INFO (MainThread) [custom_components.solax_modbus.sensor] SolisLAN returning input block: 0x81e3 0x8207 [33251, 33252, 33263, 33265, 33271, 33273, 33279, 33281, 33282, 33283, 33285]

This is what @alienatedsec provided here https://github.com/wills106/homeassistant-solax-modbus/issues/340#issuecomment-1445320276

J14111, 2023-03-11:
Thanks i will test and report back

wills106, 2023-03-13:
Just wondering if you have been able to confirm if [2023.03.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b2) has fixed the issue or not?

wills106, 2023-03-13:
Ok, well hopefully it won't need a restart over tonight / tomorrow?

J14111, 2023-03-14:
Just had a restart so its still happening, I have set the block size down to 40 so will keep an eye on it. There has also been a HA update to 2023.3.4 so I have installed that.

Where can I get the the block config to send you?

wills106, 2023-03-14:
If you add:
```
logger:
  default: info
```
To your configuration.yaml you will get more logging information.

Could you provide me with the log file with it crashing as it is now (With the above added to your config). Then we can try and work out what it's actually failing on.

J14111, 2023-03-14:
Ok will do

J14111, 2023-03-14:
Here is the startup info:

2023-03-14 13:37:10.869 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None
2023-03-14 13:37:10.869 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa7ff power_switch to block with start 0xa7ff
2023-03-14 13:37:10.869 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa841
2023-03-14 13:37:10.869 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa841 backflow_power_switch to block with start 0xa841
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa842 backflow_power to block with start 0xa841
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86c battery_chargedischarge_current to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86d battery_charge_current to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86e battery_discharge_current to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa885 timed_charge_current to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa886 timed_discharge_current to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa887 ro_timed_charge_start_h to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa888 ro_timed_charge_start_m to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa889 ro_timed_charge_end_h to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88a ro_timed_charge_end_m to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88b ro_timed_discharge_start_h to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88c ro_timed_discharge_start_m to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88d ro_timed_discharge_end_h to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88e ro_timed_discharge_end_m to block with start 0xa86c
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x80ec serialnumber to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x80fe rtc to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8105 power_generation_total to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8107 power_generation_this_month to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8109 power_generation_last_month to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810b power_generation_today to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810c power_generation_yesterday to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810d power_generation_this_year to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810f power_generation_last_year to block with start 0x80ec
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x8119
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8119 pv_voltage_1 to block with start 0x8119
2023-03-14 13:37:10.870 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x811a pv_current_1 to block with start 0x8119
2023-03-14 13:37:10.871 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x811b pv_voltage_2 to block with start 0x8119
2023-03-14 13:37:10.871 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x811c pv_current_2 to block with start 0x8119
2023-03-14 13:37:10.871 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x811d pv_voltage_3 to block with start 0x8119
2023-03-14 13:37:10.871 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x811e pv_current_3 to block with start 0x8119
2023-03-14 13:37:10.871 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x811f pv_voltage_4 to block with start 0x8119
2023-03-14 13:37:10.871 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8120 pv_current_4 to block with start 0x8119
[... cut ...]
2023-03-14 13:37:10.873 INFO (MainThread) [custom_components.solax_modbus.sensor] Solis returning input block: 0x8197 0x819d [33175, 33176, 33177, 33179, 33180]
2023-03-14 13:37:10.873 INFO (MainThread) [custom_components.solax_modbus.sensor] Solis returning input block: 0x81e3 0x8207 [33251, 33252, 33263, 33265, 33271, 33273, 33279, 33281, 33282, 33283, 33285]
2023-03-14 13:37:10.878 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {'battery_input_energy': SolisModbusSensorEntityDescription(key='battery_input_energy', device_class=<SensorDeviceClass.POWER: 'power'>, entity_category=None,

wills106, 2023-03-14:
Are you using the release 2023.03.b2 ?
That was setup with ignore_readerror = True, to work with a block size of 48 and not 40.

alienatedsec, 2023-03-14:
@J14111 you need to amend the line `1689` for block_size from `48` to `40` in `plugin_solis.py` and restart Home Assistant

https://github.com/wills106/homeassistant-solax-modbus/blob/9211bec4591909020ad5d78fc896dfb5e1b2d036/custom_components/solax_modbus/plugin_solis.py#L1683-L1692

J14111, 2023-03-14:
Ah sorry it already is, I must be on B2 because its just prompted me to upgrade to B3

[image]

So you want me to upgrade to B3 then set block_size to 40 again or stay on B2 with 40?

wills106, 2023-03-14:
@J14111 I don't want you to change anything in the Integration. I am trying to establish if https://github.com/wills106/homeassistant-solax-modbus/issues/341#issuecomment-1468161788 was the output of block_size = 40 or 48?

I released 2023.03.b2 / 2023.03.b3 which I believe had ```ignore_readerror = True,``` setup for a block_size of 48 but that output from the error log doesn't match what I expected.

J14111, 2023-03-14:
Yes that output was with block_size = 40

wills106, 2023-03-14:
@J14111 I have just released [2023.03.2b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b4https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b4) with the block_size set to 40 and I have reconfigured ```ignore_readerror = True,``` for the block_size = 40

Could you upgrade to that version and confirm in your error log you still have:

> Solis returning holding block: 0xa7ff 0xa800 [43007]
Solis returning holding block: 0xa841 0xa843 [43073, 43074]
Solis returning holding block: 0xa86c 0xa88f [43116, 43117, 43118, 43141, 43142, 43143, 43144, 43145, 43146, 43147, 43148, 43149, 43150]
Solis returning input block: 0x80ec 0x8111 [33004, 33022, 33029, 33031, 33033, 33035, 33036, 33037, 33039]
Solis returning input block: 0x8119 0x813d [33049, 33050, 33051, 33052, 33053, 33054, 33055, 33056, 33057, 33073, 33076, 33079, 33081, 33083]
Solis returning input block: 0x8145 0x816e [33093, 33094, 33126, 33128, 33129, 33130, 33132, 33133]
Solis returning input block: 0x816e 0x8197 [33134, 33135, 33139, 33140, 33141, 33142, 33143, 33144, 33147, 33148, 33149, 33161, 33163, 33164, 33165, 33167, 33168, 33169, 33171, 33172, 33173]

> So either we create new blocks explicitly with newblock statements (combined with ignore_readerror) or we extend the core system with a auto_ignore_readerror flag in the plugin_instance declaration.

It will probably worth automating this assuming this fixes the issue. Saves having to reconfigure again if we add in more registers.

J14111, 2023-03-14:
Ok updated,  log:

2023-03-14 18:40:20.711 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2023 3
2023-03-14 18:40:20.711 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2023-03-14 18:40:20.712 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-03-14 18:40:20.729 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Solis: trying to determine inverter type
2023-03-14 18:40:20.830 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read Solis 0x80ec serial number: 60310232282000, swapped: False
2023-03-14 18:40:20.841 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-03-14 18:40:20.846 INFO (MainThread) [homeassistant.setup] Setting up button
2023-03-14 18:40:20.847 INFO (MainThread) [homeassistant.setup] Setup of domain button took 0.0 seconds
2023-03-14 18:40:20.848 INFO (MainThread) [homeassistant.setup] Setting up select
2023-03-14 18:40:20.849 INFO (MainThread) [homeassistant.setup] Setup of domain select took 0.0 seconds
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa7ff power_switch to block with start 0xa7ff
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa841
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa841 backflow_power_switch to block with start 0xa841
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa842 backflow_power to block with start 0xa841
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86c battery_chargedischarge_current to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86d battery_charge_current to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86e battery_discharge_current to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa885 timed_charge_current to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa886 timed_discharge_current to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa887 ro_timed_charge_start_h to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa888 ro_timed_charge_start_m to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa889 ro_timed_charge_end_h to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88a ro_timed_charge_end_m to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88b ro_timed_discharge_start_h to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88c ro_timed_discharge_start_m to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88d ro_timed_discharge_end_h to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa88e ro_timed_discharge_end_m to block with start 0xa86c
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x80ec serialnumber to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x80fe rtc to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8105 power_generation_total to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8107 power_generation_this_month to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8109 power_generation_last_month to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810b power_generation_today to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810c power_generation_yesterday to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810d power_generation_this_year to block with start 0x80ec
2023-03-14 18:40:20.850 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x810f power_generation_last_year to block with start 0x80ec
2023-03-14 18:40:20.851 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x8119
2023-03-14 18:40:20.851 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8119 pv_voltage_1 to block with start 0x8119
[... cut ...]
2023-03-14 18:40:20.856 INFO (MainThread) [custom_components.solax_modbus.sensor] Solis returning input block: 0x816e 0x8197 [33134, 33135, 33139, 33140, 33141, 33142, 33143, 33144, 33147, 33148, 33149, 33161, 33163, 33164, 33165, 33167, 33168, 33169, 33171, 33172, 33173]
2023-03-14 18:40:20.856 INFO (MainThread) [custom_components.solax_modbus.sensor] Solis returning input block: 0x8197 0x819d [33175, 33176, 33177, 33179, 33180]
2023-03-14 18:40:20.856 INFO (MainThread) [custom_components.solax_modbus.sensor] Solis returning input block: 0x81e3 0x8207 [33251, 33252, 33263, 33265, 33271, 33273, 33279, 33281, 33282, 33283, 33285]

wills106, 2023-03-14:
So your new block groups matches up with how ```ignore_readerror = True,``` has been configured.

So we just need to wait an see how it behaves over the next day or two.
If it does fail can you provide the full log so we can try and work out what it's failing on?

J14111, 2023-03-14:
Of course will do, thanks for your help :)

infradom, 2023-03-17:
> NameError: name 'start' is not defined

Oops, should be block.start ...

wills106, 2023-03-17:
Just pushed @infradom PR out in [2023.03.2b6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b6)

J14111, 2023-03-17:
> Just pushed @infradom PR out in [2023.03.2b6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b6)

Thanks, installing now will let you know how it goes

J14111, 2023-03-20:
I had another restart yesterday, not as much in the logs that I can see this time:

2023-03-19 22:28:26.131 INFO (MainThread) [custom_components.solax_modbus] read_error : Solis cannot read input registers at device 1 position 0x81e3

and

  File "/config/custom_components/solax_modbus/number.py", line 117, in native_value
    if self._read_scale: return self._hub.data[self._key]*self._read_scale
TypeError: unsupported operand type(s) for *: 'NoneType' and 'int'

wills106, 2023-03-20:
Just pushed the PR out in [2023.03.2b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b7)

J14111, 2023-03-20:
Thanks, installing now

J14111, 2023-03-21:
Its looking a lot better my auto restart automation has not run for 2 days now:

[image]

I will keep an eye on it for another day or two, thanks! :)

kai-seipp, 2023-03-21:
As mentioned to Will within a private conversation before, I too have the problem of the integration loosing the connection during the night. But I am not sure if the problem is the same. 

In my configuration one of my two Solis inverters will actually go offline during the night as it does not have a battery and will therefore lose energy once the sun is gone. 

Before the beta 2023.03.2b4 the integration would not go to "unavailable" during that time, but just keep the last seen values. At least as long as I don't restart my HA instance. 

Since 2023.03.2b5 this behaviour has changed to all values going unavailable after some time. I think the behaviour is actually ok (the inverter actually is unavailable) but it will not regain connection by itself in the morning, when the inverter comes back online. And that is a problem - change from the behaviour before. 

I noticed that other integrations seam to have solved similar issues (e.G. the Denon integration). I power down my entertainment Center using a smart plug during the times I won't use it. So during these times the integration will go "unavailable" as well, but seems to have a loop function trying to reconnect every few minutes. This way it comes back quite quick once I power up the entertainment Center. Maybe this could be an approach to look into that could be adapted to get this integration back without the need to reload the integration by hand?

kai-seipp, 2023-03-24:
Hey, 

I did try to set the block size back to 48 again. But that did not solve the issue. 

I run 23.3.2b8 and still have the problem that the integration will not resume working in the morning. Any ideas what I could try? Thanks!

kai-seipp, 2023-03-24:
I was just thinking what else did change since the behavior of the integration has changed. So I realized that I also updated to Home Assistent 2023.03.05. So the change in behavior could actually not be caused by changes within the integration but it's surroundings. Just a thought I wanted to share with you!

J14111, 2023-03-24:
Im 99.99% sure this is fixed for me, not had any issues for 3 days, I will let you know if I see any. Thanks so much!

closed 2023-03-24 by J14111