wills106/homeassistant-solax-modbus #348: [Feature Request] Restsrt invertor button - Solis

J14111, 2023-03-11:
Hi,

Would it be possible to get a button to restart the inverter, i guess there must be a register for this?

wills106, 2023-03-11:
There doesn't seem to be an actual restart register. But there is a Power switch one. I have included it in [2023.03.2b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b2) it's called "Power Switch"

J14111, 2023-03-11:
Thanks so much will give it a test

wills106, 2023-03-21:
Have you been able to confirm if this switch works ok?

wills106, 2023-03-30:
Have you been able to test this?
Just wondering if I can close some of these issues down.

J14111, 2023-03-30:
I did a quick test and it seems to stop production and start it again but I havnt been ablet to check on the screen, I think it is more an stop/start switch rather than a restart but thats fine thanks.

closed 2023-03-30 by J14111