wills106/homeassistant-solax-modbus #351: [Feature Request] Solid Meter import/export

andybites, 2023-03-11:
The Solis Modbus map has these :

3283-32 84
Meter grid import active energy
U32 0.01kWh 100<-->1kWh 3X

3285-32 86
Meter grid export active energy
U32 0.01kW 100<-->1kWh 3X

I believe these are the live import/export states? If so, is it possible to include them - I cannot see any live import/export figures at the moment.

Many Thanks

Andrew

wills106, 2023-03-12:
Which Solis Inverter are you using? Those are registers for plugin_solis_old
Are you using that plugin or plugin_solis ?

andybites, 2023-03-12:
Hi, I'm using the Solis plug in, I've got an S5-EH1P6K-L.
I want to be able to see what I'm currently exporting and importing. 
I can see the 'Meter Active Power' entity - which usually looks right, but I expected there would be different entities for export and import.
I suppose I could template 2 entities from this one, to populate one for when the value is +'ve and another when -'ve, but I'm not sure that would be right - presumably it's possible times export and import happens simultaneously? or does it work more like water flowing and can only go one way? sorry for my naivety!

wills106, 2023-03-12:
> I expected there would be different entities for export and import.

Power Meter sensors tend to be bidirectional.

So for "Meter Active Power":
positive = export to grid
negative = import from grid

> does it work more like water flowing and can only go one way?

Yes it can only go in one direction at a time.

The only time you would appear to be importing and exporting at the same time would be in a 3 Phase setup.
You could import on one phase and export on another. Although each phase would only be flowing in a single direction at once.

andybites, 2023-03-12:
Awesome, really appreciate the explanation! I'll close this off now!

closed 2023-03-12 by andybites