wills106/homeassistant-solax-modbus #352: [BUG] ModBus Stop's working

pchaves1, 2023-03-14:
Hello,

on the last 24 hours, my integration between HA and ModBus stopped 3 times. 
Never saw this problem before with other versions of HA.
To recover I need to reload the integration on HA.
Any advice?

**Mandatory details**
1. Integration Version: 2023.03.1
2. HA Version: 2023.3.4
3. Inverter Brand: Solax
4. Plugin Used: SolaX Inverter Modbus
5. First 6 digits of Serial: H4752M
6. Connection Method: ModBus TCP using WaveShare RS485 TO ETH (B)

**Detailed Error Log**

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:472
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 17:18:41 (17 occurrences)
Last logged: 18:03:23

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 410, in read_modbus_data
    res = self.read_modbus_registers_all()
  File "/config/custom_components/solax_modbus/__init__.py", line 485, in read_modbus_registers_all
    res = res and self.read_modbus_block(block, 'holding')
  File "/config/custom_components/solax_modbus/__init__.py", line 472, in read_modbus_block
    val = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 389, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

wills106, 2023-03-14:
Does it indicate in your error log which register it failed on?

pchaves1, 2023-03-14:
Yes,

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:427
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 18:41:06 (8 occurrences)
Last logged: 18:41:06

SolaX: read failed at 0x10a: active_power_upper
SolaX: read failed at 0x10c: active_power_lower
SolaX: read failed at 0x10e: reactive_power_upper
SolaX: read failed at 0x110: reactive_power_lower
SolaX: read failed at 0x114: charge_discharge_power
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 427, in treat_address
    elif descr.unit == REGISTER_S32: val = decoder.decode_32bit_int()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 429, in decode_32bit_int
    handle = self._unpack_words(fstring, handle)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 353, in _unpack_words
    handle = unpack(f"!{wc_value}H", handle)
struct.error: unpack requires a buffer of 4 bytes

-------------------

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:424
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 18:41:06 (5 occurrences)
Last logged: 18:41:06

SolaX: read failed at 0x131: external_generation
SolaX: read failed at 0x132: external_generation_max_charge
SolaX: read failed at 0x54: lock_state
SolaX: read failed at 0xb8: meter_1_communication_state
SolaX: read failed at 0xb9: meter_2_communication_state
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 424, in treat_address
    if   descr.unit == REGISTER_U16: val = decoder.decode_16bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 389, in decode_16bit_uint
    return unpack(fstring, handle)[0]
struct.error: unpack requires a buffer of 2 bytes

-----------------

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:426
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 18:41:06 (6 occurrences)
Last logged: 18:41:06

SolaX: read failed at 0x9a: import_energy_today
SolaX: read failed at 0xaa: meter_2_export_total
SolaX: read failed at 0xac: meter_2_import_total
SolaX: read failed at 0xae: meter_2_export_today
SolaX: read failed at 0xb0: meter_2_import_today
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 426, in treat_address
    elif descr.unit == REGISTER_U32: val = decoder.decode_32bit_uint()
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 397, in decode_32bit_uint
    handle = self._unpack_words(fstring, handle)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/payload.py", line 353, in _unpack_words
    handle = unpack(f"!{wc_value}H", handle)
struct.error: unpack requires a buffer of 4 bytes

---------------

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:270
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 18:43:21 (78 occurrences)
Last logged: 18:44:06

SolaX: read failed at 0xac: meter_2_import_total
SolaX: read failed at 0xae: meter_2_export_today
SolaX: read failed at 0xb0: meter_2_import_today
SolaX: read failed at 0xb8: meter_1_communication_state
SolaX: read failed at 0xb9: meter_2_communication_state

wills106, 2023-03-14:
Have you altered your Modbus Power Meter in the past couple of days?
Do you have two Modbus Power Meters in your installation?

pchaves1, 2023-03-14:
No, I did not.
I only have one ModBus Power Meter.
Now I'm wondering if this is not related with Tigo Optimizers. I did request Tigo Energy to add the TCP integration between my Tigo CCA and WaveShare RS485 TO ETH (B). Not sure if is related or not

pchaves1, 2023-03-15:
It seems the problem is on the Tigo dev team making test for the integration (reading and sending data to the WaveShare RS485 TO ETH (B) gateway) and not on the SolaX Inverter Modbus plugin. Sorry for the time spend

wills106, 2023-03-15:
So this issue can be closed down?

pchaves1, 2023-03-15:
Yes please

closed 2023-03-15 by wills106