wills106/homeassistant-solax-modbus #353: Todays energy readings don’t start until device “wakes up”

wranglatang, 2023-03-15:
Hello,

i have noticed that the “Todays energy” sensors don’t seem to gather data until the inverter wakes up in the morning I.e when the sun rises. Which means and it also stops recording when the sun goes down (and batteries run out ) which means there are chunks of data missing at the start and end of the day
[image]

alienatedsec, 2023-03-15:
If your inverter is not powered on, it will not report some values on specific registers; hence, you cannot expect those to be retained or maintained. Thats how the inverters are designed to be powered by renewable sources - either PV or the battery.

However, I can see that your sensor is retaining today's data which resets at midnight and starts from 0 for the 'today's' value, which is 100% correct. The day (today's) starts every day from '0'.

wills106, 2023-03-15:
Like @alienatedsec mentions these Inverters whether PV Only or Hybrid / AC storage are only designed to provide energy.
So if they run out of PV or Battery storage, you will loose communications. There is nothing we can do about that.

If you need 24/7 monitoring you either need a bigger battery setup / forced charging during the night. Or another monitoring device. Bearing in mind any device monitoring will consume power somewhere.

wranglatang, 2023-03-16:
Got it, thanks for the clear explanation. Might be worth mentioning this someone in the doc's (sorry if it is and I have missed it). As for me, with only a (3kWh) battery, the value for the "Today" & "Total" sensors will be wrong as they will only start recording / calculating when the inverter is _powered_.

For example today the "Today" Sensor had it first reading of 0.1 at 7:40am but my base load is about 240w which should be about 1.4kWh

alienatedsec, 2023-03-17:
> For example today the "Today" Sensor had it first reading of 0.1 at 7:40am but my base load is about 240w which should be about 1.4kWh

That would be correct - your inverter is unable to monitor the usage if it is not operating (powered on). If you want more visibility and don't want to invest in a bigger battery, I recommend the Shelly device or something similar. An example is below. https://shellystore.co.uk/product/shelly-em-50a/

wranglatang, 2023-03-17:
Not needed :) The sensor i get from your integration for import / export power allows me to just use the home assistant integration to calculate the total energy consumption :)

closed 2023-03-17 by wranglatang