wills106/homeassistant-solax-modbus #358: [BUG]

mif11k, 2023-03-18:
**Describe the bug**
I get no data for the invertor. I get only 4 entities: Battery awaken, Unlock inverter, Unlock inverter advanced and Inverter temperature. All entities have "unknown" state. I get no inverter controls. 

As a temporary workaround I defined my own yaml sensors as described here (solaxg4.yaml): https://www.homeassistant-cz.cz/viewtopic.php?p=4397#p4397 to get data from the inverter, but this does not give me chance to change the battery operation mode.

**Mandatory details**
1. Integration Version: 2023.03.1
2. HA Version: (2023.3.5)
3. Inverter Brand: Solax X3-Hybrid G4
4. Plugin Used: homeassistant-solax-modbus
5. First 6 digits of Serial: SXKCJU
6. Connection Method: TCP

**Detailed Error Log**
2023-03-18 19:41:13.648 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-03-18 19:41:29.528 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-03-18 19:41:34.552 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-03-18 19:41:34.555 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-03-18 19:41:34.559 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2023-03-18 19:41:56.477 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ state_attr('sensor.solax', 'Data')[0] | int(default=2300) / 10}}")
2023-03-18 19:41:56.486 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 0') while processing template 'Template("{{ state_attr('sensor.solax', 'Data')[0] | int(default=2300) / 10}}")' for attribute '_attr_native_value' in entity 'sensor.solax_acu1'
2023-03-18 19:41:56.493 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ state_attr('sensor.solax', 'Data')[1] | int(default=2300) / 10}}")
2023-03-18 19:41:56.500 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 1') while processing template 'Template("{{ state_attr('sensor.solax', 'Data')[1] | int(default=2300) / 10}}")' for attribute '_attr_native_value' in entity 'sensor.solax_acu2'
2023-03-18 19:41:56.503 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{{ state_attr('sensor.solax', 'Data')[2] | int(default=2300) / 10 }}")
2023-03-18 19:41:56.508 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 2') while processing template 'Template("{{ state_attr('sensor.solax', 'Data')[2] | int(default=2300) / 10 }}")' for attribute '_attr_native_value' in entity 'sensor.solax_acu3'
2023-03-18 19:41:56.511 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[3] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[3] - 65536) | int(default=0) / 10 }} {% else %}{{ state_attr('sensor.solax', 'Data')[3] | int(default=0) / 10 }}{% endif %}")
2023-03-18 19:41:56.515 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 3') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[3] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[3] - 65536) | int(default=0) / 10 }} {% else %}{{ state_attr('sensor.solax', 'Data')[3] | int(default=0) / 10 }}{% endif %}")' for attribute '_attr_native_value' in entity 'sensor.solax_aci1'
2023-03-18 19:41:56.518 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[4] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[4] - 65536) | int(default=0) / 10 }} {% else %}{{ state_attr('sensor.solax', 'Data')[4] | int(default=0) / 10 }}{% endif %}")
2023-03-18 19:41:56.522 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 4') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[4] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[4] - 65536) | int(default=0) / 10 }} {% else %}{{ state_attr('sensor.solax', 'Data')[4] | int(default=0) / 10 }}{% endif %}")' for attribute '_attr_native_value' in entity 'sensor.solax_aci2'
2023-03-18 19:41:56.527 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[5] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[5] - 65536) | int(default=0) / 10 }} {% else %}{{ state_attr('sensor.solax', 'Data')[5] | int(default=0) / 10 }}{% endif %}")
2023-03-18 19:41:56.530 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 5') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[5] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[5] - 65536) | int(default=0) / 10 }} {% else %}{{ state_attr('sensor.solax', 'Data')[5] | int(default=0) / 10 }}{% endif %}")' for attribute '_attr_native_value' in entity 'sensor.solax_aci3'
2023-03-18 19:41:56.534 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[6] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[6] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[6] | int(default=0) }}{% endif %}")
2023-03-18 19:41:56.538 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 6') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[6] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[6] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[6] | int(default=0) }}{% endif %}")' for attribute '_attr_native_value' in entity 'sensor.solax_acp1'
2023-03-18 19:41:56.542 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[7] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[7] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[7] | int(default=0) }}{% endif %}")
2023-03-18 19:41:56.546 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 7') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[7] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[7] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[7] | int(default=0) }}{% endif %}")' for attribute '_attr_native_value' in entity 'sensor.solax_acp2'
2023-03-18 19:41:56.550 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[8] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[8] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[8] | int(default=0) }}{% endif %}")
2023-03-18 19:41:56.553 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 8') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[8] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[8] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[8] | int(default=0) }}{% endif %}")' for attribute '_attr_native_value' in entity 'sensor.solax_acp3'
2023-03-18 19:41:56.558 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template("{% if state_attr('sensor.solax', 'Data')[9] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[9] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[9] | int(default=0) }}{% endif %}")
2023-03-18 19:41:56.562 ERROR (MainThread) [homeassistant.helpers.template_entity] TemplateError('UndefinedError: None has no element 9') while processing template 'Template("{% if state_attr('sensor.solax', 'Data')[9] > 32767 %}{{ (state_attr('sensor.solax', 'Data')[9] - 65536) | int(default=0) }} {% else %}{{ state_attr('sensor.solax', 'Data')[ [... cut ...]
2023-03-18 21:38:41.746 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2023-03-18 21:38:46.775 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2023-03-18 21:38:46.779 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2023-03-18 21:38:46.784 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

wills106, 2023-03-20:
According to https://kb.solaxpower.com/data/detail/ff8080818407e2a701840a22dec20032.html you can only use Pocket WiFi v3.0 with Modbus on Firmware V3.004.03 it looks like you are on 3.001.02?

wills106, 2023-03-21:
You'll have to contact SolaX and get a FW update.
It should then work?

mif11k, 2023-03-21:
I asked SolaX to update the FW, I will report back. Thanks.

mif11k, 2023-03-22:
After FW upgrade to 3.008.10, everything is set. Integration works even via the LAN IP.

closed 2023-03-22 by mif11k