wills106/homeassistant-solax-modbus #363: [Feature Request] Solax Hybrid G4 - Add support for Charger upper limit

PxPert, 2023-03-21:
Hi,
First of all, thanks for your great work in this integrations, I'm currently using it for my Solax X1 Hybrid Gen4.

After a remote upgrade of the Inverter from the Solax Tech support, there are many (undocumented) new features supported by my Inverter. One of these is the support for setting Charger upper limit, to limit the maximum charge of my batteries.

Would it be possible to implement this feature? If you need other infos, I'm available to discover the correct registry key to get/set this value.

BTW the "Firmware Version Inverter Master" Value reported is **31**, and "Firmware Version Manager" is **30**

You can find [here](https://ibb.co/cbgPHVg) a photo of the charger setting from my inverter.

Thanks!

wills106, 2023-03-21:
Is this different to:

"Backup Nightcharge Upper SOC"
"Feedin Nightcharge Upper SOC"
"Selfuse Nightcharge Upper SOC"

PxPert, 2023-03-21:
Yes, that's not related to nightcharge: Nightcharge refers to Upper SOC charge from grid
This setting affects the global upper soc charge even during daytime (from the PV), to preverve the lifespan of the batteries

wills106, 2023-03-22:
I just wanted to double check they weren't the same.
The names in the docs and on the Inverter don't always match up ok.

PxPert, 2023-04-03:
No problem at all :)
If you need some dump over the RS485 port, just send me the commands to launch.
I'm using an USB to RS485 adapter attached to a Linux server, so it should be easy to do 

Thanks!

wills106, 2023-04-04:
You would have to use a Linux equivalent of Gmodbus to read the registers.
The problem is you will only be able to find the read only registers. On SolaX Inverters they tend to read from one register and write to an entirely different register.

So until I can find more up to date documentation there possibly isn't much we can do.

johny-mnemonic, 2023-04-11:
I just spent more than an hour trying to find which register is controlling this config variable available in the mobile app, but so far I have found nothing. Then I found this issue 😄 

These two looks the most promising:
```
register variable W/R decription unit format lenth
0x011B TargetSoc R TargetSoc 1% uint16 1
0x011C SocUpper R SocUpper 1% uint16 1
```
But the first one is always zero (X3-Hybrid-Gen4) and second one always 100 even though I have set the value to 90% using the app 🤷‍♂️ 
@wills106 could it be that for the SOC control to work we would need to mess with this one?
```
0x0100 ModbusPowerControl
```
It has following values:
```
0:disable remote control
1:enable power control
2:enable electric quantity control
3:enable SOC target control
```
So maybe setting to 3 would make the 0x011B/C work...

infradom, 2023-04-11:
> So maybe setting to 3 would make the 0x011B/C work...
I examined those parameters when we implemented modbus_power_control (option 1). The problem is that the way how to use the other options is not well understood (write multiple is involved, but unclear which other paramters participate).
I personally only use the variations of options 1 described in the Wiki (https://github.com/wills106/homeassistant-solax-modbus/wiki/Modbus-Power-Control-(remotecontrol-power)
Some very simple automations allow me to change mode if the SOC target is reached. I declare the SOC target as an input_number target_soc that the automation uses to switch modes when battery is almost full. (see same Wiki page)

These methods work extremaly well now for several months, so I do not need the options 2 or 3 this way.

wills106, 2023-04-25:
@johny-mnemonic does infradom's comment cover what you require?
Just wondering if I can close this down.

johny-mnemonic, 2023-04-28:
@wills106 honestly I already tried to read through that wiki page for power control but I am not sure how to use that to stop charging the battery at i.e. 90%. Would probably need to experiment with it or see some working automation to understand how it works.

ChequeM8, 2023-06-18:
I'd also really appreciate the 'Charger Upper Limit' adding to the Solax modbus integration if at all possible.

msteinbauer, 2023-08-08:
Would be very convenient if that works. At the moment I have an automation for that using the Solcast value for the remaining day and setting charge limit to 0 and waiting till remaining Solcast kWh are below 20 kWh. Then the charge limit is set to 30A again and battery is charged up.

wills106, 2023-09-03:
I'm going to close this down for now.
If I can get updated documentation I can add it in the future.

closed 2023-09-03 by wills106

wills106, 2023-09-24:
@PxPert Just released [2023.09.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.6)

There will be more changes this week, but wanted to get this out there first.

PxPert, 2023-09-27:
Great! I'll update my integration with the latest version and let you know how it works ;)