wills106/homeassistant-solax-modbus #364: Integration won't regain connection after inverter goes offline

kai-seipp, 2023-03-24:
I did describe the issue within #341, but since that one is closed now and I am still facing this issue I open this new issue now. 

In my configuration one of my two Solis inverters will actually go offline during the night as it does not have a battery and will therefore lose energy once the sun is gone.

Before the beta 2023.03.2b4 and before Home Assistant 2023.03.05 the integration would not go to "unavailable" during that time, but just keep the last seen values. At least as long as I don't restart my HA instance.

Since March 18th this behavior has changed to all values going unavailable after some time. I think the behavior is actually ok (the inverter actually is unavailable) but it will not regain connection by itself in the morning, when the inverter comes back online. And that is a problem - and a change from the behavior before.

Since I don't see any major changes within the integration itself, I am guessing that Home Assistant has changed something that is effecting this integration with the introduction of 2023.03.05. 

I noticed that other integrations seam to have solved similar issues (e.G. the Denon integration). I power down my entertainment Center using a smart plug during the times I won't use it. So during these times the integration will go "unavailable" as well, but seems to have a loop function trying to reconnect every few minutes. This way it comes back quite quick once I power up the entertainment Center. Maybe this could be an approach to look into that could be adapted to get this integration back without the need to reload the integration by hand?

`2023-03-23 19:25:29.445 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: reading serial number from address 0x80ec failed; other address may succeed
2023-03-23 19:25:29.465 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: cannot find serial number, even not for other Inverter
2023-03-23 19:25:29.467 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis-sued inverter type - serial number : unknown`

It would be really great if we could fix this / find a workaround for this, as this is affecting my abilities to collect statistical data for my solar system. Thanks!

alienatedsec, 2023-03-24:
@kai-seipp I get the integration behaviour........ but, have you also tried setting Waveshare to restart when no data is available - the setting is available via VirCom

[image]

kai-seipp, 2023-03-24:
Thank you 

> @kai-seipp I get the integration behaviour........ but, have you also tried setting Waveshare to restart when no data is available - the setting is available via VirCom
> 
Thank you! 
The waveshare Adapter I am using (RS232485 TO ETH) is not providing this option. 
Communication is constantly running over the same adapter with the other inverter. Therefore I was not expecting this beeing a problem - also since it did work before (until about a week ago). 

And also it is working right again, once I reload the integration for the south inverter in the morning. I am not touching the waveshare adapter for this. What do you think?

wills106, 2023-03-24:
I think it's possibly the change of adding in ```ignore_readerror = True,``` which has broken the code for when Inverters have gone to sleep. @infradom Have I not quite set ```ignore_readerror = True,``` correctly when I introduced it into plugin_solis ?
Although I haven't heard of any issues from X3-MIC users which is also using ```ignore_readerror = True,``` and those Inverters go to sleep at night.
I'm also a bit confused as I wouldn't expect ```ignore_readerror = True,``` to alter the behaviour of https://github.com/wills106/homeassistant-solax-modbus/blob/5d1a6f1e3a7381ca52b98e6549c025a48efbb854/custom_components/solax_modbus/plugin_solis.py#L2093

kai-seipp, 2023-03-26:
Hi Will, 

I am happy to confirm: going back to 2023.03.01 does solve the issue. 

Kai

wills106, 2023-03-26:
I thought it would do.

```ignore_readerror = True,``` and the previous sleepy Inverter code is either conflicting or I didn't setup ```ignore_readerror = True,``` correctly.

kai-seipp, 2023-03-26:
Well let me know if you want me to test a different implementation at some point. For now I would just stay on the last stable version. Thanks Will!

wills106, 2023-03-26:
Feedback is grateful and essential along with code input from others especially @infradom 
There are certain things I can't test as my Inverter never goes to sleep and I'm not a programmer by trade.

kai-seipp, 2023-03-26:
As said, I would be happy to help in any way I can. What is unfortunately quite limited as I can't code at all.

wills106, 2023-03-31:
@kai-seipp Could you try [2023.03.2b16](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b16) and let me know if it prevents the issue when you Inverter goes to sleep.

Are you able to make sure you add the following to your configuration.yaml

```
logger:
  default: info
```

Then could you provide a full log if you still get errors?

kai-seipp, 2023-03-31:
Did become unavailable again. 
I will see to pull the logs tomorrow for you. 

What I find quickly is: 

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2231
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 31 March 2023 at 20:32:32 (1 occurrences) 
Last logged: 31 March 2023 at 20:32:32

unrecognized solis-sued inverter type - serial number : unknown

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2217
Integration: SolaX Inverter Modbus 
First occurred: 31 March 2023 at 20:32:32 (1 occurrences) 
Last logged: 31 March 2023 at 20:32:32

solis-sued: cannot find serial number, even not for other Inverter

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2215
Integration: SolaX Inverter Modbus 
First occurred: 31 March 2023 at 20:32:32 (1 occurrences) 
Last logged: 31 March 2023 at 20:32:32

solis-sued: reading serial number from address 0x80ec failed; other address may succeed

infradom, 2023-04-01:
I am looking into this, but the failing function determineInverterType is only called once (upon startup of the integration), so I do not understand the full context . Did you restart the integration during the period that the inverter was offline ?

kai-seipp, 2023-04-01:
Hi Jan, 

Thanks! 
No I did not restart HA. For some reason - different from the current stable version - this happens once the inverter is powered off.

infradom, 2023-04-01:
@kai-seipp A full log could help us understand what happens

kai-seipp, 2023-04-01:
`2023-03-31 20:32:06.589 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-03-31 20:32:21.158 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'home_connect', 'history', 'sun', 'input_text', 'ibeacon', 'forecast_solar', 'input_datetime', 'tts', 'hardware', 'met', 'trace', 'homekit', 'application_credentials', 'ffmpeg', 'input_number', 'dlna_dms', 'mobile_app', 'zone', 'bosch', 'config', 'hacs', 'input_button', 'lovelace', 'image_upload', 'ipp', 'file_upload', 'icloud', 'counter', 'person', 'homeassistant_hardware', 'script', 'analytics', 'melcloud', 'media_source', 'onboarding', 'gardena_smart_system', 'derivative', 'co2signal', 'energy', 'remote', 'cast', 'conversation', 'tag', 'system_health', 'persistent_notification', 'ring', 'watchman', 'bosch_shc', 'downloader', 'automation', 'harmony', 'bthome', 'input_select', 'switch', 'timer', 'solax_modbus', 'upnp', 'group', 'zha', 'netatmo', 'rpi_power', 'adguard', 'template', 'dwd_weather', 'heos', 'diagnostics', 'sonos', 'apple_tv', 'input_boolean', 'stream', 'homeassistant_alerts', 'homeassistant_sky_connect', 'mqtt', 'blink', 'schedule', 'utility_meter', 'bluetooth_adapters', 'alexa_media', 'fritz', 'binary_sensor', 'blueprint', 'default_config', 'logbook', 'scene', 'homekit_controller', 'search', 'device_automation', 'switch_as_x', 'esphome', 'sensor', 'map', 'google', 'repairs', 'my', 'denonavr', 'min_max', 'shelly', 'raspberry_pi', 'auth', 'influxdb', 'easee', 'hue'}
2023-03-31 20:32:27.473 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2023-03-31 20:32:27.474 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2023-03-31 20:32:27.539 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2023 3
2023-03-31 20:32:27.539 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2023-03-31 20:32:27.540 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-03-31 20:32:27.571 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Solis-Nord: trying to determine inverter type
2023-03-31 20:32:27.646 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read Solis-Nord 0x80ec serial number: 110CA221814002, swapped: False
2023-03-31 20:32:27.647 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-03-31 20:32:27.648 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: trying to determine inverter type
2023-03-31 20:32:32.659 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: reading serial number from address 0x80ec failed; other address may succeed
2023-03-31 20:32:32.674 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read solis-sued 0x80ec serial number: None, swapped: False
2023-03-31 20:32:32.675 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: cannot find serial number, even not for other Inverter
2023-03-31 20:32:32.678 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis-sued inverter type - serial number : unknown
2023-03-31 20:32:32.691 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-03-31 20:32:32.693 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa7ff power_switch to block with start 0xa7ff
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa803 battery_minimum_soc to block with start 0xa7ff
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa810 backup_mode_soc to block with start 0xa7ff
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa841
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa841 backflow_power_switch to block with start 0xa841
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa842 backflow_power to block with start 0xa841
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa86c
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86c battery_chargedischarge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86d battery_charge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86e battery_discharge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa885 timed_charge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa886 timed_discharge_current to block with start 0xa86c
[... cut ...] native_unit_of_measurement=<UnitOfPower.WATT: 'W'>, options=None, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, suggested_display_precision=None, suggested_unit_of_measurement=None, allowedtypes=4096, scale=1, read_scale_exceptions=None, blacklist=None, register=-1, rounding=1, register_type=1, unit='_uint16', newblock=False, value_function=<function value_function_battery_output_solis at 0x7f932bdcf0>, wordcount=None, sleepmode=1, ignore_readerror=False, order16='>', order32='>')}`

kai-seipp, 2023-04-01:
It looks like I did have a restart of the system - but that would have been though a crash not initiated by me. So this is what I have since yesterday night on solax.

kai-seipp, 2023-04-01:
2023-03-31 20:32:06.589 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-03-31 20:32:21.158 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'home_connect', 'history', 'sun', 'input_text', 'ibeacon', 'forecast_solar', 'input_datetime', 'tts', 'hardware', 'met', 'trace', 'homekit', 'application_credentials', 'ffmpeg', 'input_number', 'dlna_dms', 'mobile_app', 'zone', 'bosch', 'config', 'hacs', 'input_button', 'lovelace', 'image_upload', 'ipp', 'file_upload', 'icloud', 'counter', 'person', 'homeassistant_hardware', 'script', 'analytics', 'melcloud', 'media_source', 'onboarding', 'gardena_smart_system', 'derivative', 'co2signal', 'energy', 'remote', 'cast', 'conversation', 'tag', 'system_health', 'persistent_notification', 'ring', 'watchman', 'bosch_shc', 'downloader', 'automation', 'harmony', 'bthome', 'input_select', 'switch', 'timer', 'solax_modbus', 'upnp', 'group', 'zha', 'netatmo', 'rpi_power', 'adguard', 'template', 'dwd_weather', 'heos', 'diagnostics', 'sonos', 'apple_tv', 'input_boolean', 'stream', 'homeassistant_alerts', 'homeassistant_sky_connect', 'mqtt', 'blink', 'schedule', 'utility_meter', 'bluetooth_adapters', 'alexa_media', 'fritz', 'binary_sensor', 'blueprint', 'default_config', 'logbook', 'scene', 'homekit_controller', 'search', 'device_automation', 'switch_as_x', 'esphome', 'sensor', 'map', 'google', 'repairs', 'my', 'denonavr', 'min_max', 'shelly', 'raspberry_pi', 'auth', 'influxdb', 'easee', 'hue'}
2023-03-31 20:32:27.473 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2023-03-31 20:32:27.474 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2023-03-31 20:32:27.539 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2023 3
2023-03-31 20:32:27.539 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2023-03-31 20:32:27.540 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-03-31 20:32:27.571 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Solis-Nord: trying to determine inverter type
2023-03-31 20:32:27.646 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read Solis-Nord 0x80ec serial number: 110CA221814002, swapped: False
2023-03-31 20:32:27.647 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-03-31 20:32:27.648 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: trying to determine inverter type
2023-03-31 20:32:32.659 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: reading serial number from address 0x80ec failed; other address may succeed
2023-03-31 20:32:32.674 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read solis-sued 0x80ec serial number: None, swapped: False
2023-03-31 20:32:32.675 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] solis-sued: cannot find serial number, even not for other Inverter
2023-03-31 20:32:32.678 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized solis-sued inverter type - serial number : unknown
2023-03-31 20:32:32.691 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-03-31 20:32:32.693 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa7ff power_switch to block with start 0xa7ff
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa803 battery_minimum_soc to block with start 0xa7ff
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa810 backup_mode_soc to block with start 0xa7ff
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa841
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa841 backflow_power_switch to block with start 0xa841
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa842 backflow_power to block with start 0xa841
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa86c
2023-03-31 20:32:32.718 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86c battery_chargedischarge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86d battery_charge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86e battery_discharge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa885 timed_charge_current to block with start 0xa86c
2023-03-31 20:32:32.719 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa886 timed_discharge_current to block with start 0xa86c
[... cut ...]  native_unit_of_measurement=<UnitOfPower.WATT: 'W'>, options=None, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>, suggested_display_precision=None, suggested_unit_of_measurement=None, allowedtypes=4096, scale=1, read_scale_exceptions=None, blacklist=None, register=-1, rounding=1, register_type=1, unit='_uint16', newblock=False, value_function=<function value_function_battery_output_solis at 0x7f932bdcf0>, wordcount=None, sleepmode=1, ignore_readerror=False, order16='>', order32='>')}

kai-seipp, 2023-04-02:
hmm, I really have the feeling the integration crashes my home assistant instance when my inverter goes offline. I have again no logs before 20:33 CET. 

And then getting all the startup announcements in the logs. 
I would go back to the stable version now. OK?

wills106, 2023-04-02:
Do you have any logs before these crashes at 20:32?
So we can see what exactly is causing the crash?

infradom, 2023-04-02:
Tomorrow, I will recheck the source code for potential causes of the crash. Sorry for the delay.

infradom, 2023-04-03:
@wills106 (plugin_solis) : Ignore_readerror must be set for the very entity of the first block, even if the global auto_block_ignore_readerror = True is set. This auto_bock... attribute is only used when new blocks are automatically created. The first block is not considered to be an automatically created block
This does not explain crashes of the integration however

wills106, 2023-04-26:
@kai-seipp It's not the best solution but a Tempory one... I have removed the sensor for "Serial Number" in [2023.03.2b24](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b24)
Does this prevent the Integration from crashing when you Inverter goes to sleep?

kai-seipp, 2023-04-28:
I run b24 now and it seams to work so far. :)

wills106, 2023-04-29:
Thanks for letting me know.

kai-seipp, 2023-04-29:
Well, thank you for putting all your time and work into it!

wills106, 2023-05-01:
I'm closing this issue as [2023.05.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.05.1) is released.
Although I need to relook into the issue at some point.

closed 2023-05-01 by wills106