wills106/homeassistant-solax-modbus #366: [BUG] Logs spammed with 'Detected blocking call to sleep inside the event loop'

rcourtman, 2023-03-28:
Describe the bug
The solax_modbus integration in Home Assistant is causing stability issues due to a blocking call to the sleep function inside the event loop.

Mandatory details

HA Version: 2023.3.6
Inverter Brand: Solis Hybrid
Plugin Used: solax_modbus
Connection Method: USB to RS485 (modbus)
Operating System: Raspberry Pi 4 with a 64-bit operating system
Detailed Error Log
[homeassistant.util.async_] Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/init.py, line 324: return self._client.read_input_registers(address, count, **kwargs)

Additional context
I have tried disabling and re-enabling the integration, but the issue persists. The logs also show that the same error message is being repeated frequently.

wills106, 2023-03-28:
It's a [known issue](https://github.com/wills106/homeassistant-solax-modbus#known-issues) No2.

You can also hide the errors following: https://github.com/wills106/homeassistant-solax-modbus/issues/329#issuecomment-1429600410

wills106, 2023-03-30:
Closing as I have updated the README.md

closed 2023-03-30 by wills106