wills106/homeassistant-solax-modbus #367: [BUG] Inaccurate Reading for Battery Charge & Discharge

wranglatang, 2023-03-29:
**Describe the bug**
I have recently been paying very close attention to my battery performance. I only have 1 3kW battery installed so its quite easy for me to get a full charge and discharge out of it. Today i am doing some tests where i am forcing the battery to charge to 100 then discharge to 10% then recording the values that I see reported into Home Assistant.

Below are my findings:

The Modbus integration tells me that the BMS is reporting my max Capacity to be 3072Wh. I know that the Round Trip Efficiency (RTE) of the Battery is ~95%. 
This morning I started a force charge when the battery was at 6% and it charged to 98%. On paper that charge should have been 2826.24 Wh but home assistant reported 3.1kWh's

I then started a force discharge when the battery was at 98% and it discharged to 10%. On paper that discharge (including the 95% RTE) should have been 2568.19 Wh and home assistant reported 2.4kWh's.

The discharge numbers seem pretty close but the charge numbers seem quite high? Im currently conducting a second test and will update with my results. 

[image]

wranglatang, 2023-03-29:
@alienatedsec - Sorry, Just updated this!

wranglatang, 2023-03-29:
Second Test Dataset Is in... This time it look more promising but still a slight difference (obviously i don't expect them to be bang on!)

[image]

ChequeM8, 2023-06-18:
I wouldn't expect the battery SOC to accurately represent any amount of kWh, especially at the lower and upper ends of the scale, it's only an approximation.

wills106, 2023-09-03:
I'm going to close this down as it isn't and Integration issue.
The Integration can only report what the Inverter provides over Modbus.

closed 2023-09-03 by wills106