wills106/homeassistant-solax-modbus #368: EV Charger modbus TCP not working

CristiXP, 2023-03-30:
**Describe the bug**
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version:2023.03.2b11
2. HA Version:Home Assistant 2023.3.6
3. Inverter Brand: Solax EV Charger
4. Plugin Used:Solax Modbus
5. First 6 digits of Serial:C32201
6. Connection Method: modbus TCP

**Detailed Error Log**
Detailed Error Log, if you press on the individual error further information is displayed.
This is the content we need, the description alone doesn't provide enough context.

If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:

Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”

Now the full logs are loaded. If you scroll down, you will see them.
Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.

Search for

solax

and report us the logs. Make sure to replace sensitive information by xxxx (if any)

**Additional context**
Add any other context about the problem here.
        
I have also bought a EV Charger. I have the inverter connected on modbus TCP which worked as a charge and also want to connect the EV charger the same, but I am getting this error: 

Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_solax_ev_charger.py:792 
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues)) 
First occurred: 8:10:42 PM (1 occurrences) 
Last logged: 8:10:42 PM

Error setting up entry SolaX_EV for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 107, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
  File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 792, in <module>
    plugin_instance = solax__ev_charger_plugin(
NameError: name 'solax__ev_charger_plugin' is not defined

_Originally posted by @CristiXP in https://github.com/wills106/homeassistant-solax-modbus/discussions/349#discussioncomment-5480214_

wills106, 2023-03-30:
Just released [2023.03.2b12](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b12) there was a typo in the plugin.

wills106, 2023-03-30:
Is C32201 the actual serial of the charger?

What charger do you have? Single / 3 Phase. 7kW, 11kW or 22kW etc?

I need to add in some serial numbers as the Integration will only return a "Unknown Serial" error at the min.

wills106, 2023-03-30:
Ok I'll release another update with the Serial in the plugin.

CristiXP, 2023-03-30:
The b12 version doesn't work either.

Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_solax_ev_charger.py:758 
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues)) 
First occurred: 8:26:58 PM (1 occurrences) 
Last logged: 8:26:58 PM

Error setting up entry Solax_EV for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 127, in async_setup_entry
    hub = SolaXModbusHub(hass, name, host, port, modbus_addr, interface, serial_port, baudrate, scan_interval, plugin, config)
  File "/config/custom_components/solax_modbus/__init__.py", line 211, in __init__
    self._invertertype = self.plugin.determineInverterType(self, config)
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 758, in determineInverterType
    seriesnumber                       = _read_serialnr(hub, 0x600,  swapbytes = False)
TypeError: _read_serialnr() got an unexpected keyword argument 'swapbytes'

wills106, 2023-03-30:
Ok, I'll correct that as well. Don't try 2023.03.2b13 that will give the same issue.

CristiXP, 2023-03-30:
Testing on my HA from the VM for testing, so no issues, also it runs very fast.

wills106, 2023-03-30:
Hopefully you'll get somewhere with [2023.03.2b14](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b14)

wills106, 2023-03-30:
How exactly is it all connected? Also where are you connecting in via Modbus?

CristiXP, 2023-03-30:
maybe I am doing something wrong? 
[image]
[image]
[image]

CristiXP, 2023-03-30:
The HA is connected to the inverter directly to the IP of the dongle... it seems it works this way.
I am trying to do the same thing with the EV Charger, connect directly modbus to the IP of the EV Charger. No modbus, no wires connected to the Inverter/ EV Charger.

wills106, 2023-03-30:
Is the WiFi on the EV Charger a Pocket WiFi?

wills106, 2023-03-30:
I don't know if the Pocket WiFI on the Inverter can pass the EV Charger over or not. But you would have to change the Modbus address to something other than 1 (For the plugin_solax_ev_charger). As 1 would default to the Inverter.

You may only be to communicate over RS485 though. I don't know.

CristiXP, 2023-03-30:
They both have 2 separate Wifi and connected to router. One is xxx.188 for the Inverter, the other is xxx.111 for the EvCharger. Both can be accessed LAN as the screenshot from above. I was trying to avoid serial cables as I don't really like that type of connection and was happy that I tried today with the inverter and it worked. Also the EVCharger at this moment is not connected via serial with the inverter, but only cloud + separate sets of CTs.

CristiXP, 2023-03-30:
I am still not sure if I will connect with serial to the inverter...waitinfg for Solax support, no ideea why the Charger doesn't start automatically when the injection is above 4kW.

wills106, 2023-03-31:
I would imagine you will get more accurate control directly connected to the Inverter.

I would imagine you would at least need to connected the charger to HA through RS485 to be able to communicate to the charger. The docs only mention Modbus over RTU. Where the Gen3 & Gen4 Hybrid mention RTU and TCP. So potentially SolaX don't plan on communicating with the charger by any means other than RS485.

wills106, 2023-04-04:
It looks like you do get better working of the Charger and Inverter when they are connected together like intended. https://github.com/wills106/homeassistant-solax-modbus/discussions/349#discussioncomment-5517629 which I would expect as they are designed to work together.

wills106, 2023-05-02:
Closing down as EV Charger needs RS485 connection.
For any further development follow along with #349

Philipp-X1, 2024-01-28:
I would like to open this up again since im trying to integrate my Solax X3-EVC-11 EV Charger to HA

**Describe the bug**
I made the Modbus connection with the connectors A1 and B1 to a Waveshare RS485 to POE ETH (B) Device. 
The Waveshare is set to Work Mode TCP Server and Protocol Modbus TCP to RTU.
When it is connected the Link is green. When i try to load the integration it just pops up blue and gets green again.
Konfiguration of the Integration should be right ( IP, Modbus Adress etc...)

But it shows no Device and no Entitys.

**Mandatory details**

    Integration Version:2024.01.7
    HA Version:Home Assistant 2024.1.5
    Inverter Brand: Solax EV Charger
    Plugin Used:Solax Modbus
    First 6 digits of Serial:C31103
    Connection Method: modbus TCP

**Detailed Error Log**
NameError: name 'ALL_PM_GROUP' is not defined
2024-01-28 10:49:02.133 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_EV: reading serial number from address 0x600 failed; other address may succeed
2024-01-28 10:49:02.133 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_EV: cannot find serial number for EV Charger
2024-01-28 10:49:02.134 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] unrecognized inverter type - serial number : unknown
2024-01-28 10:49:02.713 ERROR (MainThread) [homeassistant.components.button] Error while setting up solax_modbus platform for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/button.py", line 30, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype, button_info.allowedtypes, hub.seriesnumber, button_info.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-01-28 10:49:02.714 ERROR (MainThread) [homeassistant.components.number] Error while setting up solax_modbus platform for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/number.py", line 34, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype,newdescr.allowedtypes, hub.seriesnumber ,newdescr.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-01-28 10:49:02.716 ERROR (MainThread) [homeassistant.components.select] Error while setting up solax_modbus platform for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/select.py", line 27, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype, select_info.allowedtypes, hub.seriesnumber , select_info.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-01-28 10:49:02.718 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solax_modbus platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/sensor.py", line 89, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype,sensor_description.allowedtypes, hub.seriesnumber, sensor_description.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined

**Additional context**

Is it a Problem with the Serial Number like mentioned above?

PatrikTrestik, 2024-02-01:
This is not an issue of this plugin, but most probably technical problem on Modbus connection.
Have you tested if you can communicate on Modbus? Check correct converter setting also.
You should be able to read registers on Modbus (there is many test tools) before trying to configure the plugin.

Error above is only manifestation of error in communication.

Philipp-X1, 2024-02-01:
I tested it now and it looks like this:

[image]

is this ok?

Still no entities

PatrikTrestik, 2024-02-03:
Yes it shows that your modbus communication works well.
I see from data that you have 11kW, 3phase charger with firmware 1.12.

Try to read Holding register 1536 (0x600) and you should see your SN (char encoded).

If with same setup you try the plugin, what is the result?

Philipp-X1, 2024-02-03:
Actually I don’t know how to read the data… But I could try to get into it. It shows Data on Holding register 1536:

[image]

Do I need to delete the integration and set it up again?
Or is it enough to configure and reload it?
When I reload it in HA the Link LED on my converter shows activity ( blue ) and changes to green afterwards. 

But no Entities:

[image]

PatrikTrestik, 2024-02-03:
Check plugin configuration. 
IP address, port, Modbus ID. 

Also in logs you can find if it reads from correct device.
I expect that because you see activity, that IP and port is ok. Check Modbus ID.

You are able to read manually, so your wiring is OK.

Philipp-X1, 2024-02-03:
Where can I find the Modbus ID?

Plugin configuration looks like this:

[image]
[image]

**This ist the Log:** 

NameError: name 'ALL_PM_GROUP' is not defined
2024-02-03 21:15:24.291 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_EV: reading serial number from address 0x600 failed; other address may succeed
2024-02-03 21:15:24.291 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaX_EV: cannot find serial number for EV Charger
2024-02-03 21:15:24.291 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] unrecognized inverter type - serial number : unknown
2024-02-03 21:15:24.309 ERROR (MainThread) [homeassistant.components.button] Error while setting up solax_modbus platform for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/button.py", line 30, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype, button_info.allowedtypes, hub.seriesnumber, button_info.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-02-03 21:15:24.315 ERROR (MainThread) [homeassistant.components.number] Error while setting up solax_modbus platform for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/number.py", line 34, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype,newdescr.allowedtypes, hub.seriesnumber ,newdescr.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-02-03 21:15:24.321 ERROR (MainThread) [homeassistant.components.select] Error while setting up solax_modbus platform for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/select.py", line 27, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype, select_info.allowedtypes, hub.seriesnumber , select_info.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-02-03 21:15:24.325 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solax_modbus platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/sensor.py", line 89, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype,sensor_description.allowedtypes, hub.seriesnumber, sensor_description.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined

Where do i need to search to see if it reads from the correct device?

PatrikTrestik, 2024-02-03:
You have incorrect setting. It has to be Modbus TCP not Modbus RTU.  All other is ok.

Philipp-X1, 2024-02-04:
I have changed it to Modbus TCP and now the LINK LED on the Waveshare Converter stays blue. Seems to be good.
But still no Entities.

Log now shows this:

2024-02-04 13:45:26.014 ERROR (MainThread) [homeassistant.components.button] Error while setting up solax_modbus platform for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/button.py", line 30, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype, button_info.allowedtypes, hub.seriesnumber, button_info.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-02-04 13:45:26.018 ERROR (MainThread) [homeassistant.components.number] Error while setting up solax_modbus platform for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/number.py", line 34, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype,newdescr.allowedtypes, hub.seriesnumber ,newdescr.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-02-04 13:45:26.019 ERROR (MainThread) [homeassistant.components.select] Error while setting up solax_modbus platform for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/select.py", line 27, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype, select_info.allowedtypes, hub.seriesnumber , select_info.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined
2024-02-04 13:45:26.021 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solax_modbus platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 360, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solax_modbus/sensor.py", line 89, in async_setup_entry
    if plugin.matchInverterWithMask(hub._invertertype,sensor_description.allowedtypes, hub.seriesnumber, sensor_description.blacklist):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 783, in matchInverterWithMask
    pmmatch = ((inverterspec & entitymask & ALL_PM_GROUP)  != 0) or (entitymask & ALL_PM_GROUP  == 0)
                                            ^^^^^^^^^^^^
NameError: name 'ALL_PM_GROUP' is not defined

[image]

Any hints?

PatrikTrestik, 2024-02-04:
Do you have latest version of integration?
I have impression that this error was fixed already.

I'll look tomorrow to code.

Philipp-X1, 2024-02-04:
Yes, everything latest Version.
Integration 2024.01.8

PatrikTrestik, 2024-02-05:
Fixed here https://github.com/wills106/homeassistant-solax-modbus/pull/722

Philipp-X1, 2024-02-05:
Awesome! Thank you so much!

But there are quite some values that seem not to be right.

[image]

For example sensor.solax_ev_charge_added_total should be somewhere around 1500kWh at my charger but it shows jumping numbers like xxx.xxx,xx kWh

also no value for sensor.solax_ev_charge_voltage_l1 sensor.solax_ev_charge_voltage_l2 and sensor.solax_ev_charge_voltage_l3 ?

sensor.solax_ev_charge_frequency_l1 sensor.solax_ev_charge_frequency_l2 sensor.solax_ev_charge_frequency_l3 seem not to be right as well

same with sensor.solax_ev_charge_current_l1 sensor.solax_ev_charge_current_l12 sensor.solax_ev_charge_current_l13

And sensor.solax_ev_charge_power_total ist somewhere around 5000W even when it ist not charging at all

sensor.solax_ev_charge_added doesnt get any value at all

I am charging only on 1 Phase. And I am getting Charge Power L3 Values which seem to be right.

I hope I'm not annoying anyone. It just seems like there are some bugs? Or do i still have some issues with setting up?

Philipp-X1, 2024-02-05:
Well it seems like i am the only one... But i know that there are a bunch of people interested in this.
Theres a German Photovoltaic community with quite some Solax EV Charger owners and some of them are interested.
Problem is often that people don´t even know about the integrations.

But I wanted to test this in advance bevore telling people about it.

I tried your HTTP API which seemed to work good for me.
Is it ok if i make it public in German Community so you might get some more users? 

And theres no need to hurry. Maybe theres later some time to fix the modbus plugin. Or someboy else got time :)

Thanks anyway your work here is highly appreciated!

PatrikTrestik, 2024-02-05:
Feel free. Http api integration is HACS compatible and can be added through it.

I created it because there is no option to connect Modbus and keep smart functions.
I need Green charging to work and so I need Modbus connected to  Inverter.

wills106, 2024-02-07:
Just released [2024.02.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.1) with the missing flag.

> But there are quite some values that seem not to be right.

You probably are the first person to use the Modbus connection. So there probably are values that need to be adjusted.

Would need to go back over the docs and see if any of the registers are mismatched.

> And sensor.solax_ev_charge_power_total ist somewhere around 5000W

Was this the amount from the last charge session?

Philipp-X1, 2024-02-07:
I don’t think so, because I can only charge 1Phase 16A.

The value changes continuously somewhere around 5000W:

[image]

And there’s no change even when I start or stop charging.

How can I help with the mismatched registers? 
I got GModbus to test but I’m unfortunately not into all this Modbus related things.

But maybe I can contribute somehow.

PatrikTrestik, 2024-02-07:
> Just released [2024.02.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.1) with the missing flag.
>
>
> You probably are the first person to use the Modbus connection. So [...]

I had similar issues on HTTP. Values are same as on Modbus. 
Look to my HTTP integration, you may find value formatting solution there.

I don't want to commit, but there is chance that I find time to contribute over weekend.

PatrikTrestik, 2024-02-13:
I checked the plugin code and there seems to be no obvious error in reading these values. 
I'll connect my changer on Modbus to see if I can find a bug.

sonypsx, 2024-02-16:
So finally i added EV Charger (11k) to HA (2024.2.1) and its connected via MODBus TCP. Cable  runs from the inverter to the Modbus converter and from there to the EV charger. EV Charger is shown in the Solax App and reports its duty.
But i am really unsure if HA can even talk to it.
I have all my values for the Inverter (which is using the same Waveshare Adapter) and this is working fine since months!
I see 2 devices and 31 Entities for the EV Charger and for the Inverter i see 290 entities.
The inverter entities are up2date and refreshing fine, but the ones for the EV charger are empty.
Any hint what i did wrong?

marbeja, 2024-02-20:
While trying on my SPF Inverter via ethernet modbus I accidentally put the charger  integration and got a lot of empty entities, even without connection

Philipp-X1, 2024-02-20:
> So finally i added EV Charger (11k) to HA (2024.2.1) and its connected via MODBus TCP. Cable runs from the inverter to the Modbus converter and from there to the EV charger. EV Charger is shown in [...]

Are you using 2 Waveshare Adapter? Or are you trying to daisy chain? ( As far as i know daisy chaining isnt going to work )

sonypsx, 2024-02-21:
> Are you using 2 Waveshare Adapter? Or are you trying to daisy chain? ( As far as i know daisy chaining isnt going to work )

Currently i am daisy chaining. Why? Because the Inverter and the EV Box has to be connected together, otherwise the INV won't know the EV and vice versa. Thats my current knowledge.
But enlight me , is there a different way to cable this and take another Waveshare AND have HA and the Solax App up2date?
Would be cool to solve this riddle! Thank you!

sonypsx, 2024-02-21:
> > > > So finally i added EV Charger (11k) to HA (2024.2.1) and its connected via MODBus TCP. Cable runs from the inverter to the Modbus converter and from there to the EV charger. EV Charger is shown in the Solax App and reports its duty. But i am really unsure if HA can even talk to it. I have all my values for the Inverter (which is using the same Waveshare Adapter) and this is working fine since months! I see 2 devices and 31 Entities for the EV Charger and for the Inverter i see 290 entities. The inverter entities are up2date and refreshing fine, but the ones for the EV charger are empty. Any hint what i did wrong?
> > > 
> > > 
> > > Are you using 2 Waveshare Adapter? Or are you trying to daisy chain? ( As far as i know daisy chaining isnt going to work )
> > 
> > 
> > Currently i am daisy chaining. Why? Because the Inverter and the EV Box has to be connected together, otherwise the INV won't know the EV and vice versa. Thats my current knowledge. But enlight me , is there a different way to cable this and take another Waveshare AND have HA and the Solax App up2date? Would be cool to solve this riddle! Thank you!
> 
> I'll be short, because there is lots of written about it here.
> 
> You could not daisy chain. Modbus is Master-Slave protocol on wire. Only ONE master is allowed on chain. Here Inverter is Master and you could not add another. You can daisy-chain to other Slave (for example metter) but never Master.
> 
> You are lucky that all devices has protections. In theory your Inverter should by destroyed by now. But it has good protection circuits so it survives all this.
> 
> :)

Uff! No i am a bit scared, but also glad that nothing serious did happen.
I have read many threads, but didn't stumbled over that info.

So what would be the proper way to cable Inverter, EV and Waveshare (maybe second Waveshare) to have everything working (HA,Solax App and communication between Inverter and EV)?

sonypsx, 2024-02-21:
Thank you about that HA warning, already being obsessed with it, runnning HA now over months, and having fun to try out new stuff while enjoying accurate monitoring of PV and several other things!

Already ordered a second Waveshared while reading the Wiki for it, which states that it can do Both Modes.
[https://www.waveshare.com/wiki/RS485_TO_ETH_(B)](url)

Thank you very much for pointing that out! Will post soon results about working or not, but i guess this time its a go!

Philipp-X1, 2024-02-21:
If you want to use a second Converter for your EV Charger you cant use the Connection Inverter <-> EV-Charger anymore. Even if the RS 485 connector in your EV-Charger isnt connected it is internally daisy chained to the RJ45 Port. 

As Patrik said this is mentioned bevore somewhere here :)

But i am glad that there will be a second User to integrate the wired connection to HA.
( Patrik has got a working Integration which uses HTTP API over the Wifi connection of the EV-Charger. Maybe you want to try this as well )

Speaking about the RS485 - Converter connection: 
There seem to be some issues as for now and not all Data transmitted to HA is right. ( At least at my setup )

Patrik I know you dont have the Waveshare but can you check if my converter settings appear to be right?

[image]

Thank you :)

PatrikTrestik, 2024-02-23:
Fixed plugin is here https://github.com/wills106/homeassistant-solax-modbus/pull/772

Unfortunately not tested a lot, but I hope it helps you.

@wills106 please review and merge when you have a minute.

sonypsx, 2024-02-24:
And here is my small Update...
Connected the second Waveshare to the EV without any other connection except Power an WLAN.
Link Led went from Green to Blue so at least there is a connection -> got the same settings as Philipp have.
But no luck in HA.

The only Info whichare in the HA log are not very promising:
`2024-02-24 17:13:04.659 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax_ev_charger] SolaXEV: attempt to read serialnumber failed at 0x600
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax_ev_charger.py", line 59, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 370, in async_read_holding_registers
    return await self._client.read_holding_registers(address, count, **kwargs)`

However i think i will stick to the HTTP Local API which is working perfect!

Philipp-X1, 2024-02-24:
Are you running the latest version of the integration? There was an issue which has been fixed

sonypsx, 2024-02-24:
Everything is up2date HACS, HA, all Plugins etc. I'll simply go for the HTTP API - much easier to handle and most of all -> It works!

PatrikTrestik, 2024-02-26:
Please don't continue in this thread, it is too long and confusing.

If you have a question "How to ...?", start Discussion.
For errors please add new Issue with simple and exact title. Better to have 10 small issues open than this long one.

Most of errors are solved in https://github.com/wills106/homeassistant-solax-modbus/pull/772
@wills106 Please close this spaghetti issue.

wills106, 2024-02-26:
Closing down as issue has got too long to follow.

closed 2024-02-26 by wills106