wills106/homeassistant-solax-modbus #372: [BUG] sofar HYD 3-6 data inconsistencies

Matz88, 2023-04-01:
First of all thank you very much for the great work here!

I'm trying to figure out few odd things with the data retrieval to a monopjase sofar hyd 3-6 inverter branded as zcs azzurro hyd. I'm on 2023.03.2b14 beta version.

I'm getting almost all data as in the app but I miss the inverter temperature, I have only the heatsink temperature which is much lower.

Also the PV power is odd, PV1 and PV2 are ok but PV total always shows a value that is not the sum of the two above but less, and not necessarily equal to one of the two.

Battery active control: what is for? And in my case does toggle on request but then comes back immediately to disabled. Also I cannot modify the maximum depth of discharges, they just revert to the old values immediately after.

At your disposal for any test and additional information! Many thanks

wills106, 2023-04-05:
Does one of the Module Temps match the Inverter Temp?

PV Total is an internal register of the Inverter. How far out is it? If it's only a small amount it might be internal precision as PV1 & PV2 are 0.01 scaled and PV Total is 0.1

> General exception: unpack requires a buffer of 4 bytes
> Modbus IO exception Modbus Error: [Input/Output] Unable to decode request`

Which register do you get this error with?

Matz88, 2023-04-05:
PV tot is quite far from the sum of PV1 and PV2, see below, but could be indeed only a matter of scaling as you said, here powers are quite low since I have return to grid set to 0 so far:
[image]

About the temps, I have checked also all other modules temperature, since there was an open issue on this, but none is different from 0 but module 1 and heatsink.

About the exception, how can I know the register triggering the error? I just know that is linked to a tentative of setting the mode to passive.

Matz88, 2023-05-03:
@wills106 as in my last comment, all fine for the PV power, understood, what about the inverter temperature? I see only the heatsink temperature but on the solarman app I can see also the inverter temperature that is higher.

Also is there any chance to command remotely the inverter? Looks like I can change the inverter mode but not the depth of discharge, any change is ignore and value on the slider comes back to the original value in few seconds

Matz88, 2023-06-08:
@wills106 I found the issue...for the monophase sofar hyd 3-6 inverter branded as zcs azzurro hyd, serial ZM2ESxxxxxx the internal inverter temperature is under ambient temperature, would you just mind to add the following two? Tested manually and works:
[image]

wills106, 2023-06-08:
I'm sure I added them in previously? Maybe I deleted them by mistake.
Unless I'm thinking of 0420-0422.

I'll add them in.

wills106, 2023-06-08:
I have just released [2023.06.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.06.3) with the two Inverter Temperatures

Matz88, 2023-06-08:
thank you very much!

closed 2023-06-08 by Matz88