wills106/homeassistant-solax-modbus #373: sensor.solis_power_generation_total is reporting this year's total instead of the sum of all years

rcourtman, 2023-04-01:
*note - this looks like it's probably an issue with the solis inverter itself as I'm seeing the inverter reporting this number incorrectly. But if there is a workout to it then it would be great to know. 

The sensor.solis_power_generation_total entity in our system is reporting the total power generation for this year instead of the sum of all years. 

{'entity_id': 'sensor.solis_power_generation_last_year', 'state': '624', 'unit_of_measurement': 'kWh'}
{'entity_id': 'sensor.solis_power_generation_this_year', 'state': '1099', 'unit_of_measurement': 'kWh'}
{'entity_id': 'sensor.solis_power_generation_total', 'state': '1099', 'unit_of_measurement': 'kWh'}

Additional Information

sensor.solis_power_generation_last_year and sensor.solis_power_generation_this_year entities are reporting accurate data (when checked against the inverters display).

Solis Hybrid, modbus. 

[image]
[image]

Fantastic work by the way on this intergration, it works beautifully.

wills106, 2023-04-02:
'sensor.solis_power_generation_total' is mapped to 33029
'sensor.solis_power_generation_this_year' is mapped to 33037

The only thing I can think of is that in the firmware 'sensor.solis_power_generation_total' is mapped to 33037 so it's reading the wrong value?

Have you asked Solis if there is a firmware update?

rcourtman, 2023-04-07:
I'm waiting to hear back, will post a response if/when I get one.

alienatedsec, 2023-04-07:
@rcourtman what is your model and firmware version at the moment?

rcourtman, 2023-04-14:
> @rcourtman what is your model and firmware version at the moment?

I believe it's the RHI-(3-6)K-48ES-5G
HMI - 42
DSP - 3F

alienatedsec, 2023-04-14:
@rcourtman Which exact RHI?

`RHI-6K-48ES-5G` is the 6kWh model

rcourtman, 2023-04-14:
Update*

The Solis team responded and resolved the issue with my inverter remotely. They simply requested me to plug in the Wifi Logger stick, allowing them to access and fix it. It is uncertain whether it was a firmware problem or just an isolated issue with my device. As a result, I will consider this issue resolved. Thanks!  

[image]

Thanks

rcourtman, 2023-04-14:
> 

The 3.6kW model

closed 2023-04-14 by rcourtman