wills106/homeassistant-solax-modbus #374: [BUG] Growatt plugin missing swapbyte argument

JohNan, 2023-04-05:
**Describe the bug**

Growatt plugin is missing an argument when trying to read serial number

**Mandatory details**
1. Integration Version: 2023.03.1
2. HA Version: 2023.3.6
3. Inverter Brand: Growatt
4. Plugin Used: growatt
5. First 6 digits of Serial: PZLOCGR
6. Connection Method: Serial, USB to RS485

**Detailed Error Log**
```
Logger: homeassistant.config_entries
Source: custom_components/solax_modbus/plugin_growatt.py:1855
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 19:40:15 (2 occurrences)
Last logged: 19:50:34

Error setting up entry growatt_solax for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solax_modbus/__init__.py", line 127, in async_setup_entry
    hub = SolaXModbusHub(hass, name, host, port, modbus_addr, interface, serial_port, baudrate, scan_interval, plugin, config)
  File "/config/custom_components/solax_modbus/__init__.py", line 211, in __init__
    self._invertertype = self.plugin.determineInverterType(self, config)
  File "/config/custom_components/solax_modbus/plugin_growatt.py", line 1855, in determineInverterType
    seriesnumber                       = _read_serialnr(hub, 23)
TypeError: _read_serialnr() missing 1 required positional argument: 'swapbytes'
```

wills106, 2023-04-05:
I have removed the leftover swapbyte code in [2023.03b20](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03b20) can you try this version and let me know how you get on?

wills106, 2023-04-05:
Which Growatt Inverter are you using? As I'm not using the Serial number for detection but the Firmware version.

JohNan, 2023-04-05:
The integration starts now and the previous error is gone, thank you!

But as I suspected my Inverter is not recognized. It's not int the list of known working Inverters, so it was a bit of a gamble.

The model is Growatt MOD 6KTL3-XH

JohNan, 2023-04-05:
Wow, it is working! Thank a lot for such quick response!

However some values is a bit off.

Left side is from another integration that pulls from the cloud. Right side is your integration. 

[image]

wills106, 2023-04-09:
> However some values is a bit off.

Are you able to get some real time values from off you Inverter LCD matched to what's reported in the Integration? Then I can look at the rounding better.
Using the values from a cloud Integration can be difficult as there is normally a 5 - 10 min delay in what it reports. So you can end up with two completely different readings.

wills106, 2023-04-29:
Closing down due to lack of feedback.

closed 2023-04-29 by wills106