wills106/homeassistant-solax-modbus #377: [BUG]unrecognized Growatt inverter type

super-bedst, 2023-04-10:
Read Growatt 0x9 serial number before potential swap: RA1.0 ZCBA
A clear and concise description of what the bug is.

**Mandatory details**
1. Integration Version: 2023.03.2b22
2. HA Version: 2023.03.5
3. Inverter Brand: Growatt SPH 3600
4. Plugin Used: Growatt
5. First 6 digits of Serial: OMCUCC
6. Connection Method: IP Waveshare

**Detailed Error Log**
2023-04-10 11:43:55.321 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized Growatt inverter type - serial number : RA1.0 ZCBA

Can the type be adden it is a Growatt SPH 3600 Hybrid inverter with Battery

wills106, 2023-04-10:
Is this a single phase one?

wills106, 2023-04-10:
It looks like it is Single Phase so I have released [2023.03.2b23](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.03.2b23)

Are you able to have a read of https://github.com/wills106/homeassistant-solax-modbus/issues/374#issuecomment-1501065014 and let me know any corrections.
Could you reply in this issue though, as that issue is for a different model. So corrections might not pass between them as they are using a different register set between them.

super-bedst, 2023-04-10:
It is a Single Phase inverter. at the beginning i have the same issue as #374 but after going to b2 it have connection.
i got this for the log:
2023-04-10 11:43:55.261 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating host: returning data: {'host': '192.168.0.32', 'port': 4196}
2023-04-10 11:43:55.269 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: growatt
2023-04-10 11:43:55.270 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Growatt: trying to determine inverter type
2023-04-10 11:43:55.320 INFO (MainThread) [custom_components.solax_modbus.plugin_growatt] Read Growatt 0x9 serial number before potential swap: RA1.0 ZCBA
2023-04-10 11:43:55.321 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized Growatt inverter type - serial number : RA1.0 ZCBA
2023-04-10 11:43:55.328 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2023-04-10 11:43:55.329 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2023-04-10 11:43:55.330 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2023-04-10 11:43:55.333 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-04-10 11:43:55.341 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None
2023-04-10 11:43:55.342 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xf language to block with start 0xf
2023-04-10 11:43:55.342 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x2d rtc to block with start 0xf
2023-04-10 11:43:55.342 INFO (MainThread) [custom_components.solax_modbus.sensor] Growatt returning holding block: 0xf 0x33 [15, 45]
2023-04-10 11:43:55.342 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}

super-bedst, 2023-04-10:
I Have connection after upgrade to [2023.03.2b23] is it possible to add function to change operation types:
1 Grid First - where the system prioritises export before sending to battery/house
2 Battery First - where the system prioritises the charging of batteries before anything else.
3 Load First - where the system prioritises supplying the house before charging the batteries or exporting to the grid.

https://www.amosplanet.org/wp-content/uploads/2022/12/%E9%80%9A%E4%BF%A1%E5%8D%8F%E8%AE%AEMAX_Growatt-Inverter-Modbus-RTU-Protocol_II-V1_13_20190116-English.pdf

Page 41 register 118

closed 2023-04-10 by super-bedst

wills106, 2023-04-10:
Yes, I can add in write support. I just wanted to ensure the Inverters worked ok first.

wills106, 2023-04-22:
Just had a look at the register set.

118 is an input register. So read only. There must be a separate holding register to write to, so you can change the mode.

wills106, 2023-04-24:
Do you know what the difference between Grid First and Battery First is?
Does Grid First still charge the battery? Or is it time to only use the Grid?